Xilinx/video-sdk #1: U30 FFMPEG not passing Audio

ssellaySkreens, 2021-08-31:
I have an mp4 file that I am transcoding to multiple rtmp endpoints. I can not figure out how to get the audio to be encoded in the new streams.

**My command:**
/tmp/ffmpeg/bin/ffmpeg -re -stream_loop -1 -c:v mpsoc_vcu_h264 -entropy_buffers_count 2 -low_latency 1 -c:a aac -i ~/Braves-Nationals_MLB.mp4 -filter_complex "multiscale_xma=ots=4: \
out_1_width=1280: out_1_height=720:  out_1_rate=full: \
out_2_width=848:  out_2_height=480:  out_2_rate=half: \
out_3_width=640:  out_3_height=360:  out_3_rate=half: \
out_4_width=288:  out_4_height=160:  out_4_rate=half  \
[a][b][c][d]; [a]split[aa][ab];[ab]fps=30[abb]" \ 
-map "[aa]" -c:a aac  -b:v 4M -profile:v baseline -c:v mpsoc_vcu_h264 -f flv rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/1  \
-map "[abb]" -c:a aac -b:v 3M -profile:v baseline -c:v mpsoc_vcu_h264 -f flv rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/2 \
-map "[b]" -c:a aac -b:v 2500K -profile:v baseline -c:v mpsoc_vcu_h264 -f flv rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/3 \
-map "[c]" -c:a aac -b:v 1250K -profile:v baseline -c:v mpsoc_vcu_h264 -f flv rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/4 \
-map "[d]" -c:a aac -b:v 625K -profile:v baseline -c:v mpsoc_vcu_h264 -f flv rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/5

**The results from FFMPEG:**
ffmpeg version 4.1.9 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.5.0-3ubuntu1~18.04)
  configuration: --prefix=/tmp/ffmpeg --datadir=/tmp/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/skreens/Braves-Nationals_MLB.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf59.3.101
  Duration: 02:46:47.07, start: 0.008005, bitrate: 5203 kb/s
    Stream #0:0(und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709), 1280x720 [SAR 1:1 DAR 16:9], 5063 kb/s, 59.94 fps, 59.94 tbr, 60k tbn, 119.88 tbc (default)
    Metadata:
      handler_name    : VideoHandler
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 128 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
------------------------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0 
------------------------------------------------------------

XMA INFO: XILINX_XRT env variable not set. Trying default /opt/xilinx/xrt
Stream mapping:
  Stream #0:0 (mpsoc_vcu_h264) -> multiscale_xma
  split:output0 -> Stream #0:0 (mpsoc_vcu_h264)
  fps -> Stream #1:0 (mpsoc_vcu_h264)
  multiscale_xma:output1 -> Stream #2:0 (mpsoc_vcu_h264)
  multiscale_xma:output2 -> Stream #3:0 (mpsoc_vcu_h264)
  multiscale_xma:output3 -> Stream #4:0 (mpsoc_vcu_h264)
Press [q] to stop, [?] for help
[mpsoc_vcu_h264 @ 0x558655487740] Custom Rate Control Mode is Disabled
Output #0, flv, to 'rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/1':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.20.100
    Stream #0:0: Video: h264 (mpsoc_vcu_h264) ([7][0][0][0] / 0x0007), xlnx_xvbm, 1280x720 [SAR 1:1 DAR 16:9], q=2-31, 4000 kb/s, 59.94 fps, 1k tbn, 59.94 tbc (default)
    Metadata:
      encoder         : Lavc58.35.100 mpsoc_vcu_h264
[mpsoc_vcu_h264 @ 0x558655574e00] Custom Rate Control Mode is Disabled
Output #1, flv, to 'rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/2':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.20.100
    Stream #1:0: Video: h264 (mpsoc_vcu_h264) ([7][0][0][0] / 0x0007), xlnx_xvbm, 1280x720 [SAR 1:1 DAR 16:9], q=2-31, 3000 kb/s, 30 fps, 1k tbn, 30 tbc (default)
    Metadata:
      encoder         : Lavc58.35.100 mpsoc_vcu_h264
[mpsoc_vcu_h264 @ 0x558655579680] Custom Rate Control Mode is Disabled
Output #2, flv, to 'rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/3':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.20.100
    Stream #2:0: Video: h264 (mpsoc_vcu_h264) ([7][0][0][0] / 0x0007), xlnx_xvbm, 848x480 [SAR 1:1 DAR 53:30], q=2-31, 2500 kb/s, 29.97 fps, 1k tbn, 29.97 tbc (default)
    Metadata:
      encoder         : Lavc58.35.100 mpsoc_vcu_h264
[mpsoc_vcu_h264 @ 0x55865556b400] Custom Rate Control Mode is Disabled
Output #3, flv, to 'rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.20.100
    Stream #3:0: Video: h264 (mpsoc_vcu_h264) ([7][0][0][0] / 0x0007), xlnx_xvbm, 640x360 [SAR 1:1 DAR 16:9], q=2-31, 1250 kb/s, 29.97 fps, 1k tbn, 29.97 tbc (default)
    Metadata:
      encoder         : Lavc58.35.100 mpsoc_vcu_h264
[mpsoc_vcu_h264 @ 0x5586558740c0] Custom Rate Control Mode is Disabled
Output #4, flv, to 'rtmp://bcast-2f8x62sjlrgs6ylhv5fha2hvym.skreens.tv:8100/live/5':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.20.100
    Stream #4:0: Video: h264 (mpsoc_vcu_h264) ([7][0][0][0] / 0x0007), xlnx_xvbm, 288x160 [SAR 1:1 DAR 9:5], q=2-31, 625 kb/s, 29.97 fps, 1k tbn, 29.97 tbc (default)
    Metadata:
      encoder         : Lavc58.35.100 mpsoc_vcu_h264
frame=44502 fps= 60 q=-0.0 q=-0.0 q=-0.0 q=-0.0 q=-0.0 size=  363293kB time=00:12:22.37 bitrate=4008.9kbits/s speed=   1x   

as you can see there is no audio stream on the outputs even though there is on the inputs.

khamamot-xlnx, 2021-09-04:
The reason ffmpeg is not outputting audio is because it is not being told to. It has 1 input and 5 outputs and since it is not being told to split the audio, it doesn't know which output should get it. Use something along the lines of the following script:
ffmpeg -re -stream_loop -1 -c:v mpsoc_vcu_h264 -entropy_buffers_count 2 -low_latency 1 -c:a aac -i ~/Ninja-1-H264-source-1080p60.flv -filter_complex "multiscale_xma=outputs=4: \
out_1_width=1280: out_1_height=720: out_1_rate=full: \
out_2_width=848: out_2_height=480: out_2_rate=half: \
out_3_width=640: out_3_height=360: out_3_rate=half: \
out_4_width=288: out_4_height=160: out_4_rate=half \
[a][b][c][d]; [a]split[aa][ab];[ab]fps=30[abb];**[0:1]asplit=outputs=5[aud1][aud2][aud3][aud4][aud5]**" \
-map "[aa]" -b:v 4M -profile:v baseline -c:v mpsoc_vcu_h264 **-map "[aud1]"** -c:a aac -f flv out1.flv \
-map "[abb]" -b:v 3M -profile:v baseline -c:v mpsoc_vcu_h264 **-map "[aud2]"** -c:a aac -f flv out2.flv \
-map "[b]" -b:v 2500K -profile:v baseline -c:v mpsoc_vcu_h264 **-map "[aud3]"** -c:a aac -f flv out3.flv \
-map "[c]" -b:v 1250K -profile:v baseline -c:v mpsoc_vcu_h264 **-map "[aud4]"** -c:a aac -f flv out4.flv \
-map "[d]" -b:v 625K -profile:v baseline -c:v mpsoc_vcu_h264 **-map "[aud5]"** -c:a aac -f flv out5.flv

ThomasXilinx, 2021-09-14:
@ssellaySkreens, I believe this issue can now be closed. Feel free to reopen it if your issue with audio persists.

closed 2021-09-14 by ThomasXilinx

ssellaySkreens, 2021-09-14:
This is closed... audio is working fine.

Thank you!

---
Spencer Sellay
Sr Director of Field Engineering and Business Development
email: ***@***.***
mobile: +1.303.523.6420

www.skreens.com

NOTICE: This communication and any attachments may contain confidential
and/or privileged information of Skreens Technologies. This communication
is intended solely for the use of the individual or entity to which it is
addressed. If you are not the intended recipient of this communication, you
are hereby notified that any dissemination, distribution, copying or other
use of this communication or the information it contains is strictly
prohibited and may be unlawful. If you have received this communication in
error, please notify the sender immediately by reply e-mail and destroy all
copies of this communication (including attachments), electronic or
otherwise. If you are a party to a written Non-Disclosure Agreement ("NDA")
with Skreens Technologies, you should maintain the contents of this
communication (including attachments) in confidence to the full extent
specified in the NDA.