Xilinx/video-sdk #2: ABR Ladder example: 1080p60 to 720p60, 720p30, 480p30, 360p30, 160p30 takes 1/4 of decoder

ssellaySkreens, 2021-09-06:
The documentation states that each card can handle a total aggregate bandwidth of two 3840x2160@60 fps per card.  When I try to run more than 4 of these configurations the 5 job always fails.  According to the documentation I should be able to get 8 of these jobs running simultaneously.

When I run a single input with the following command it takes up 1/4 of the decoder available:

ffmpeg -c:v mpsoc_vcu_h264 -entropy_buffers_count 2 -low_latency 1 -c:a aac -i rtmp://35.199.33.235:7162/live/stream -filter_complex "[0:1]multiscale_xma=outputs=5: out_1_width=1280: out_1_height=720: out_1_rate=full: out_2_width=1280: out_2_height=720: out_2_rate=half: out_3_width=852: out_3_height=480: out_3_rate=half: out_4_width=640: out_4_height=360: out_4_rate=half: out_5_width=288: out_5_height=160: out_5_rate=half [a][b][c][d][e]; [0:0]asplit=outputs=5[auda][audb][audc][audd][aude]" -map [a] -b:v 4M -profile:v baseline -c:v mpsoc_vcu_h264 -map [auda] -c:a aac -f flv rtmp://35.199.33.235:7000/live/stream -map [b] -b:v 3M -profile:v baseline -c:v mpsoc_vcu_h264 -map [audb] -c:a aac -f flv rtmp://35.199.33.235:7045/live/stream -map [c] -b:v 2500K -profile:v baseline -c:v mpsoc_vcu_h264 -map [audc] -c:a aac -f flv rtmp://35.221.49.159:7275/live/stream -map [d] -b:v 2150K -profile:v baseline -c:v mpsoc_vcu_h264 -map [audd] -c:a aac -f flv rtmp://35.221.49.159:7325/live/stream -map [e] -b:v 625K -profile:v baseline -c:v mpsoc_vcu_h264 -map [aude] -c:a aac -f flv rtmp://35.221.49.159:7030/live/stream
ffmpeg version 4.1.9 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.5.0-3ubuntu1~18.04)
  configuration: --prefix=/tmp/ffmpeg --datadir=/tmp/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
Input #0, flv, from 'rtmp://35.199.33.235:7162/live/stream':
  Metadata:
    Server          : NGINX RTMP (github.com/arut/nginx-rtmp-module)
    displayWidth    : 1920
    displayHeight   : 1080
    fps             : 60
    profile         : 
    level           : 
  Duration: 00:00:00.00, start: 356251.370000, bitrate: N/A
    Stream #0:0: Audio: aac (LC), 48000 Hz, stereo, fltp, 128 kb/s
    Stream #0:1: Video: h264 (Main), yuv420p(progressive), 1920x1080 [SAR 1:1 DAR 16:9], 5113 kb/s, 60 fps, 60 tbr, 1k tbn, 120 tbc
------------------------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0 
------------------------------------------------------------

XMA INFO: XILINX_XRT env variable not set. Trying default /opt/xilinx/xrt
Stream mapping:
  Stream #0:0 (aac) -> asplit
  Stream #0:1 (mpsoc_vcu_h264) -> multiscale_xma
  multiscale_xma:output0 -> Stream #0:0 (mpsoc_vcu_h264)
  asplit:output0 -> Stream #0:1 (aac)
  multiscale_xma:output1 -> Stream #1:0 (mpsoc_vcu_h264)
  asplit:output1 -> Stream #1:1 (aac)
  multiscale_xma:output2 -> Stream #2:0 (mpsoc_vcu_h264)
  asplit:output2 -> Stream #2:1 (aac)
  multiscale_xma:output3 -> Stream #3:0 (mpsoc_vcu_h264)
  asplit:output3 -> Stream #3:1 (aac)
  multiscale_xma:output4 -> Stream #4:0 (mpsoc_vcu_h264)
  asplit:output4 -> Stream #4:1 (aac)
Press [q] to stop, [?] for help
[mpsoc_vcu_h264 @ 0x5556306db540] Custom Rate Control Mode is Disabled
Output #0, flv, to 'rtmp://35.199.33.235:7000/live/stream':
  Metadata:
    Server          : NGINX RTMP (github.com/arut/nginx-rtmp-module)
    displayWidth    : 1920
    displayHeight   : 1080
    fps             : 60
    profile         : 
    level           : 
    encoder         : Lavf58.20.100
    Stream #0:0: Video: h264 (mpsoc_vcu_h264) ([7][0][0][0] / 0x0007), xlnx_xvbm(progressive), 1280x720 [SAR 1:1 DAR 16:9], q=2-31, 4000 kb/s, 60 fps, 1k tbn, 60 tbc (default)
    Metadata:
      encoder         : Lavc58.35.100 mpsoc_vcu_h264
    Stream #0:1: Audio: aac (LC) ([10][0][0][0] / 0x000A), 48000 Hz, stereo, fltp, 128 kb/s (default)
    Metadata:
      encoder         : Lavc58.35.100 aac
[mpsoc_vcu_h264 @ 0x555630551f00] Custom Rate Control Mode is Disabled
Output #1, flv, to 'rtmp://35.199.33.235:7045/live/stream':
  Metadata:
    Server          : NGINX RTMP (github.com/arut/nginx-rtmp-module)
    displayWidth    : 1920
[... 1048 lines cut ...]
                    "cuName       ": "kernel_vcu_encoder",
                    "kernelPlugin ": "",
                    "maxCapacity  ": "",
                    "numChanInuse ": "0",
                    "usedLoad     ": "0 of 1000000",
                    "reservedLoad ": "0 of 1000000",
                    "resrvUsedLoad": "0 of 1000000"
                }
            },
            "xrmPluginNumber": "3",
            "xrmPlugin_0": {
                "xrmPluginName    ": "xrmU30DecPlugin",
                "xrmPluginFileName": "/opt/xilinx/xrm/plugin/libxrmU30Decoder.so"
            },
            "xrmPlugin_1": {
                "xrmPluginName    ": "xrmU30ScalPlugin",
                "xrmPluginFileName": "/opt/xilinx/xrm/plugin/libxrmU30Scaler.so"
            },
            "xrmPlugin_2": {
                "xrmPluginName    ": "xrmU30EncPlugin",
                "xrmPluginFileName": "/opt/xilinx/xrm/plugin/libxrmU30Encoder.so"
            }
        }
    }
}

ThomasXilinx, 2021-09-07:
Hello,

A U30 card with its two devices is indeed capable of handling 8 such jobs. But as explained in the documentation, you will need to either use "explicit device IDs" or "job description files" to distribute jobs across multiple devices. See here: https://xilinx.github.io/video-sdk/v1.0/running_multiple_jobs.html

Using explicit device IDs (with the ffmpeg -xlnx_hwdev option) is the easiest method. You will want to run 4 of your jobs with "-xlnx_hwdev 0" and the other 4 with "-xlnx_hwdev 1" to distribute your jobs across both devices.

Thanks for closing this issue if this response answers your question.

ssellaySkreens, 2021-09-07:
Ok... except I can't seem get the second device to be accessible.  Only device 0 shows up.  How can I get the second device to be available?

$xbutil examine
System Configuration
  OS Name              : Linux
  Release              : 4.15.0-154-generic
  Version              : #161-Ubuntu SMP Fri Jul 30 13:04:17 UTC 2021
  Machine              : x86_64
  CPU Cores            : 12
  Memory               : 7631 MB
  Distribution         : Ubuntu 18.04.5 LTS
  GLIBC                : 2.27
  Model                : To Be Filled By O.E.M.

XRT
  Version              : 2.10.43
  Branch               : u30
  Hash                 : 41f6054b2c19619ceceb07bf7b662e5003388d89
  Hash Date            : 2021-05-11 09:17:17
  XOCL                 : 2.10.43, 41f6054b2c19619ceceb07bf7b662e5003388d89
  XCLMGMT              : 2.10.43, 41f6054b2c19619ceceb07bf7b662e5003388d89

Devices present
  [0000:02:00.1] : xilinx_u30_gen3x4_base_1 

$./u30flashall.sh
XILINX_XRT        : /opt/xilinx/xrt
PATH              : /opt/xilinx/xrt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
LD_LIBRARY_PATH   : /opt/xilinx/xrt/lib:
PYTHONPATH        : /opt/xilinx/xrt/python:
Discovering installed devices....

=====================================================
[1]: Updating flash image(s) for device: 0000:02:00.0
=====================================================
xbmgmt program --device 0000:02:00.0 --base
----------------------------------------------------
Device : [0000:02:00.0]

Current Configuration
  Platform             : xilinx_u30_gen3x4_base_1
  SC Version           : 6.2.10(FIXED)
  Platform ID          : 0x323002f54d799c04

Incoming Configuration
  Deployment File      : partition.xsabin
  Deployment Directory : /lib/firmware/xilinx/323002f54d799c04786db52be50c3dae
  Size                 : 64,121,465 bytes
  Timestamp            : Tue Sep  7 15:07:04 2021

  Platform             : xilinx_u30_gen3x4_base_1
  SC Version           : 6.3.8
  Platform UUID        : 323002F5-4D79-9C04-786D-B52BE50C3DAE
----------------------------------------------------
----------------------------------------------------
Report

Device(s) up-to-date and do not need to be flashed.

ThomasXilinx, 2021-09-07:
Each Alveo U30 card has 2 PCIe Gen 3x4 compliant interfaces (one for each device on the card). 
Is your host sever configured to support PCIe bifurcation on the PCIe slot with your Alveo U30 card?

ssellaySkreens, 2021-09-07:
I do see both interfaces.

lspci -vvv produces this for the xilinx board:

02:00.0 Processing accelerators: Xilinx Corporation Device 513c
	Subsystem: Xilinx Corporation Device 000e
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Region 0: Memory at a6000000 (64-bit, prefetchable) [size=32M]
	Region 2: Memory at a8040000 (64-bit, prefetchable) [size=256K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [60] MSI-X: Enable+ Count=32 Masked-
		Vector table: BAR=2 offset=00030000
		PBA: BAR=2 offset=00034000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 1024 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 25.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM not supported, Exit Latency L0s unlimited, L1 unlimited
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x4, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range BC, TimeoutDis+, LTR-, OBFF Not Supported
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+, EqualizationPhase1+
			 EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [1c0 v1] #19
	Capabilities: [400 v1] Access Control Services
		ACSCap:	SrcValid- TransBlk- ReqRedir- CmpltRedir- UpstreamFwd- EgressCtrl+ DirectTrans-
		ACSCtl:	SrcValid- TransBlk- ReqRedir- CmpltRedir- UpstreamFwd- EgressCtrl- DirectTrans-
	Capabilities: [410 v1] #15
	Capabilities: [480 v1] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>
	Kernel driver in use: xclmgmt
	Kernel modules: xclmgmt

02:00.1 Processing accelerators: Xilinx Corporation Device 513d
	Subsystem: Xilinx Corporation Device 000e
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 16
	Region 0: Memory at a4000000 (64-bit, prefetchable) [size=32M]
	Region 2: Memory at a8000000 (64-bit, prefetchable) [size=256K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [60] MSI-X: Enable+ Count=32 Masked-
		Vector table: BAR=2 offset=00030000
		PBA: BAR=2 offset=00034000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 1024 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 25.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM not supported, Exit Latency L0s unlimited, L1 unlimited
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x4, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range BC, TimeoutDis+, LTR-, OBFF Not Supported
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [400 v1] Access Control Services
		ACSCap:	SrcValid- TransBlk- ReqRedir- CmpltRedir- UpstreamFwd- EgressCtrl+ DirectTrans-
		ACSCtl:	SrcValid- TransBlk- ReqRedir- CmpltRedir- UpstreamFwd- EgressCtrl- DirectTrans-
	Capabilities: [410 v1] #15
	Capabilities: [480 v1] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>
	Kernel driver in use: xocl
	Kernel modules: xocl

dp018, 2021-09-07:
Each card has 2 PCie Gen 3x4 compliant devices/interfaces. Each interface has 2 functions -  user function  used by xocl and a management function used by xclmgmt. lpsci should show 4 entries, like that on  on pages 11/12 of https://www.xilinx.com/support/documentation/boards_and_kits/accelerator-cards/1_0/ug1425-alveo-u30-install.pdf

Your output shows is only showing the expected outputs for the 02:xx.x device/interface. The other device/interface is missing.

Have you told the MB bios to split the PCIe slot and look for multiple devices/interfaces? There would normally be a "Bifurcation" setting of x4x4 for a x8 slot or x4x4x4x4 for a x16 slot. The location changes between the various BIOS implementations and should be documented in the system manuals.

dp018, 2021-09-09:
Are you able to see both devices/interfaces on the card via lspci?