Xilinx/video-sdk #4: xocl build fails (U30_Ubuntu_20.04_v1.5_20210827) on ubuntu 20.04 w/ kernel 5.11

amrragab8080, 2021-09-16:
xocl build fails in ubuntu 20.04 w/ kernel 5.11 the build make log error below
```
    amazon-ebs: /var/lib/dkms/xrt/2.10.50/build/driver/xocl/userpf/xocl_bo.c: In function ‘xocl_gem_prime_mmap’:
    amazon-ebs: /var/lib/dkms/xrt/2.10.50/build/driver/xocl/userpf/xocl_bo.c:1231:39: error: ‘const struct drm_driver’ has no member named ‘gem_vm_ops’
    amazon-ebs:  1231 |   vma->vm_ops = xobj->base.dev->driver->gem_vm_ops;
    amazon-ebs:       |                                       ^~
    amazon-ebs: make[3]: *** [scripts/Makefile.build:288: /var/lib/dkms/xrt/2.10.50/build/driver/xocl/userpf/xocl_bo.o] Error 1
    amazon-ebs: make[2]: *** [Makefile:1848: /var/lib/dkms/xrt/2.10.50/build/driver/xocl/userpf] Error 2
    amazon-ebs: make[2]: Leaving directory '/usr/src/linux-headers-5.11.0-1017-aws'
    amazon-ebs: make[1]: *** [Makefile:124: all] Error 2
    amazon-ebs: make[1]: Leaving directory '/var/lib/dkms/xrt/2.10.50/build/driver/xocl/userpf'
    amazon-ebs: make: *** [Makefile:2: all] Error 2
```
kernel 5.11 has deprecated `gem_vm_ops`

amrragab8080, 2021-09-16:
linking issue in the XRT
https://github.com/Xilinx/XRT/issues/5168

koolscooby, 2022-06-09:
This is still broken in the 2.0 SDK release, and I thought I'd share the workaround for Ubuntu 20.04: roll back to kernel 5.4 LTS. You could do this on any Ubuntu 20.04.X release.

* Launch Ubuntu 20.04 from the AWS Marketplace (that has kernel 5.13 out of the box) and follow the [instructions here](https://discourse.ubuntu.com/t/how-to-downgrade-the-kernel-on-ubuntu-20-04-to-the-5-4-lts-version/26459/2) (Note: there is a bug in the last command - it should be `update-grub` not `upgrade-grub`.)
* `sudo reboot`
* Then, follow the [SDK instructions](https://xilinx.github.io/video-sdk/v2.0/getting_started_on_vt1.html#installing-the-sdk-on-an-existing-ami) to install it.
* `sudo reboot`

This Ubuntu 20.04 LTS (with kernel 5.4 LTS) w/ 2.0 SDK worked for me.

closed 2022-10-20 by NastoohX