Xilinx/video-sdk #12: [vt1] Precompiled ffmpeg generates h264 bitstream incompatible with Shaka packager

occasionallydavid, 2021-11-25:
Shaka version: https://github.com/google/shaka-packager/releases/tag/v2.6.1

Video SDK version: ami-008b1e1c9d46b4e84 (eu-west-1 Amazon Linux image, booted ~5 days ago)

```
[ec2-user@ip-172-31-8-46 ladder]$ ffmpeg -version

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
ffmpeg version 4.1.9 Copyright (c) 2000-2018 the FFmpeg developers
built with gcc 7 (GCC)
configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
libavutil      56. 22.100 / 56. 22.100
libavcodec     58. 35.100 / 58. 35.100
libavformat    58. 20.100 / 58. 20.100
libavdevice    58.  5.100 / 58.  5.100
libavfilter     7. 40.101 /  7. 40.101
libswscale      5.  3.100 /  5.  3.100
libswresample   3.  3.100 /  3.  3.100
[ec2-user@ip-172-31-8-46 ladder]$ ls -l `which ffmpeg`
-rwxr-xr-x 1 root root 255192 Aug 28 00:34 /opt/xilinx/ffmpeg/bin/ffmpeg
```

---

Given an MP4 file produced with:

```
ffmpeg -nostdin -y -c:v mpsoc_vcu_h264 -i 5sec.mp4 -filter_complex 'multiscale_xma=outputs=5:out_1_width=1920:out_1_height=1280:out_1_rate=half:out_2_width=1280:out_2_height=720:out_2_rate=half:out_3_width=640:out_3_height=360:out_3_rate=half:out_4_width=480:out_4_height=280:out_4_rate=half:out_5_width=1920:out_5_height=1280:out_5_rate=full[v1280][v720][v360][v280][vfull];[vfull]nullsink' -map '[v1280]' -b:v 3000K -profile high -level 5.2 -c:v mpsoc_vcu_h264 -f mp4 /tmp/ladder_v1280p_zsfydh9v.mp4 -map '[v720]' -b:v 1300K -profile high -level 5.2 -c:v mpsoc_vcu_h264 -f mp4 /tmp/ladder_v720p_k_dk3u2v.mp4 -map '[v360]' -b:v 600K -profile high -level 5.2 -c:v mpsoc_vcu_h264 -f mp4 /tmp/ladder_v360p_s8jgn0ku.mp4 -map '[v280]' -b:v 400K -profile high -level 5.2 -c:v mpsoc_vcu_h264 -f mp4 /tmp/ladder_v280p_yahw1yvq.mp4
```

And given the Shaka command line:

```
/home/ec2-user/bin/packager 'in=5sec.mp4.1280p.mp4,stream=video,init_segment=out/1280p.mp4,segment_template=out/1280p_$Number$.m4s' 'in=5sec.mp4.720p.mp4,stream=video,init_segment=out/720p.mp4,segment_template=out/720p_$Number$.m4s' 'in=5sec.mp4.360p.mp4,stream=video,init_segment=out/360p.mp4,segment_template=out/360p_$Number$.m4s' 'in=5sec.mp4.280p.mp4,stream=video,init_segment=out/280p.mp4,segment_template=out/280p_$Number$.m4s' --generate_static_live_mpd --mpd_output out/h264.mpd
```

Shaka packager will fail with:

````
[1125/080458:INFO:demuxer.cc(155)] Initialize Demuxer for file '5sec.mp4.720p.mp4'.
[1125/080459:WARNING:nalu_reader.cc(115)] nal_ref_idc shall not be equal to 0 for nalu type 7 (header 0x7).
[1125/080459:ERROR:avc_decoder_configuration_record.cc(52)] Failure while processing: nalu.Initialize(Nalu::kH264, nalu_data, size)
[1125/080459:ERROR:mp4_media_parser.cc(614)] Failed to parse avcc.
````

The `nal_ref_idc` of type 7 is referring to h.264 SPS NAL unit.

occasionallydavid, 2021-11-25:
Note: ffmpeg command lines above include various random changes to profile/level settings, usually both flags are unset. None of these seem related.

occasionallydavid, 2021-11-25:
Shaka operates correctly if generating .ts from ffmpeg rather than .mp4, but it would still be good to know why it is complaining about what seems to be a spec conformance test when outputting MP4

ThomasXilinx, 2021-11-29:
Hello, this is a known issue which is already fixed in our nightly builds. The fix will be available in our upcoming release.

closed 2021-11-29 by ThomasXilinx