Xilinx/video-sdk #19: [Docker] ffmpeg: command not found

android71Francis, 2022-01-05:
I would like to ask for some assistance on the issue that when running the a task definition using a docker build using the file from the SDK. 

When I run the task the docker couldn't find the ffmpeg command. I did modified it a little by adding the the sample bash file and AWS cli

```
FROM amazonlinux:2
WORKDIR /usr/src/build

RUN yum -y install libxml2-devel sudo git scl-utils

#Install XRT dependencies
RUN git clone https://github.com/Xilinx/XRT.git -b master
RUN sed -i 's/devtoolset-6/devtoolset-9/g' XRT/src/runtime_src/tools/scripts/xrtdeps.sh
RUN ./XRT/src/runtime_src/tools/scripts/xrtdeps.sh -docker

RUN yum -y install fontconfig-devel.x86_64
RUN yum -y install freetype-devel
RUN yum -y install git libxml2-devel scl-utils

RUN git clone https://github.com/gdraheim/docker-systemctl-replacement.git /usr/local/share/docker-systemctl-replacement
RUN echo "alias systemctl='python3 /usr/local/share/docker-systemctl-replacement/files/docker/systemctl3.py'" >> /root/.bashrc
RUN ln -s /usr/local/share/docker-systemctl-replacement/files/docker/systemctl3.py /usr/local/bin/systemctl
RUN export PATH=/usr/local/bin/:$PATH

# Dockerfile assumes that the video-sdk repo has been cloned locally
COPY release ./release
RUN ls -l /usr/src/build/release
RUN cd /usr/src/build/release/U30_Amazon_2.0_v1.5_20210827; ./install.sh -sw
RUN source /opt/xilinx/xcdr/setup.sh

# Build and/or install FFmpeg dependencies
RUN yum -y install libva-dev libsdl2-dev wget autoconf automake build-essential
RUN wget --no-check-certificate https://www.nasm.us/pub/nasm/releasebuilds/2.14.02/nasm-2.14.02.tar.bz2 && \
    tar xjvf nasm-2.14.02.tar.bz2 && \
    cd nasm-2.14.02 && \
    ./autogen.sh && \
    ./configure --prefix="/usr" --bindir="/usr/bin" && \
    make && \
    make install

RUN pip3 install awscli

COPY 03_ffmpeg_transcode_only.sh /usr/src/build

ENTRYPOINT bash 03_ffmpeg_transcode_only.sh
```

03_ffmpeg_transcode_only.sh 

```
aws s3 cp s3://${BUCKET_NAME}/${INPUT_VIDEO} ${INPUT_VIDEO}

OUTPUT_VIDEO_FILE="vidoe_sample.mp4"

ffmpeg -c:v mpsoc_vcu_h264 -i ${INPUT_VIDEO} \
-f mp4 -b:v 8M -c:v mpsoc_vcu_h264 -y ${OUTPUT_VIDEO_FILE}

aws s3 cp ${OUTPUT_VIDEO_FILE} s3://${BUCKET_NAME}/${SESSION}/${OUTPUT_VIDEO_FILE} 
```

The results: 

```
timestamp,message
1641398522999,Completed 256.0 KiB/33.2 MiB (1.5 MiB/s) with 1 file(s) remaining
Completed 33.2 MiB/33.2 MiB (85.9 MiB/s) with 1 file(s) remaining 
download: s3://canapii-assets-dev/sample_video_2.mp4 to ./sample_video_2.mp4
1641398523047,03_ffmpeg_transcode_only.sh: line 32: ffmpeg: command not found
1641398523376,The user-provided path vidoe_sample.mp4 does not exist.
```

ThomasXilinx, 2022-01-05:
Hello,
The error message indicates that ffmpeg is not found. Two reasons for this come to mind:
1. The Video SDK packages were not correctly installed  
2. FFmpeg is not in the PATH

Installation is handled by the following line:
RUN cd /usr/src/build/release/U30_Amazon_2.0_v1.5_20210827; ./install.sh -sw
This should install the ffmpeg executable in /opt/xilinx/ffmpeg/bin/ffmpeg
Can you check that this is done correctly?

Then the environment is set up with "source /opt/xilinx/xcdr/setup.sh". This should set the PATH and other variables.
Can you check that PATH is indeed set to include /opt/xilinx/ffmpeg/bin ?

ThomasXilinx, 2022-01-05:
Can you try adding these two lines at the start of your 03_ffmpeg_transcode_only.sh script?
```
systemctl start xrmd
source /opt/xilinx/xcdr/setup.sh
```

android71Francis, 2022-01-06:
@ThomasXilinx  Thanks ffmpeg now is working but I encountered a different issue. I log it a different ticket for other people reference too. https://github.com/Xilinx/video-sdk/issues/20

closed 2022-01-06 by android71Francis