Xilinx/video-sdk #20: [AWS ECS] No Xilinx device found

android71Francis, 2022-01-06:
I encounter an issue when running the Xilinx on a AWS ECS. My container configuration is 4CPU and 8GB memory. The purposed of this is to utilize the instance vt1.3xlarge. 

This is continuation of this ticket [#19 ](https://github.com/Xilinx/video-sdk/issues/19)

Here the results: 
```
timestamp,message
1641428030819,="---------------------------------------"
1641428030819,="-----Source Xilinx U30 setup files-----"
1641428030844,XILINX_XRT        : /opt/xilinx/xrt
1641428030844,PATH              : /opt/xilinx/xrt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
1641428030844,LD_LIBRARY_PATH   : /opt/xilinx/xrt/lib:
1641428030844,PYTHONPATH        : /opt/xilinx/xrt/python:
1641428030867,XILINX_XRM      : /opt/xilinx/xrm
1641428030867,PATH            : /opt/xilinx/xrm/bin:/opt/xilinx/xrt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
1641428030867,LD_LIBRARY_PATH : /opt/xilinx/xrm/lib:/opt/xilinx/xrt/lib:
1641428032509,No U30 devices found
1641428032541,{
1641428032541,"    ""response"": {"
1641428032541,"        ""name"": ""load"","
1641428032541,"        ""requestId"": ""1"","
1641428032541,"        ""status"": ""failed"","
1641428032541,"        ""data"": {"
1641428032541,"            ""failed"": """""
1641428032541,        }
1641428032541,    }
1641428032541,}
1641428032544,="-----Load xrm plugins-----"
1641428032576,{
1641428032576,"    ""response"": {"
1641428032576,"        ""name"": ""loadXrmPlugins"","
1641428032576,"        ""requestId"": ""1"","
1641428032576,"        ""status"": ""ok"""
1641428032576,    }
1641428032576,}
1641428032579,="---------------------------------------"
1641428033309,Completed 256.0 KiB/33.2 MiB (1.4 MiB/s) with 1 file(s) remaining
download: s3://canapii-assets-dev/sample_video_2.mp4 to ./sample_video_2.mp4
1641428033362,ffmpeg version 4.1.9 Copyright (c) 2000-2018 the FFmpeg developers
1641428033362,  built with gcc 7 (GCC)
1641428033362,  configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
1641428033362,  libavutil      56. 22.100 / 56. 22.100
1641428033362,  libavcodec     58. 35.100 / 58. 35.100
1641428033362,  libavformat    58. 20.100 / 58. 20.100
1641428033362,  libavdevice    58.  5.100 / 58.  5.100
1641428033362,  libavfilter     7. 40.101 /  7. 40.101
1641428033362,  libswscale      5.  3.100 /  5.  3.100
1641428033362,  libswresample   3.  3.100 /  3.  3.100
1641428033362,<<<<<<<==  FFmpeg xrm ===>>>>>>>>
1641428033423,"Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'sample_video_2.mp4':"
1641428033423,  Metadata:
1641428033423,    major_brand     : mp42
1641428033423,    minor_version   : 0
1641428033423,    compatible_brands: mp42mp41isomavc1
1641428033423,    creation_time   : 2019-10-30T14:20:33.000000Z
1641428033423,"  Duration: 00:00:13.08, start: 0.000000, bitrate: 21310 kb/s"
1641428033423,"    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, bt709), 3840x2160, 21306 kb/s, 29.97 fps, 29.97 tbr, 30k tbn, 60k tbc (default)"
1641428033423,    Metadata:
1641428033423,      creation_time   : 2019-10-30T14:20:33.000000Z
1641428033423,      handler_name    : L-SMASH Video Handler
1641428033423,      encoder         : AVC Coding
1641428033423,="------------------------------------------------------------"
1641428033423,   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
1641428033423,   device_id   :  0 
1641428033423,="------------------------------------------------------------"
1641428033424,XRT build version: 2.10.50
1641428033424,Build hash: 244440c52fdc373a80a17f5b54d751648f69b9ec
1641428033424,Build date: 2021-08-27 23:28:22
1641428033424,Git branch: u30
1641428033424,PID: 157
1641428033424,UID: 0
1641428033424,[Thu Jan  6 00:13:53 2022 GMT]
1641428033424,HOST: 2cbb60130ca2
1641428033424,EXE: /opt/xilinx/ffmpeg/bin/ffmpeg
1641428033424,[XMA] ERROR: ffmpeg xmahw_hal ERROR: No Xilinx device found
1641428033424,ERROR: XMA Initialization failed. Program exiting
1641428033822,upload: ./vidoe_sample.mp4 to s3://canapii-assets-dev/session_1/vidoe_sample.mp4
```

ThomasXilinx, 2022-01-06:
Did you launch the container as explained in the following instructions:
https://xilinx.github.io/video-sdk/v1.5/container_setup.html#launching-a-docker-container-on-vt1

If you do not map the devices using the docker --device option, they will not be visible from within the container.

ThomasXilinx, 2022-01-06:
Note that AWS has created a helper script to automatically generate an env var with the list of all visible devices formatted as docker --device option.
The script can be found here: https://aws.amazon.com/blogs/compute/deep-dive-on-amazon-ec2-vt1-instances/
And more details on this page: https://aws.amazon.com/blogs/compute/deep-dive-on-amazon-ec2-vt1-instances/

android71Francis, 2022-01-06:
Reading the the document it seem you need to have Xilinx on the EC2 itself to get the device. Our goal is to run multiple request throw the ECS from our Lambda by running only a task to execute the conversion. I don't know if this is possible on the ecs run task. 

 If you have reference similar with or approach or idea to accomplish.

ThomasXilinx, 2022-01-06:
Xilinx devices are only available on AWS EC2 VT1 instance. So you need to select this type of instance.
In addition, you need to install the Xilinx VIdeo SDK in your AMI. And then you will be able to work with containers.

I will reach out to AWS to see if they have experience or reference for working with VT1 and Lamdba.

NastoohX, 2022-10-26:
Hi,
As you have noted, portion of software stack needs to run on the host VT1 instance. Furthermore, given the singleton nature of some of these modules, we only support a single execution environment, which can contain multiple video pipelines. As such, if you are able to target a host VT1 or a single container within it, then there should not be any issues or imitations in executing your pipelines. However, if you require multiple isolated pipelines, then we do not support such a mode.

Cheers,

closed 2022-10-26 by NastoohX