Xilinx/video-sdk #21: [AWS VT1 EC2] ffmpeg: error while loading shared libraries: libavdevice.so.58

tkamiya0831, 2022-01-12:
Hello Xilinx team,

I have tried an AWS VT1 instance and used the accelerated ffmpeg.
It worked properly when I run it manually.
Also I have created a .sh file to make it easier to use.

Then, I set the .sh file in the crontab but it did not work properly.
I checked the cron log for this action, then I found the following error log.

`ffmpeg: error while loading shared libraries: libavdevice.so.58: cannot open shared object file: No such file or directory`

I searched the same error message then someboby said the following solution.

1) sudo find / -name libavdevice.so.58
Find the libavdevice.so.58. The location is the following.
/opt/xilinx/ffmpeg/lib/libavdevice.so.58

2) Copy the libavdevice.so.58 to /etc/ld.so.conf.d/
I have copied the /opt/xilinx/ffmpeg/lib/libavdevice.so.58 to /etc/ld.so.conf.d/

3) Read the config
sudo ldconfig

I have tried the above actions.
However, it can not run with cron.

Would you tell me, how do I use the accelerated ffmpeg with cron?

jmouroux, 2022-01-12:
Hello @tkamiya0831 .   This appears to be an environment issue.  Can you try to update the your shell script to include the environment PATH and LD_LIBRARY_PATH updates normally performed by our xcdr_setup.sh before attempting to run ffmpeg:

<pre>
source /opt/xilinx/xrt/setup.sh
export LD_LIBRARY_PATH=/opt/xilinx/ffmpeg/lib:$LD_LIBRARY_PATH
export PATH=/opt/xilinx/ffmpeg/bin:/opt/xilinx/xcdr/bin:/opt/xilinx/launcher/bin:/opt/xilinx/jobSlotReservation/bin:$PATH
</pre>

tkamiya0831, 2022-01-12:
Hello @jmouroux. Thank you for your quick reply.
I have tried to add your sentence into my shell script.
It worked fine.
Thank you for your cooperation.

closed 2022-01-12 by tkamiya0831