Xilinx/video-sdk #22: [AWS VT1 EC2] ffmpeg with sudo command not working

tkamiya0831, 2022-01-14:
Hello Xilinx team,

When I tried to use sudo command with ffmpeg, then it could not work properly.
Is there a way to run the ffmpeg with sudo?

I would like to integrate with Fsx for Windows storage service.
When I mount the storage service in VT1 instance, it was mounted as a root permission.
Therefore, ffmpeg with ec2-user can not write an output there.

jmouroux, 2022-01-14:
Hello @tkamiya0831 .  When running a command with 'sudo', you will loose your environment variable settings (i.e. PATH, LD_LIBRARY_PATH) unless you use the '-E' flag with the command.  That might be what is happening here.  Can you try running your command with that option?

tkamiya0831, 2022-01-17:
Hello @jmouroux. Thank you for your cooperation.
I have created other workaround for this case.

Anyway, which command should I type in my environment?
I have tried `sudo ffmpeg -E`.
However, it did not show any information for the environment variable setting.

jmouroux, 2022-01-18:
Hello @tkamiya0831.  The -E is an option to the sudo command, not ffmpeg.   `sudo -E <cmd>`

tkamiya0831, 2022-01-18:
Hello @jmouroux.
Thank you for your advice.
I have tried with the following command.
`sudo -E ffmpeg`
However, the result is the following.
`sudo: ffmpeg: command not found`

It seems sudo can not find the ffmpeg command.

jmouroux, 2022-01-18:
Hello @tkamiya0831. Did you prepare your env before running the command (i.e. the PATH and LD_LIBRARY_PATH that is set in xcdr_setup.sh)?

tkamiya0831, 2022-01-19:
Hello @jmouroux.
I have just run the `source /opt/xilinx/xcdr/setup.sh`.
Is this included the command?
Or should I run the "/opt/xilinx/xcdr/setup.sh"?

tkamiya0831, 2022-01-21:
Hello @jmouroux.

I have tried the following command.
`sudo -E "PATH=$PATH" -E "LD_LIBRARY_PATH=$LD_LIBRARY_PATH" /opt/xilinx/ffmpeg/bin/ffmpeg -n -i input.ts -f mp4 -b:v 5M -s 1920x1080 -g 60 -xlnx_hwdev 1 -c:v mpsoc_vcu_h264 -r 29.97 -acodec aac -b:a 256k output.mp4`

And also, I did run `sudo -E "PATH=$PATH" -E "LD_LIBRARY_PATH=$LD_LIBRARY_PATH" /home/ec2-user/transcode.sh`

However, the ffmpeg did not run.
Also, there is no error information.

NastoohX, 2022-10-21:
Hi,
Not sure if this issue has already been resolved. If not, would the following approach meet your needs?
`
:~$ sudo bash -c "source /opt/xilinx/xcdr/setup.sh && ffmpeg -version"
`
This results in:

`-----Source Xilinx U30 setup files-----`
XILINX_XRT        : /opt/xilinx/xrt
PATH              : /opt/xilinx/xrt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
LD_LIBRARY_PATH   : /opt/xilinx/xrt/lib:
PYTHONPATH        : /opt/xilinx/xrt/python:
XILINX_XRM      : /opt/xilinx/xrm
PATH            : /opt/xilinx/xrm/bin:/opt/xilinx/xrt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
LD_LIBRARY_PATH : /opt/xilinx/xrm/lib:/opt/xilinx/xrt/lib:
xrmd is already active and running 
Number of U30 devices found : 2 
{
    "response": {
        "name": "load",
        "requestId": "1",
        "status": "ok"
    }
}
`-----Load xrm plugins-----`
{
    "response": {
        "name": "loadXrmPlugins",
        "requestId": "1",
        "status": "failed",
        "data": {
            "failed": "plugin xrmU30DecPlugin (version 2) is already loaded"
        }
    }
}
`---------------------------------------`

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
No device set hence falling to default device 0
`------------------i=0------------------------------------------`

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0 
`------------------------------------------------------------`

ffmpeg version n4.4.xlnx.1 Copyright (c) 2000-2021 the FFmpeg developers
built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
libavutil      56. 70.100 / 56. 70.100
libavcodec     58.134.100 / 58.134.100
libavformat    58. 76.100 / 58. 76.100
libavdevice    58. 13.100 / 58. 13.100
libavfilter     7.110.100 /  7.110.100
libswscale      5.  9.100 /  5.  9.100
libswresample   3.  9.100 /  3.  9.100

Note that the above has invoked the correct version of FFmpeg, with proper access rights.
Cheers,

NastoohX, 2022-10-26:
Closing this issue for now. Kindly, open a new ticket if further assistance is required.
Cheers,

closed 2022-10-26 by NastoohX