Xilinx/video-sdk #23: [AWS VT1 EC2] Can not create an interlaced output

tkamiya0831, 2022-01-20:
Hello Xilinx team,

I have tried to create an interlaced MP4.
However, it could not create an proper interlaced file.

I used an interlaced media file as an input file and output via ffmpeg.
The result is the following information.
This report was created by mediainfo.

Scan type                                : Interlaced
Original scan type                       : Progressive
Scan order                               : Top Field First

And also, I have added an interlaced output `vf` for the ffmpeg.
But the result was the same.
I tried to import these files into the Adobe Premiere, then it shows frame duplocating error.

Would you tell me, how do I create proper interlaced file?
Or, mpsoc_vcu_h264 encoder can create only progressive file?

jmouroux, 2022-01-20:
Hello @tkamiya0831 .  Can you paste your full ffmpeg command?   Bear in mind that our solution does not support decode of interlaced video; interlaced video must be decoded in software and converted to progressive scan video first for further processing.

tkamiya0831, 2022-01-21:
Hello @jmouroux.
Sure, here is my command.
`ffmpeg -n -i input.ts -flags +ilme+ildct -top 1 -f mp4 -b:v 5M -s 1920x1080 -g 60 -xlnx_hwdev 1 -c:v mpsoc_vcu_h264 -r 29.97 -acodec aac -b:a 256k output.mp4`

The `-flags +ilme+ildct -top 1` will create an interlaced scan type.
However, I understood about the decoding process.
I will use your product for a progressive.

closed 2022-01-21 by tkamiya0831