Xilinx/video-sdk #24: Issue saving mpsoc_vcu_h264 on iphone cameral/roll

Mudpuppy12, 2022-02-09:
I'm having a problem with video processed with the mpsoc_vcu_h264 vs something done with the native ffmpeg libx264.

Native ffmpeg libx264 command:
ffmpeg -loglevel verbose -i V0.mp4  -i 1637340981225-convwater.png -filter_complex "[0:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=3.966666666666667:d=0.5[v0];[v0][0:a]concat=n=1:v=1:a=1[outv1][a],[1:v]scale=160:-1[outv2],[outv1][outv2]overlay=main_w-overlay_w-20:main_h-overlay_h-77.0[outv]" -map [outv] -map [a] -g 2 -preset faster -crf 18 test2.mp4 -y

Will output a video in which can be saved on an iphone devices

This command from xilinux V2 Instance w/ hardware card mpsoc_vcu_h264

ffmpeg -loglevel verbose  -nostdin -i /tmp/dennis/V0.mp4  -i /tmp/dennis/1637340981225-convwater.png -filter_complex "[0:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=10.06501133786848:d=0.5[v0];[v0][0:a]concat=n=1:v=1:a=1[outv1][a],[1:v]scale=160:-1[outv2],[outv1][outv2]overlay=main_w-overlay_w-20:main_h-overlay_h-77.0[outv]" -map [outv] -map [a] -b:v 10675372 -c:v mpsoc_vcu_h264 /tmp/dennis/test.mp4 -y

Will not.

I've ffprobed the streams, then diffed them. The first is the mpsoc_vcu_h264 the 2nd l ibx264

< color_range=tv
---
> color_range=unknown
37c37
< bit_rate=10805871
---
> bit_rate=10675372
43c43
< extradata_size=54
---
> extradata_size=45

Is there some thing with the yuv420p color_range (tv) that the cameral roll will not import?

I think I tracked it down with xcode looking at this error. I was using slack to download the video to my phone (Same issue happens with dropbox not able to save the format into camera roll)

PhotoKit changes: [A68EE991-B0EF-4232-8739-E0C8FB054F83] Reply sent to client com.tinyspeck.chatlyio(2375) at QOS_CLASS_USER_INITIATED, success: NO, error: Error Domain=PHPhotosErrorDomain Code=3302, inserts: [Asset: 1]

I'm guessing, Not sure this is right, but:
 PHPhotosErrorMultipleIdentifiersFound API_AVAILABLE(macos(12), ios(15), tvos(15)) = 3202, // More than one identifier was found

I'm at a loss on how to convert the original mov, using the mpsoc_vcu_h264 that will be compatible with iphone save into the camera roll.

Original below

ffprobe f64417a2-066a-4380-b9ea-dd761df2b2c1.mov -select_streams v:0 -show_streams
ffprobe version 4.1.9 Copyright (c) 2007-2018 the FFmpeg developers
  built with gcc 7 (GCC)
  configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'f64417a2-066a-4380-b9ea-dd761df2b2c1.mov':
  Metadata:
    major_brand     : mp42
    minor_version   : 1
    compatible_brands: isommp41mp42
    creation_time   : 2022-02-08T16:19:33.000000Z
  Duration: 00:00:10.57, start: 0.000000, bitrate: 2440 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1280x720, 2370 kb/s, 30 fps, 30 tbr, 600 tbn, 1200 tbc (default)
    Metadata:
      creation_time   : 2022-02-08T16:19:33.000000Z
      handler_name    : Core Media Video
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, mono, fltp, 64 kb/s (default)
    Metadata:
      creation_time   : 2022-02-08T16:19:33.000000Z
      handler_name    : Core Media Audio
[STREAM]
index=0
codec_name=h264
codec_long_name=H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
profile=High
codec_type=video
codec_time_base=2113/126800
codec_tag_string=avc1
codec_tag=0x31637661
width=1280
height=720
coded_width=1280
coded_height=720
has_b_frames=1
sample_aspect_ratio=N/A
display_aspect_ratio=N/A
pix_fmt=yuv420p
level=31
color_range=unknown
color_space=unknown
color_transfer=unknown
color_primaries=unknown
chroma_location=left
field_order=unknown
timecode=N/A
refs=1
is_avc=true
nal_length_size=4
id=N/A
r_frame_rate=30/1
avg_frame_rate=63400/2113
time_base=1/600
start_pts=0
start_time=0.000000
duration_ts=6339
duration=10.565000
bit_rate=2370702
max_bit_rate=N/A
bits_per_raw_sample=8
nb_frames=317
nb_read_frames=N/A
nb_read_packets=N/A
DISPOSITION:default=1
DISPOSITION:dub=0
DISPOSITION:original=0
DISPOSITION:comment=0
DISPOSITION:lyrics=0
DISPOSITION:karaoke=0
DISPOSITION:forced=0
DISPOSITION:hearing_impaired=0
DISPOSITION:visual_impaired=0
DISPOSITION:clean_effects=0
DISPOSITION:attached_pic=0
DISPOSITION:timed_thumbnails=0
TAG:creation_time=2022-02-08T16:19:33.000000Z
TAG:language=und
TAG:handler_name=Core Media Video
[/STREAM]

Mudpuppy12, 2022-02-09:
I also tried to just go back to basics:

ffmpeg -c:v mpsoc_vcu_h264 -i V0.mp4 -vf xvbm_convert -pix_fmt yuv420p -y output.yuv
ffmpeg -f rawvideo -s 1280x720 -r 30 -pix_fmt yuv420p -i wierd.yuv -b:v 768K -c:v mpsoc_vcu_h264 -f mp4 -y output.mp4

It still will not import to iphone cameral roll. (stripped out all audio)

Mudpuppy12, 2022-02-10:
This is the data I'm getting with Xcode console when trying to save an video endcoded with mpsoc_vcu264. Used dropbox as medium for transfer

PhotoKit changes: [9A53A84D-7080-4A68-B992-D814149BE2B6] Request received from client: com.getdropbox.Dropbox(3885) at QOS_CLASS_USER_INITIATED, inserts: [Asset: 1]
Unable to create asset from validated resources: Error Domain=PHPhotosErrorDomain Code=-1
PhotoKit changes: [9A53A84D-7080-4A68-B992-D814149BE2B6] Reply sent to client com.getdropbox.Dropbox(3885) at QOS_CLASS_USER_INITIATED, success: NO, error: Error Domain=PHPhotosErrorDomain Code=-1, inserts: [Asset: 1]
[ERROR : SaveMediaPhotoLibraryHelper] saved file with error: invalidVideoFormat

ThomasXilinx, 2022-02-11:
Thanks for providing all these details, including the reference clips. Our team is currently looking into this issue. We will get back to you as quickly as possible with an update.

sureshreddyavula, 2022-02-15:
I tried moving the shared video files from mac to iphone through airdrop. The hardware encoded files are failed to transfer giving error: _Failed to save item. Save to iCloud drive instead?_

Able to reproduce the issue with the commands given. The hardware encoded output file has nal_ref_idc as 0 in SPS and PPS headers at the end of the mp4 file. nal_ref_idc shouldn't be 0 for SPS or PPS headers. 

With the upcoming v2.0 of SDK, the issue is resolved and able to transfer the encoded files from mac to iphone photos app. It doesn't look like codec/hardware issue but needs to confirm. We are doubting it could be an mp4 containerization issue that could have been resolved in ffmpeg 4.4 which is the base version for v2.0 SDK. Current v1.5 SDK uses ffmpeg 4.1 as base.

sureshreddyavula, 2022-02-16:
I went through the header related changes done on top of our video-sdk v1.5 for the upcoming v2.0. We have a fixed a header metadata related issue in v2.0 that's stopping copying of mp4 files to iphone photos app(camera roll).

ThomasXilinx, 2022-06-06:
v2.0 of the Video SDK is now available and has a fix for this issue.

closed 2022-06-06 by ThomasXilinx