Xilinx/video-sdk #25: Cannot turn RTMP input into HLS segments

moerradi, 2022-05-01:
I tried to run the following command that is inspired by the streaming examples section in the xilinx video sdk docs:
https://xilinx.github.io/video-sdk/v1.5/examples/ffmpeg/tutorials.html#streaming-examples
`
#1920x1080
ffmpeg -c:v mpsoc_vcu_h264 -f flv -i rtmp://localhost/transcode/app \
-filter_complex "multiscale_xma=outputs=5: \
out_1_width=1280: out_1_height=720:  out_1_rate=full: \
out_2_width=1920: out_2_height=1080:  out_2_rate=full: \
out_3_width=848:  out_3_height=480:  out_3_rate=full: \
out_4_width=640:  out_4_height=360:  out_4_rate=full: \
out_5_width=288:  out_5_height=160:  out_5_rate=full \
[vid1][vid2][vid3][vid4][vid5]; [0:1]asplit=outputs=5[aud1][aud2][aud3][aud4][aud5]" \
-map "[vid1]" -b:v:1 5050k  -maxrate:v:1 5050k    -bufsize:v:1 5050k -c:v:1 mpsoc_vcu_h264 -map "[aud1]" -c:a aac \
-map "[vid2]" -b:v:2 6000k  -maxrate:v:2 6000k    -bufsize:v:2 6000k -c:v:2 mpsoc_vcu_h264 -map "[aud2]" -c:a aac \
-map "[vid3]" -b:v:3  750K  -maxrate:v:3 750K -bufsize:v:3 750K -c:v:3 mpsoc_vcu_h264 -map "[aud3]" -c:a aac \
-map "[vid4]" -b:v:4  375K -maxrate:v:4 375K  -bufsize:v:4 375K  -c:v:4 mpsoc_vcu_h264 -map "[aud4]" -c:a aac \
-map "[vid5]" -b:v:5 250k  -maxrate:v:5 250k -bufsize:v:5 250k -c:v:5 mpsoc_vcu_h264 -map "[aud5]" -c:a aac \
-var_stream_map "v:0,a:0 v:1,a:1 v:2,a:2 v:3,a:3 v:4,a:4" \
-f hls -master_pl_name "test.m3u8" \
-hls_segment_filename  "/hls/test-_%v-%d.ts" \
-max_muxing_queue_size 9999  "/hls/tes_%v.m3u8"
`
and it always ends up failing with the following output:
`<<<<<<<==  FFmpeg xrm ===>>>>>>>>
Input #0, flv, from 'rtmp://localhost/transcode/app':
  Metadata:
    Server          : NGINX RTMP (github.com/arut/nginx-rtmp-module)
    displayWidth    : 1920
    displayHeight   : 1080
    fps             : 60
    profile         : 
    level           : 
  Duration: 00:00:00.00, start: 151.777000, bitrate: N/A
    Stream #0:0: Audio: aac (LC), 48000 Hz, stereo, fltp, 163 kb/s
Cannot find a matching stream for unlabeled input pad 0 on filter Parsed_multiscale_xma_0`
I am running the script after sourcing the setup script located at : /opt/xilinx/xcdr/setup.sh
I am using an AWS VT1  instance with the default Xilinx Video SDK AMI for VT1 Instances.

jmouroux, 2022-05-02:
@moerradi Thank you for bringing this issue to our attention.  We will investigate further and reply with our findings.  One quick note looking over your command:  your multiscale_xma filter should be structured to generate output resolutions from highest to lowest in that order; our ABR scaler IP cascades input 1 to form the basis for scaling input 2.   As such, output 1 should be 1080p and output 2 720p (and so on) for the best quality.

jmouroux, 2022-05-03:
Hello @moerradi .  Upon review of this condition, a couple of observations to share/review:
1) It seems FFmpeg is unable to identify the video source (i.e. only the audio stream is listed).  Can you verify your RTMP source?
2) When using the multiscale_xma and mapping output to different encoder streams, it is not necessary nor recommended to use ordinals for each encoder instance (i.e. -c:v:<#>) as each scaler output is identified as video instance '0'.  Please specify simply using -c:v mpsoc_vcu_h264

cdeshpan-xlnx, 2022-05-03:
Hi @moerradi,

Just curious, could you please tell me what are we trying achieve here using -var_stream_map here?

moerradi, 2022-05-03:
I am using var stream map to map all outputs to an hls master playlist in order to acheive the auto quality mode in the video js front end i used it in normal ffmpeg without xilinx acceleration and it seemed to work fine but when I try using multiscale xma it fails

moerradi, 2022-05-03:
In order to set the master playlist properly when i don't use the -c:v[0-9] method for some wierd reason the proper bitrate for each sub playlist of the hls master playlist doesn't get set correctly

cdeshpan-xlnx, 2022-05-03:
Ok. Got it @moerradi. 

@jmouroux 
If that's the case then in order get this working we need to update multiscale_xma in order to get it working. I will need to dig more to see what changes we would need to make in the multiscale_xma  filter.

moerradi, 2022-05-03:
Also, can you please tell me if there is any way that the multiscale_xma filter detect the RTMP input resolution and only transcode to resolutions lower than input quality ?
or hard coding it using ffprobe and ffmpeg in a bash script is the only way ?

cdeshpan-xlnx, 2022-05-03:
@moerradi  If you chose .mp4 source, then does your current use case works for you?

jmouroux, 2022-05-06:
@moerradi We noticed that your video stream identifiers start enumerating from 1-5.  The first stream is actually '0' (so, 0-4 would be correct).   Can you update your encoder portion from:
```
-map "[vid1]" -b:v:1 5050k  -maxrate:v:1 5050k    -bufsize:v:1 5050k -c:v:1 mpsoc_vcu_h264 -map "[aud1]" -c:a aac \
-map "[vid2]" -b:v:2 6000k  -maxrate:v:2 6000k    -bufsize:v:2 6000k -c:v:2 mpsoc_vcu_h264 -map "[aud2]" -c:a aac \
-map "[vid3]" -b:v:3  750K  -maxrate:v:3 750K -bufsize:v:3 750K -c:v:3 mpsoc_vcu_h264 -map "[aud3]" -c:a aac \
-map "[vid4]" -b:v:4  375K -maxrate:v:4 375K  -bufsize:v:4 375K  -c:v:4 mpsoc_vcu_h264 -map "[aud4]" -c:a aac \
-map "[vid5]" -b:v:5 250k  -maxrate:v:5 250k -bufsize:v:5 250k -c:v:5 mpsoc_vcu_h264 -map "[aud5]" -c:a aac \
```
to 
```
-map "[vid1]" -b:v:0 5050k -maxrate:v:0 5050k -bufsize:v:0 5050k -c:v:0 mpsoc_vcu_h264 -map "[aud1]" -c:a aac \
-map "[vid2]" -b:v:1 6000k -maxrate:v:1 6000k -bufsize:v:1 6000k -c:v:1 mpsoc_vcu_h264 -map "[aud2]" -c:a aac \
-map "[vid3]" -b:v:2 750K -maxrate:v:2 750K -bufsize:v:2 750K -c:v:2 mpsoc_vcu_h264 -map "[aud3]" -c:a aac \
-map "[vid4]" -b:v:3 375K -maxrate:v:3 375K -bufsize:v:3 375K -c:v:3 mpsoc_vcu_h264 -map "[aud4]" -c:a aac \
-map "[vid5]" -b:v:4 250k -maxrate:v:4 250k -bufsize:v:4 250k -c:v:4 mpsoc_vcu_h264 -map "[aud5]" -c:a aac \
-var_stream_map "v:0,a:0 v:1,a:1 v:2,a:2 v:3,a:3 v:4,a:4" \
-f hls -master_pl_name "test.m3u8" \
hls_segment_filename  "test_%v-%d.ts" \
-max_muxing_queue_size 9999  "tes_%v.m3u8"
```

NastoohX, 2022-05-11:
@moerradi We were wondering if the above suggested mapping resolved your issue?

NastoohX, 2022-05-16:
Solution provided. Issue closed.

closed 2022-05-16 by NastoohX