Xilinx/video-sdk #26: ffmpeg at 29.97 fps

peykens, 2022-05-13:
Hi,

We need to have the ffmpeg encoding for fps=30000/1001.
But we can't find a way to set this properly.
We are using the multiscale_xma  :
"multiscale_xma=outputs=2: out_1_width=1920: out_1_height=1080: out_1_rate=full: out_2_width=1920: out_2_height=1080: out_2_rate=full: [vid-raw1][vid-raw2]; [vid-raw1]fps=30000/1001[vid1]; [vid-raw2]fps=30000/1001[vid2];"

Error at startup :
Error initializing output stream 1:0 -- Error while opening encoder for output stream #1:0 - maybe incorrect parameters such as bit_rate, rate, width or height

How should we set this ?

NastoohX, 2022-05-13:
Hi,

Thank you for your post. Seems that we are not able to reproduce your issue in-house. For example, the following command works fine:

ffmpeg  -f lavfi -i testsrc=duration=9999:size=1920x1080:rate=30  -filter_complex "multiscale_xma=outputs=2: out_1_width=1920: out_1_height=1080: out_1_rate=full: out_2_width=1920: out_2_height=1080: out_2_rate=full [vid-raw1][vid-raw2]; [vid-raw1]fps=30000/1001[vid1]; [vid-raw2]fps=30000/1001[vid2]" -map "[vid1]" -c:v mpsoc_vcu_h264 -b:v 8M -minrate 8M -maxrate 8M -bufsize 16M -f mpegts udp://127.0.0.1:8000 -map "[vid2]" -c:v mpsoc_vcu_h264 -b:v 1M -minrate 1M -maxrate 1M -bufsize 2M -f mpegts udp://127.0.0.1:8001

Would it be possible to provide a complete pipeline that exhibits your problem?

Cheers,

peykens, 2022-05-14:
Hi, Here the full command :

ffmpeg -hide_banner -loglevel info -analyzeduration 8M -probesize 500M -entropy_buffers_count 2 -low_latency 1 -fflags nobuffer -flags low_delay \
-i rtmp://b2cac7b6-0a62-47c5-84ce-b25d3c703dbd.service.consul:1936/live/b2cac7b6-0a62-47c5-84ce-b25d3c703dbd \
-filter_complex "multiscale_xma=outputs=8: out_1_width=1920: out_1_height=1080: out_1_rate=full: out_2_width=1920: out_2_height=1080: out_2_rate=full: out_3_width=1280: out_3_height=720: out_3_rate=full: out_4_width=1280: out_4_height=720: out_4_rate=full: out_5_width=848: out_5_height=480: out_5_rate=full: out_6_width=848: out_6_height=480: out_6_rate=full: out_7_width=640: out_7_height=360: out_7_rate=full: out_8_width=640: out_8_height=360: out_8_rate=full: [vid-raw1][vid-raw2][vid-raw3][vid-raw4][vid-raw5][vid-raw6][vid-raw7][vid-raw8]; [vid-raw1]fps=30000/1001[vid1]; [vid-raw2]fps=30000/1001[vid2]; [vid-raw3]fps=30000/1001[vid3]; [vid-raw4]fps=30000/1001[vid4]; [vid-raw5]fps=30000/1001[vid5]; [vid-raw6]fps=30000/1001[vid6]; [vid-raw7]fps=30000/1001[vid7]; [vid-raw8]fps=30000/1001[vid8]" \
-map [vid1] -c:v mpsoc_vcu_h264 -b:v 8000000 -bufsize 16000000 -control-rate 1 -pix_fmt yuv420p -coder 1 -an -bf 0 -g 1 -scaling-list 0 -f flv -profile:v main rtmp://nginx-rtmp:1936/backend/key_only_1080p \
-map [vid2] -c:v mpsoc_vcu_h264 -b:v 4000000 -bufsize 8000000 -control-rate 1 -pix_fmt yuv420p -coder 1 -an -bf 0 -g 60000 -scaling-list 0 -refs 1 -f flv -profile:v main rtmp://nginx-rtmp:1936/backend/30s_gop_1080p \
-map [vid3] -c:v mpsoc_vcu_h264 -b:v 5000000 -bufsize 10000000 -control-rate 1 -pix_fmt yuv420p -coder 1 -an -bf 0 -g 1 -scaling-list 0 -f flv -profile:v main rtmp://nginx-rtmp:1936/backend/key_only_720p \
-map [vid4] -c:v mpsoc_vcu_h264 -b:v 2500000 -bufsize 5000000 -control-rate 1 -pix_fmt yuv420p -coder 1 -an -bf 0 -g 60000 -scaling-list 0 -refs 1 -f flv -profile:v main rtmp://nginx-rtmp:1936/backend/30s_gop_720p \
-map [vid5] -c:v mpsoc_vcu_h264 -b:v 2400000 -bufsize 4800000 -control-rate 1 -pix_fmt yuv420p -coder 1 -an -bf 0 -g 1 -scaling-list 0 -f flv -profile:v main rtmp://nginx-rtmp:1936/backend/key_only_480p \
-map [vid6] -c:v mpsoc_vcu_h264 -b:v 1200000 -bufsize 2400000 -control-rate 1 -pix_fmt yuv420p -coder 1 -an -bf 0 -g 60000 -scaling-list 0 -refs 1 -f flv -profile:v main rtmp://nginx-rtmp:1936/backend/30s_gop_480p \
-map [vid7] -c:v mpsoc_vcu_h264 -b:v 1800000 -bufsize 3600000 -control-rate 1 -pix_fmt yuv420p -coder 1 -an -bf 0 -g 1 -scaling-list 0 -f flv -profile:v main rtmp://nginx-rtmp:1936/backend/key_only_360p \
-map [vid8] -c:v mpsoc_vcu_h264 -b:v 900000 -bufsize 1800000 -control-rate 1 -pix_fmt yuv420p -coder 1 -an -bf 0 -g 60000 -scaling-list 0 -refs 1 -f flv -profile:v main rtmp://nginx-rtmp:1936/backend/30s_gop_360p \
-c:a aac -vn -b:a 128000 -ar 44100 -f flv rtmp://nginx-rtmp:1936/backend/audio

Output logging on vt1 instance :

2022-05-13 02:59:26,755 DEBUG    ffmpeg        : [aac @ 0x56292836ee00] Qavg: 1990.488
2022-05-13 02:59:25,916 DEBUG    ffmpeg        : Error initializing output stream 1:0 -- Error while opening encoder for output stream #1:0 - maybe incorrect parameters such as bit_rate, rate, width or height
2022-05-13 02:59:25,916 DEBUG    ffmpeg        : [MPSOC VCU H264 encoder @ 0x562928270580] encoder error: ERROR: Unable to allocate MPSoC encoder session : ffmpeg pid 72 on device index =  0 cu index = 39
2022-05-13 02:59:25,915 DEBUG    ffmpeg        : 
2022-05-13 02:59:25,915 DEBUG    ffmpeg        : [XMA] ERROR: ffmpeg xmaencoder Initalization of encoder plugin failed. Return code -1
2022-05-13 02:59:25,875 DEBUG    ffmpeg        : 
2022-05-13 02:59:25,875 DEBUG    ffmpeg        : [XMA] ERROR: ffmpeg xma-vcu-encoder VCU_PREINIT failed : device error: 1 errors(s). Invalid settings, check the [SETTINGS] section of your configuration file or check your commandline
2022-05-13 02:59:25,874 DEBUG    ffmpeg        : EXE: /opt/xilinx/ffmpeg/bin/ffmpeg
2022-05-13 02:59:25,874 DEBUG    ffmpeg        : HOST: transcoder-859bc7dd84-c2m5s
2022-05-13 02:59:25,874 DEBUG    ffmpeg        : [Fri May 13 09:59:25 2022 GMT]
2022-05-13 02:59:25,874 DEBUG    ffmpeg        : UID: 0
2022-05-13 02:59:25,874 DEBUG    ffmpeg        : PID: 72
2022-05-13 02:59:25,874 DEBUG    ffmpeg        : Git branch: u30
2022-05-13 02:59:25,874 DEBUG    ffmpeg        : Build date: 2021-08-27 16:26:34
2022-05-13 02:59:25,874 DEBUG    ffmpeg        : Build hash: 244440c52fdc373a80a17f5b54d751648f69b9ec
2022-05-13 02:59:25,874 DEBUG    ffmpeg        : XRT build version: 2.10.50
2022-05-13 02:59:25,845 DEBUG    ffmpeg        : [mpsoc_vcu_h264 @ 0x5629282bafc0] Custom Rate Control Mode is Disabled
2022-05-13 02:59:25,845 DEBUG    ffmpeg        : encoder         : Lavc58.35.100 mpsoc_vcu_h264
2022-05-13 02:59:25,845 DEBUG    ffmpeg        : Metadata:
2022-05-13 02:59:25,845 DEBUG    ffmpeg        : Stream #0:0: Video: h264 (mpsoc_vcu_h264) ([7][0][0][0] / 0x0007), xlnx_xvbm, 1920x1080 [SAR 1:1 DAR 16:9], q=2-31, 8000 kb/s, 29.97 fps, 1k tbn, 29.97 tbc
2022-05-13 02:59:25,844 DEBUG    ffmpeg        : encoder         : Lavf58.20.100
2022-05-13 02:59:25,844 DEBUG    ffmpeg        : level           :
2022-05-13 02:59:25,844 DEBUG    ffmpeg        : profile         :
2022-05-13 02:59:25,844 DEBUG    ffmpeg        : fps             : 29
2022-05-13 02:59:25,844 DEBUG    ffmpeg        : displayHeight   : 1080
2022-05-13 02:59:25,844 DEBUG    ffmpeg        : displayWidth    : 1920
[... cut ...]
2022-05-13 02:59:23,712 DEBUG    ffmpeg        : displayWidth    : 1920
2022-05-13 02:59:23,712 DEBUG    ffmpeg        : Server          : NGINX RTMP (github.com/arut/nginx-rtmp-module)
2022-05-13 02:59:23,712 DEBUG    ffmpeg        : Metadata:
2022-05-13 02:59:23,712 DEBUG    ffmpeg        : Input #0, flv, from 'rtmp://b2cac7b6-0a62-47c5-84ce-b25d3c703dbd.service.consul:1936/live/b2cac7b6-0a62-47c5-84ce-b25d3c703dbd':
2022-05-13 02:59:23,711 DEBUG    ffmpeg        : <<<<<<<==  FFmpeg xrm ===>>>>>>>>
2022-05-13 02:59:21,232 DEBUG    ffmpeg        :

peykens, 2022-05-14:
If we just set fps=30 the ffmpeg is started, but with fps=30000/1001 it returns an error

peykens, 2022-05-16:
OK, we have found our problem. With the change of scale, we made an error in setting the gop size which is way too high.
After fixing this, the ffmpeg gets started.

closed 2022-05-16 by peykens

NastoohX, 2022-05-16:
HI Peykens,
Great to hear that. (Nice way to start the week.)
Cheers,

peykens, 2022-05-16:
@NastoohX , just fyi. When we have source at 29.97 fps and we want to transcode to 30 fps, the ffmpeg on vt1 fails continuously.
I have open xilinx support ticket for that.