Xilinx/video-sdk #28: [XMA] ERROR: ffmpeg xma-vcu-decoder profile 578 is not supported

LiorHesed, 2022-05-21:
Hello,
I'm trying to transcode a hls video chunk file on a vt1 instance using the Xilinx marketplace AMI for Amazon Linux 2.
This works:
`ffmpeg -copyts -i originalChunk.ts -y -copyts -muxdelay 0 -vsync passthrough -filter_complex "scale=1280:720" -codec:v mpsoc_vcu_h264 -b:v 1000k -codec:a copy transcodedChunk.ts`

In all the examples I saw the input file was preceeeded with `-codec:v mpsoc_vcu_h264` as well so I tried this:
`ffmpeg -c:v mpsoc_vcu_h264 -i originalChunk.ts -y -copyts -muxdelay 0 -vsync passthrough -filter_complex "scale=1280:720" -codec:v mpsoc_vcu_h264 -b:v 1000k -codec:a copy transcodedChunk.ts`
Which resulted in the error I put in the title (see also attached screenshot). Am I missing something?
[image]

ThomasXilinx, 2022-05-23:
From the error message, it looks like your input stream has a profile not currently supported by the decoder. Are you able to share your input file with us?

Your second command also has what should be an unrelated issue. You are decoding the stream using the HW accelerator, then doing scaling using a SW filter on the host, then encoding the output using the HW accelerator. 
As explained in the documentation, you need to use the xvbm_convert filter to transfer the output of the decoder to the host to run the SW filter: https://xilinx.github.io/video-sdk/v1.5/using_ffmpeg.html#moving-data-through-the-video-pipeline
Alternatively, you can use the HW accelerated scaler, in which case you do not need to transfer data back and forth:
https://xilinx.github.io/video-sdk/v1.5/using_ffmpeg.html#video-scaling

ThomasXilinx, 2022-05-23:
Upon further inspection of your log, we noticed that your input stream is "Constrained Baseline" which is not supported. The list of supported codecs, formats and resolutions can be found here:
https://xilinx.github.io/video-sdk/v1.5/specs_and_features.html#hardware-features

closed 2022-05-24 by ThomasXilinx

LiorHesed, 2022-05-25:
Great, thank you for the update !