Xilinx/video-sdk #32: 1080p output is glitchy on xffmpeg

moerradi, 2022-05-25:
Environnement:
```
Dockerfile: dockerfiles/vt1/Dockerfile.ubuntu20.04_sw
machine: AWS VT1.3xlarge
streaming software: OBS latest release
video-sdk version: 1.5
nginx_version: 1.21.6
video player: aws ivs player
ingest: 1080p60
```
I am using xffmpeg to transcode an RTMP stream into multiple HLS outputs: 1080p60 720p60 720 480  360 160
All of the above qualities work perfectly except for 1080p which is the ingest quality, I even tried to ingest using a higher quality (2K) but the problem persisted
the scripts ouptuts 1080 hls files but the output is really laggy and glitchy look at the following gif compared the the input which is this youtube video [test-vide](https://www.youtube.com/watch?v=LT-oNf3A7IU)
here is the output: [output](https://cdn.discordapp.com/attachments/858031919714795550/979006164492767252/problem.gif)
here is the script I used:
```
#!/bin/bash
source /opt/xilinx/xrt/setup.sh
source /opt/xilinx/xrm/setup.sh
export LD_LIBRARY_PATH=/opt/xilinx/ffmpeg/lib:$LD_LIBRARY_PATH
export PATH=/opt/xilinx/ffmpeg/bin:/opt/xilinx/xcdr/bin:/opt/xilinx/launcher/bin:/opt/xilinx/jobSlotReservation/bin:$PATH

#check this for refernce
#https://docs.peer5.com/guides/production-ready-hls-vod/#how-to-choose-the-right-bitrate
on_die ()
{
    # kill all children
    pkill -KILL -P $$
}

trap 'on_die' TERM

mkdir -p /hls/$1/1080p60
mkdir -p /hls/$1/720p60
mkdir -p /hls/$1/720
mkdir -p /hls/$1/480
mkdir -p /hls/$1/360
mkdir -p /hls/$1/160

 cat /scripts/exmple.m3u8 | sed "s/XXXX/$1/g" > /hls/$1/index.m3u8``

ffmpeg -i  rtmp://localhost:1935/transcode/$1 \
-filter_complex "[0:v]split=2[origin][mux]; \
[origin]multiscale_xma=outputs=6: \
out_2_width=1280: out_2_height=720:  out_2_rate=full: \
out_3_width=848:  out_3_height=480:  out_3_rate=half: \
out_4_width=640:  out_4_height=360:  out_4_rate=half: \
out_5_width=288:  out_5_height=160:  out_5_rate=half  \
[a][b][c][d]; [a]split[aa][ab]; [ab]fps=30[abb]; asplit=outputs=6[aud1][aud2][aud3][aud4][aud5][aud6]" \
-map "[mux]"  -b:v 7M    -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 10 -hls_flags delete_segments -map "[aud1]" -c:a aac -y /hls/$1/1080p60/index.m3u8  \
-map "[aa]"  -b:v 4M    -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 10 -hls_flags delete_segments -map "[aud2]" -c:a aac -y /hls/$1/720p60/index.m3u8  \
-map "[abb]" -b:v 3M    -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 10 -hls_flags delete_segments -map "[aud3]" -c:a aac -y /hls/$1/720/index.m3u8  \
-map "[b]"   -b:v 2500K -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 10 -hls_flags delete_segments -map "[aud4]" -c:a aac -y /hls/$1/480/index.m3u8  \
-map "[c]"   -b:v 1250K -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 10 -hls_flags delete_segments -map "[aud5]" -c:a aac -y /hls/$1/360/index.m3u8  \
-map "[d]"   -b:v 625K  -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 10 -hls_flags delete_segments -map "[aud6]" -c:a aac -y /hls/$1/160/index.m3u8  2>&1 | tee /hls/test.log

wait
```

NastoohX, 2022-05-25:
Thank you for bringing this issue to our attention. Given that the number of outputs from the scaler or its starting numeration do not match the expected values, i.e., 4 vs 6 and starting from 2 vs 1, we expect undefined behaviour within the pipeline, e.g., in our local tests we've noted the following family of messages: "[XMA] WARNING: ffmpeg xma-vcu-encoder process_cmd_response: b_handle 0x2539d60 already released, something went wrong here."

If you could, kindly, re-run your test with the pipeline modified as follows, and report back your observations, it would be greatly appreciated:

...

    -filter_complex "[0:v]split=2[origin][mux]; \
    [origin]multiscale_xma=outputs=4: \
    out_1_width=1280: out_1_height=720:  out_1_rate=full: \
    out_2_width=848:  out_2_height=480:  out_2_rate=half: \
    out_3_width=640:  out_3_height=360:  out_3_rate=half: \
    out_4_width=288:  out_4_height=160:  out_4_rate=half  \

...

Cheers,

moerradi, 2022-05-25:
Worked perfectly I didn't notice the number of outputs was 6, Thank you very much you made my day !

closed 2022-05-25 by moerradi