Xilinx/video-sdk #33: VT1.3xlarge instance doesn't handle promissed number of streams in ABR

moerradi, 2022-05-28:
I tied to transcode some live streams using multiscale_xma ABR, It works fine as long as I have less than 4 streams in 1080 to 480, once i Have a 5th streamers it keeps giving the following message :
```
ffmpeg version 4.1.9 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
  configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
Input #0, flv, from 'rtmp://localhost:1935/transcode/yataousi67':
  Metadata:
    Server          : NGINX RTMP (github.com/arut/nginx-rtmp-module)
    displayWidth    : 1920
    displayHeight   : 1080
    fps             : 60
    profile         :
    level           :
  Duration: 00:00:00.00, start: 76.086000, bitrate: N/A
    Stream #0:0: Audio: aac (LC), 48000 Hz, stereo, fltp, 163 kb/s
    Stream #0:1: Video: h264 (High), yuv420p(tv, bt709, progressive), 1920x1080 [SAR 1:1 DAR 16:9], 2560 kb/s, 60 fps, 60 tbr, 1k tbn, 120 tbc
------------------------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0
------------------------------------------------------------

Stream mapping:
  Stream #0:0 (aac) -> asplit
  Stream #0:1 (h264) -> split
  split:output1 -> Stream #0:0 (mpsoc_vcu_h264)
  asplit:output0 -> Stream #0:1 (aac)
  multiscale_xma:output0 -> Stream #1:0 (mpsoc_vcu_h264)
  asplit:output1 -> Stream #1:1 (aac)
  multiscale_xma:output1 -> Stream #2:0 (mpsoc_vcu_h264)
  asplit:output2 -> Stream #2:1 (aac)
  multiscale_xma:output2 -> Stream #3:0 (mpsoc_vcu_h264)
  asplit:output3 -> Stream #3:1 (aac)
  multiscale_xma:output3 -> Stream #4:0 (mpsoc_vcu_h264)
  asplit:output4 -> Stream #4:1 (aac)
Press [q] to stop, [?] for help
[mpsoc_vcu_h264 @ 0x563a6e111bc0] Custom Rate Control Mode is Disabled
xrm_allocation: failed to allocate encoder resources from device 0
[MPSOC VCU H264 encoder @ 0x563a6e177e40] xrm_allocation: resource allocation failed
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
Conversion failed!
```
here is the shell script i use to execute ABR, i am using ABR inside nginx_rtmp_module exec 
```
#!/bin/bash
source /opt/xilinx/xrt/setup.sh
source /opt/xilinx/xrm/setup.sh
export LD_LIBRARY_PATH=/opt/xilinx/ffmpeg/lib:$LD_LIBRARY_PATH
export PATH=/opt/xilinx/ffmpeg/bin:/opt/xilinx/xcdr/bin:/opt/xilinx/launcher/bin:/opt/xilinx/jobSlotReservation/bin:$PATH

#check this for refernce
#https://docs.peer5.com/guides/production-ready-hls-vod/#how-to-choose-the-right-bitrate
on_die ()
{
    # kill all children
    pkill -KILL -P $$
}

source /var/quality_data/$1.sh
while [ -v $HEIGHT -o -v $WIDTH -o -v $FPS ]
do
[... 105 lines cut ...]
		out_2_width=640:  out_2_height=360:  out_2_rate=$OUT_RATE: \
		out_3_width=288:  out_3_height=160:  out_3_rate=$OUT_RATE: \
		[a][b][c]; asplit=outputs=3[aud1][aud2][aud3]" \
		-map "[a]"	-b:v 2200K			-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud1]" -c:a aac -y /hls/$1/480/index.m3u8	\
		-map "[b]"	-b:v BITRATE_360 	-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud2]" -c:a aac -y /hls/$1/360/index.m3u8	\
		-map "[c]"	-b:v BITRATE_160 	-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud3]" -c:a aac -y /hls/$1/160/index.m3u8	 2>&1 | tee "/hls/$1.log"
elif [ $HEIGHT -ge "640" ] && [ $WIDTH -ge "360" ]
then
	mkdir -p /hls/$1/360
	mkdir -p /hls/$1/160

	export PL_BITRATE_360=900000
	envsubst < /scripts/templates/template_360.m3u8 > /hls/$1.m3u8

	ffmpeg -i  rtmp://localhost:1935/transcode/$1 \
		-filter_complex "multiscale_xma=outputs=2: \
		out_1_width=640:  out_1_height=360:  out_1_rate=full: \
		out_2_width=288:  out_2_height=160:  out_2_rate=$OUT_RATE: \
		[a][b]; asplit=outputs=2[aud1][aud2]" \
		-map "[a]"	-b:v 1200K			-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud1]" -c:a aac -y /hls/$1/360/index.m3u8	\
		-map "[a]"	-b:v BITRATE_160 	-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud2]" -c:a aac -y /hls/$1/160/index.m3u8	 2>&1 | tee "/hls/$1.log"
fi
wait
```
can you please help me out asap I have a deployment due soon please

moerradi, 2022-05-28:
I am using aws vt1.3xlarge so the minimum should be 8 1080p60 streams in ABR

NastoohX, 2022-05-30:
Hi moerradi,
Thank you for your engagement. Kindly note, as indicated in sdk's documentation, e.g., Section https://xilinx.github.io/video-sdk/v1.5/managing_compute_resources.html, each device is capable of handling 4 1080p streams and each card contains 2 devices. Therefore, 5th stream, and above, must be processed on the 2nd device. See the above link and  https://xilinx.github.io/video-sdk/v1.5/examples/ffmpeg/tutorials.html#running-multiple-ffmpeg-jobs for more details and examples.
Cheers,

moerradi, 2022-06-03:
it is the scripts that gets executed and the command is as following : 
```
ffmpeg -i  rtmp://localhost:1935/transcode/$1 \
	-filter_complex "[0:v]split=2[origin][mux]; \
	[origin]multiscale_xma=outputs=4: \
	out_1_width=1280: out_1_height=720:  out_1_rate=full: \
	out_2_width=848:  out_2_height=480:  out_2_rate=$OUT_RATE: \
	out_3_width=640:  out_3_height=360:  out_3_rate=$OUT_RATE: \
	out_4_width=288:  out_4_height=160:  out_4_rate=$OUT_RATE: \
	[a][b][c][d]; asplit=outputs=5[aud1][aud2][aud3][aud4][aud5]" \
	-map "[mux]"	-b:v $BITRATE_1080	-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud1]" -c:a aac -y /hls/$1/1080/index.m3u8	\
	-map "[a]"		-b:v $BITRATE_720	-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud2]" -c:a aac -y /hls/$1/720/index.m3u8	\
	-map "[b]"		-b:v $BITRATE_480	-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud3]" -c:a aac -y /hls/$1/480/index.m3u8	\
	-map "[c]"		-b:v $BITRATE_360 	-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud4]" -c:a aac -y /hls/$1/360/index.m3u8	\
	-map "[d]"		-b:v $BITRATE_160	-c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments -map "[aud5]" -c:a aac -y /hls/$1/160/index.m3u8	2>&1 | tee "/hls/$1.log"
```

NastoohX, 2022-06-03:
Hi Moe,
Not knowing the exact parameters, we tried a very similar command as your:

ffmpeg -f flv -i  tst.rtmp -filter_complex "[0:v]split=2[origin][mux]; \
[origin]multiscale_xma=outputs=4: \
out_1_width=1280: out_1_height=720:  out_1_rate=full:  \
out_2_width=848:  out_2_height=480:  out_2_rate=half:  \
out_3_width=640:  out_3_height=360:  out_3_rate=half:  \
out_4_width=288:  out_4_height=160:  out_4_rate=half:  \
[a][b][c][d]; asplit=outputs=5[aud1][aud2][aud3][aud4][aud5]" \
-map "[mux]"   -b:v 2M -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments \
-map "[aud1]" -c:a aac -y /tmp/hls/1080/index.m3u8 \
-map "[a]" -b:v 1M -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments \
-map "[aud2]" -c:a aac -y /tmp/hls/720/index.m3u8 \
-map "[b]" -b:v 1M -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments \
-map "[aud3]" -c:a aac -y /tmp/hls/480/index.m3u8 \
-map "[c]" -b:v 1M -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments \
-map "[aud4]" -c:a aac -y /tmp/hls/360/index.m3u8 \
-map "[d]" -b:v 1M -c:v mpsoc_vcu_h264 -f hls -hls_time 2 -hls_list_size 13 -hls_flags delete_segments \
-map "[aud5]" -c:a aac -y /tmp/hls/160/index.m3u8
, with the attached test file. On our vt13xlarge this runs fine:
<<<<<<<==  FFmpeg xrm ===>>>>>>>>
Input #0, flv, from 'tst.rtmp':
  Metadata:
    encoder         : Lavf58.29.100
  Duration: 00:00:30.02, start: 0.000000, bitrate: 1834 kb/s
    Stream #0:0: Audio: aac (LC), 44100 Hz, mono, fltp, 69 kb/s
    Stream #0:1: Video: flv1, yuv420p, 1920x1080, 200 kb/s, 30 fps, 30 tbr, 1k tbn
------------------------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0 
...
[hls @ 0x7b97c0] Opening '/tmp/hls/1080/index6.ts' for writingA time=00:00:23.86 bitrate=N/A dup=2 drop=0 speed=1.87x    
[hls @ 0x7b7980] Opening '/tmp/hls/720/index6.ts' for writing
[hls @ 0x7e03c0] Opening '/tmp/hls/480/index3.ts' for writing
[hls @ 0x827580] Opening '/tmp/hls/360/index3.ts' for writing
[hls @ 0x7bbd00] Opening '/tmp/hls/160/index3.ts' for writing
[hls @ 0x7b97c0] Opening '/tmp/hls/1<<<<<<<==  FFmpeg xrm ===>>>>>>>>
Input #0, flv, from 'tst.rtmp':
  Metadata:
    encoder         : Lavf58.29.100
  Duration: 00:00:30.02, start: 0.000000, bitrate: 1834 kb/s
    Stream #0:0: Audio: aac (LC), 44100 Hz, mono, fltp, 69 kb/s
    Stream #0:1: Video: flv1, yuv420p, 1920x1080, 200 kb/s, 30 fps, 30 tbr, 1k tbn
------------------------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0 080/index7.ts' for writingA time=00:00:27.74 bitrate=N/A dup=2 drop=0 speed=1.87x    
[hls @ 0x7b7980] Opening '/tmp/hls/720/index7.ts' for writing
frame=  901 fps= 56 q=-0.0 Lq=-0.0 q=-0.0 q=-0.0 q=-0.0 size=N/A time=00:00:30.02 bitrate=N/A dup=2 drop=0 speed=1.85x    
video:13425kB audio:1140kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
[aac @ 0x7b7000] Qavg: 61880.094
[aac @ 0x7df700] Qavg: 61880.094
[aac @ 0x826880] Qavg: 61880.094
[aac @ 0x7cafc0] Qavg: 61880.094
[aac @ 0x80c340] Qavg: 61880.094
[ec2-user@ip-172-31-10-24 tests]$ 
 
Could you kindly try this on your end and let us know if there are any issues?
Cheers,
[tst.zip](https://github.com/Xilinx/video-sdk/files/8836117/tst.zip)

moerradi, 2022-06-10:
Hello !
I figured out the trick I just had to assign a device Id to my ffmpeg command, even tho keep track of what each card is doing to assign the device Id properly is a burden it is working