Xilinx/video-sdk #34: [XMA] WARNING: ffmpeg xma-vcu-encoder device warning: !! The specified Level is too low and will be adjusted !!

LiorHesed, 2022-06-07:
Running the following command:
`ffmpeg -c:v mpsoc_vcu_h264 -i /tmp/originalChunk0.ts -loglevel info -filter_complex "multiscale_xma=outputs=1: out_1_width=1280: out_1_height=720: out_1_rate=full: [a]" -map "[a]" -b:v 1000K  -c:v mpsoc_vcu_h264 -y /tmp/originalChunk0.ts.ts` 
returns the warning in the title. This didn't happen in the previous version 1.5.
What does it mean? (see attached screenshot for more details)
[image]
)

NastoohX, 2022-06-07:
Hi,
Thank you for bringing this issue to our attention. Could you, kindly, provide the following details:
1- Deployment type ,i.e., in-prem or AWS
2- Running on a container or host
3- Distro type

Also, is it possible to provide the input file 

Dev team is looking into this; meanwhile, can you, kindly, try the following and see if you are observing the same issue:
ffmpeg -f lavfi -i testsrc=duration=40:size=1920x1080:rate=60  -c:v mpsoc_vcu_h264 -loglevel panic -filter_complex "multiscale_xma=outputs=1: out_1_width=1280: out_1_height=720: out_1_rate=full: [a]" -map "[a]" -b:v 1000K -c:v mpsoc_vcu_h264 -y /tmp/originalChunk0.ts

koolscooby, 2022-06-08:
I think that message is caused because the default `level` parameter, if unspecified, is `1` [1] and your resolution, profile, framerate, and bitrate require a higher `level`. [2] So, the encoder driver sets the level to the minimum required to support your resolution, profile, framerate, and bitrate. [3]

You could safely ignore the message or set `level` based on the characteristics of the source you're trying to encode. For example, I was encoding 1080p30 asset at 3Mbps to AVC/h.264 and used `-level 40` in the encoder settings, and the warning went away.

Refs:
[1] FFmpeg XMA plugin default for `level` parameter = "10" for 1.0 - https://github.com/Xilinx/app-ffmpeg4-xma/blob/c2817fbeb0cad3e8cb497792b8e8c52e7a0eb12f/libavcodec/mpsoc_vcu_enc.c#L144
[2] Error from `vcu-ctrl-sw` - https://github.com/Xilinx/vcu-ctrl-sw/blob/1cb5281d319ea4f3c0eb5514864c80d95e78fe6e/lib_common_enc/Settings.c#L294-L300
[3] Comment in `vcu-ctrl-sw` - https://github.com/Xilinx/vcu-ctrl-sw/blob/1cb5281d319ea4f3c0eb5514864c80d95e78fe6e/lib_common_enc/Settings.c#L1180

LiorHesed, 2022-06-08:
Thanks @koolscooby , setting `level` indeed resolved this warning !

@NastoohX :
1. I am using AWS VT1 instance with Xilinx's SDK v2.0 for AL2 image. Kernel version: 4.14.276-211.499.amzn2.x86_64
2. The warning still persists when running the command you've asked. (see attached screenshot)
[image]

NastoohX, 2022-06-09:
Great news. Thank you @koolscooby. Also, dev team has been notified of this behaviour.
Cheers,

closed 2022-06-09 by NastoohX