Xilinx/video-sdk #35: Hardware Annotation changed on AWS VT instances

natehudson, 2022-06-07:
This issue/question is about the Kubernetes Xilinx FPGA driver and VT AWS hardware. I know this is not the best place but I couldn't find a better place to ask and it is related to the SDK a bit.

We have kubernetes deployments that use the fpga plugin image `public.ecr.aws/xilinx_dcg/k8s-fpga-device-plugin:1.0.100`
and were using the recommended yaml from https://xilinx.github.io/video-sdk/v2.0/deploying_with_kubernetes.html in order to target a xilinx u30 device:
```
    resources:
      limits:
        xilinx.com/fpga-xilinx_u30_gen3x4_base_1-0: 1  # Number of Alveo U30 cards which should be mounted
```

Sometime in the last day or so, new containers stopped coming online and when we dug into it, it looks like the latest `vt1.3xlarge` instances were coming up with a different hardware capacity name, specifically:
```
Capacity:
...
  xilinx.com/fpga-xilinx_u30_gen3x4_base_2-0:  1
```

Notice the difference between `base_1-0` and `base_2-0`. Is this expected behavior when the underlying hardware revision changes or something?

I can't seem to find this breaking change announced anywhere or anyone else running into the problem. Thanks for any help or insight you can shed on this.

natehudson, 2022-06-07:
It seems that the kubernetes documentation has been updated to reflect what we are seeing with the AWS hardware. I'm still curious if these changes are expected and how/why they came about. Thanks.

NastoohX, 2022-06-07:
Hi,
Indeed, there was an update yesterday and as you have noted requires an update to pod's yaml file.
Cheers,

closed 2022-06-09 by NastoohX