Xilinx/video-sdk #37: Muxing video/Audio

LiorHesed, 2022-06-13:
Hi,
I have a hls chunk containing a video stream and an audio stream.
I'd like to transcode the video using the C API (and not ffmpeg) then mux it with the audio.
Any suggestions how to do the muxing ?

NastoohX, 2022-06-13:
Hi LiorHesed,
Our C APIs are specific to video processing only, i.e., no muxing or audio processing. (See https://xilinx.github.io/video-sdk/v2.0/c_apis.html#c-api-programming-guide) For audio and muxing we rely on open source projects, i.e., ffmpeg and gstreamer. Perhaps, you may be able to include libav, https://libav.org/, in your application for such purposes.
Cheers,

LiorHesed, 2022-06-14:
Thanks @NastoohX ,
I'm aware the C-API's are for video processing but was wondering if you know/have transcoding examples that handle containers as input instead of a raw stream as this is usually the use case

NastoohX, 2022-06-14:
Indeed, that is the case; however, as you've noted our sample codes are limited to ES. (There are a few open source projects that may be helpful, a more notable one being https://github.com/gpac/gpac, )
Cheers,

closed 2022-06-29 by NastoohX