Xilinx/video-sdk #40: [AWS VT1] Slowdown when transcoding multiple jobs on the same SoC

thzhub, 2022-07-25:
Hello Xilinx team,

I'm doing some load testing using an vt1.3xlarge ec2 instance. I like to to use as much bandwidth as possible on a single device by doing multiple transcodes (i.e. 4 1080p input videos at the same time). While testing, I noticed some slowdown while running multiple ffmpeg transcoding commands on the same device.

The test video is 1080p at 24 fps. For simplicity, I'm using the same input video in the parallel tests; the actual videos will be different for the use case.
This is the [sample video](https://studio.blender.org/films/elephants-dream/gallery/?asset=1065). 

With one job, using a very simple command
```
/usr/bin/time -f "%E real,%U user,%S sys, %P CPU" /opt/xilinx/ffmpeg/bin/ffmpeg -loglevel info -xlnx_hwdev 0 -c:v mpsoc_vcu_h264 -i /tmp/xilinx/tests/videos/big-elephants-dream-1080p.mp4 -f mp4 -c:v mpsoc_vcu_h264 -y output/out1.mp4
2:53.65 real,22.89 user,2.20 sys, 14% CPU, 0 mem_use
```
The real elapsed time is around 3 minutes.

With 3 jobs on different terminals,
```
/usr/bin/time -f "%E real,%U user,%S sys, %P CPU" /opt/xilinx/ffmpeg/bin/ffmpeg -loglevel info -xlnx_hwdev 0 -c:v mpsoc_vcu_h264 -i /tmp/xilinx/tests/videos/big-elephants-dream-1080p.mp4 -f mp4 -c:v mpsoc_vcu_h264 -y output/out1.mp4
/usr/bin/time -f "%E real,%U user,%S sys, %P CPU" /opt/xilinx/ffmpeg/bin/ffmpeg -loglevel info -xlnx_hwdev 0 -c:v mpsoc_vcu_h264 -i /tmp/xilinx/tests/videos/big-elephants-dream-1080p.mp4 -f mp4 -c:v mpsoc_vcu_h264 -y output/out2.mp4
/usr/bin/time -f "%E real,%U user,%S sys, %P CPU" /opt/xilinx/ffmpeg/bin/ffmpeg -loglevel info -xlnx_hwdev 0 -c:v mpsoc_vcu_h264 -i /tmp/xilinx/tests/videos/big-elephants-dream-1080p.mp4 -f mp4 -c:v mpsoc_vcu_h264 -y output/out3.mp4
5:44.28 real,23.45 user,2.51 sys, 7% CPU
5:44.24 real,23.53 user,2.53 sys, 7% CPU
2:56.37 real,23.38 user,2.27 sys, 14% CPU
```
The real elapsed time for one job around 3 minutes, but the other two are close to 6 minutes.

With 4 jobs on different terminals,
```
/usr/bin/time -f "%E real,%U user,%S sys, %P CPU" /opt/xilinx/ffmpeg/bin/ffmpeg -loglevel info -xlnx_hwdev 0 -c:v mpsoc_vcu_h264 -i /tmp/xilinx/tests/videos/big-elephants-dream-1080p.mp4 -f mp4 -c:v mpsoc_vcu_h264 -y output/out1.mp4
/usr/bin/time -f "%E real,%U user,%S sys, %P CPU" /opt/xilinx/ffmpeg/bin/ffmpeg -loglevel info -xlnx_hwdev 0 -c:v mpsoc_vcu_h264 -i /tmp/xilinx/tests/videos/big-elephants-dream-1080p.mp4 -f mp4 -c:v mpsoc_vcu_h264 -y output/out2.mp4
/usr/bin/time -f "%E real,%U user,%S sys, %P CPU" /opt/xilinx/ffmpeg/bin/ffmpeg -loglevel info -xlnx_hwdev 0 -c:v mpsoc_vcu_h264 -i /tmp/xilinx/tests/videos/big-elephants-dream-1080p.mp4 -f mp4 -c:v mpsoc_vcu_h264 -y output/out3.mp4
/usr/bin/time -f "%E real,%U user,%S sys, %P CPU" /opt/xilinx/ffmpeg/bin/ffmpeg -loglevel info -xlnx_hwdev 0 -c:v mpsoc_vcu_h264 -i /tmp/xilinx/tests/videos/big-elephants-dream-1080p.mp4 -f mp4 -c:v mpsoc_vcu_h264 -y output/out4.mp4
5:43.60 real,24.76 user,4.57 sys, 8% CPU
5:44.93 real,23.83 user,2.81 sys, 7% CPU
5:45.68 real,24.53 user,3.54 sys, 8% CPU
5:44.28 real,23.40 user,2.38 sys, 7% CPU
```

Is the almost 2x latency increase expected while running multiple jobs on the same device? Is there anything I can do to speed up these parallel processing?

jmouroux, 2022-07-25:
Hello @thzhub .   Based on your use case, there does not appear to be any rate emulation applied simulating 24 fps processing rate for each stream.  As such, each stream is being processed as fast as can be managed through the hardware.  The U30 is designed to support processing up to 3840x2160 (aka 4K) at 60 fps aggregate processing rate and is designed to ensure real-time processing performance at that rate and/or derivates of that aggregate pixel processing rate.  As you are not rate limiting your streams, you are processing at faster-than-real-time.   Your 4 stream use case is processing, roughly at 1080p181.   For faster-than-real-time use cases, scaling is no longer strictly linear.  I would suggest you confirm by introducing rate emulation (-re) and re-running your 4 x 1080p use case to ensure that 24 fps real-time processing rate is maintained for all 4 streams.  Hope that clarifies.

thzhub, 2022-07-26:
Thank you, that clarifies things for me.

closed 2022-07-26 by thzhub