Xilinx/video-sdk #41: aac Error handling w/ mpsoc_vcu_h264

Mudpuppy12, 2022-08-10:
Have an interesting behavior. When processing videos using libx264 we are able to get an output.mp4 (which looks fine when viewing) having two aac Invalid data found on our 6:1 and 8:1 errors)

but with VT1 ffmpeg mpsoc_vcu_h264 encoder, it will fail hard on the first error and not output a completed video.

The question is, can we duplicate the same libx264 behavior? It doesn't seem to be hard errors, unlike the vt1 mpsoc_vcu_h264 encoder which will toss it.

Thanks,
Dennis

Output below

Non VT1

ffmpeg version 4.4.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 11.2.1 (Alpine 11.2.1_git20220219) 20220219
  configuration: --enable-version3 --enable-gpl --enable-nonfree --enable-small --enable-libmp3lame --enable-libx264 --enable-libx265 --enable-libvpx --enable-libtheora --enable-libvorbis --enable-libopus --enable-libass --enable-libwebp --enable-librtmp --enable-postproc --enable-avresample --enable-libfreetype --enable-openssl --disable-debug

ffmpeg -loglevel repeat+level+error -i V0.mp4 -i V1.mp4 -i V2.mp4 -i V3.mp4 -i V4.mp4 -i V5.mp4 -i V6.mp4 -i V7.mp4 -i V8.mp4 -i V9.mp4  -i 1644596044291-whit
e.png -i 1646412774623-catching-waves-looped.m4a -filter_complex "[0:v]scale=1280:720,setdar=16/9,setsar=1/1[v0];[1:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:s
t=0:d=0.5,fade=out:st=3.5:d=0.5[v1];[2:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=19.47:d=0.5[v2];[3:v]scale=1280:720,setdar=16/9,setsar=
1/1,fade=in:st=0:d=0.5,fade=out:st=3.5:d=0.5[v3];[4:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=15.825:d=0.5[v4];[5:v]scale=1280:720,setda
r=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=3.5:d=0.5[v5];[6:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=21.1114375:d=0.5[v6];[7:v]sc
ale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=3.5:d=0.5[v7];[8:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=31.8333:d=
0.5[v8];[9:v]scale=1280:720,setdar=16/9,setsar=1/1[v9];[v0][0:a][v1][1:a][v2][2:a][v3][3:a][v4][4:a][v5][5:a][v6][6:a][v7][7:a][v8][8:a][v9][9:a]concat=n=10:v=1:a=1[
outv1][a1],[10:v]scale=-1:160[outv2],[outv1][outv2]overlay=main_w-overlay_w-20:main_h-overlay_h-20[outv];[11:a]volume=enable='between(t,0.0,7.018)+between(t,7.018,11
.018)+between(t,11.018,30.988)+between(t,30.988,34.988)+between(t,34.988,51.313)+between(t,51.313,55.313)+between(t,55.313,76.92443750000001)+between(t,76.9244375000
0001,80.92443750000001)+between(t,80.92443750000001,113.25773750000002)+between(t,113.25773750000002,120.27573750000002)':volume=1.0,afade=t=out:st=117.2757375000000
2:d=3[a2];[a1][a2]amix=inputs=2[a]" -map [outv] -map [a] -r 60 -g 2 -preset faster -crf 18 output.mp4 -y

[aac @ 0x7f16adda1b40] [error] decode_band_types: Input buffer exhausted before END element found
[error] Error while decoding stream #6:1: Invalid data found when processing input
[aac @ 0x7f16add5f4c0] [error] decode_band_types: Input buffer exhausted before END element found
[error] Error while decoding stream #8:1: Invalid data found when processing input

Errors are reported, but it will create an output.mp4

VT1

ffmpeg -version

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
No device set hence falling to default device 0
------------------i=0------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0
------------------------------------------------------------

ffmpeg version n4.4.xlnx.1 Copyright (c) 2000-2021 the FFmpeg developers
built with gcc 7 (GCC)
configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
libavutil      56. 70.100 / 56. 70.100
libavcodec     58.134.100 / 58.134.100
libavformat    58. 76.100 / 58. 76.100
libavdevice    58. 13.100 / 58. 13.100
libavfilter     7.110.100 /  7.110.100
libswscale      5.  9.100 /  5.  9.100
libswresample   3.  9.100 /  3.  9.100

ffmpeg -loglevel repeat+level+error  -i V0.mp4 -i V1.mp4 -i V2.mp4 -i V3.mp4 -i V4.mp4 -i V5.mp4 -i V6.mp4 -i V7.mp4 -i V8.mp4 -i V9.mp4  -i 1644596044291-white.png -i 1646412774623-catching-waves-looped.m4a -filter_complex "[0:v]scale=1280:720,setdar=16/9,setsar=1/1[v0];[1:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=3.5:d=0.5[v1];[2:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=19.47:d=0.5[v2];[3:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=3.5:d=0.5[v3];[4:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=15.825:d=0.5[v4];[5:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=3.5:d=0.5[v5];[6:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=21.1114375:d=0.5[v6];[7:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=3.5:d=0.5[v7];[8:v]scale=1280:720,setdar=16/9,setsar=1/1,fade=in:st=0:d=0.5,fade=out:st=31.8333:d=0.5[v8];[9:v]scale=1280:720,setdar=16/9,setsar=1/1[v9];[v0][0:a][v1][1:a][v2][2:a][v3][3:a][v4][4:a][v5][5:a][v6][6:a][v7][7:a][v8][8:a][v9][9:a]concat=n=10:v=1:a=1[outv1][a1],[10:v]scale=-1:160[outv2],[outv1][outv2]overlay=main_w-overlay_w-20:main_h-overlay_h-20[outv];[11:a]volume=enable='between(t,0.0,7.018)+between(t,7.018,11.018)+between(t,11.018,30.988)+between(t,30.988,34.988)+between(t,34.988,51.313)+between(t,51.313,55.313)+between(t,55.313,76.92443750000001)+between(t,76.92443750000001,80.92443750000001)+between(t,80.92443750000001,113.25773750000002)+between(t,113.25773750000002,120.27573750000002)':volume=1.0,afade=t=out:st=117.27573750000002:d=3[a2];[a1][a2]amix=inputs=2[a]" -map [outv] -map [a] -r 60 -g 120 -b:v 8000k -level 5.2 -c:v mpsoc_vcu_h264 output.mp4 -y

[aac @ 0x1ff8000] [error] decode_band_types: Input buffer exhausted before END element found
[error] Error while decoding stream #6:1: Invalid data found when processing input
XRT build version: 2.11.691
Build hash: 3e695ed86d15164e36267fb83def6ff2aaecd758
Build date: 2021-11-19 02:16:32
Git branch: 2021.1
PID: 10629
UID: 1000
[Wed Aug 10 18:51:36 2022 GMT]
HOST: ip-10-0-0-55.ec2.internal
EXE: /opt/xilinx/ffmpeg/bin/ffmpeg
[XMA-Session-Stats] WARNING: Session id: 1, type: encoder still has unused completd cu cmds

NastoohX, 2022-08-11:
Hi,
Thank you for bringing this issue to our attention. If you can provide feedback to the following questions, it will help us with debugging this problem:
1- Does the problem persists if audio inputs and related processing, from filter_complex, were removed. (Trying to establish if problem happens during encoding or muxing.)
2- Is this problem reproducible with smaller number of input? Specifically, can it be reproduced using only the offending audio and video streams? If so, would you be able to share these sample clips?
3- Noticed that gop sizes on the 2 sample cli are different: 2 and 120. Do you notice a similar/different behaviour when -g in both cases are the same?
Looking forward to your reply.
Cheers,

Mudpuppy12, 2022-08-12:
I've tried -g both 2, 120. I notice similar behavior.

Mudpuppy12, 2022-08-12:
Videos are @ https://www.dropbox.com/sh/19vasquztuwsndb/AACYYVUV9pQbUDLc5m7j80FVa?dl=0

NastoohX, 2022-08-18:
Hi,
Thank you for the sample clips. We have indeed been able to verify your observation and are working towards a solution. I'll update this issue, as soon as there is a viable solution available.
Cheers,

Mudpuppy12, 2022-09-21:
Hello, Any update on this issue?

NastoohX, 2022-09-21:
At this point, we believe this to be an issue with our base 4.4 FFmpeg. We'll test this issue again, once we have updated our base FFmpeg.
Cheers,

Mudpuppy12, 2022-09-22:
What does that timescale look for the updated FFmpeg?

NastoohX, 2022-09-23:
Hi,
We are still working on this and will update this ticket as soon as possible.

NastoohX, 2022-09-28:
Hi,
We have a fix for this issue, which will become official in our next release. However, for now, you can apply the attached patch file on top of the procedure noted in https://xilinx.github.io/video-sdk/v2.0/using_ffmpeg.html#using-the-git-patch-file, i.e. after going through steps 1-5, do another patch: **git am a1531598439ff21e2fc00f1fbc10c6387f47624f.patch --ignore-whitespace --ignore-space-change**. Once this attached patch is applied, proceed with steps 6-8. 
With the above procedure, I have tested your pipeline and have obtained a complete video clip:
$ ffprobe -hide_banner -loglevel info output.mp4
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'output.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso6iso2avc1mp41
    encoder         : Lavf58.76.100
  Duration: 00:02:00.32, start: 0.000000, bitrate: 6741 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv), 1280x720 [SAR 1:1 DAR 16:9], 6608 kb/s, 60 fps, 60 tbr, 15360 tbn, 120 tbc (default)

    Metadata:
      handler_name    : VideoHandler
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 128 kb/s (default)
    Metadata:
      handler_name    : SoundHandler

Let us know if this resolves your issue.
Cheers,

[a1531598439ff21e2fc00f1fbc10c6387f47624f.patch.zip](https://github.com/Xilinx/video-sdk/files/9669415/a1531598439ff21e2fc00f1fbc10c6387f47624f.patch.zip)

Mudpuppy12, 2022-11-14:
What is the ETA on the next release? It is easier for us to wait at this time.

NastoohX, 2022-12-07:
Hi,
Closing this issue given the offline conversion.
Cheers,