Xilinx/video-sdk #42: xbmgmt ERROR after flash update

shaunclimbon, 2022-08-25:
Installing SDK and flashing cards on a supported env (Ubuntu 18.04 w/ 5.4 kernel).  Install script ran fine and card update seemed fine as well but after a cold re-boot xbmgmt gives the following strange error (indicating a Windows path):

[xbmgmt] ERROR: boost::filesystem::status: Permission denied: "C:\Xilinx"

Full log is below.  Note that card passes validation despite this error.

$ sudo /opt/xilinx/xrt/bin/xball --device-filter u30 xbmgmt program --base
Discovering installed devices....
Skip: xilinx_u250_gen3x16_base_1
Skip: xilinx_u200_xdma_201830_2
Match: xilinx_u30_gen3x4_base_2
Match: xilinx_u30_gen3x4_base_2

=====================================================================
1 / 2 [0000:8a:00.0] : xilinx_u30_gen3x4_base_2
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Command: /opt/xilinx/xrt/bin/xbmgmt program --base --device 0000:8a:00.0

XRT build version: 2.11.691
Build hash: 3e695ed86d15164e36267fb83def6ff2aaecd758
Build date: 2021-11-18 18:16:39
Git branch: 2021.1
PID: 3680
UID: 0
[Wed Aug 24 22:11:20 2022 GMT]
HOST: xowalveoaccel40
EXE: /opt/xilinx/xrt/bin/unwrapped/xbmgmt2
[xbmgmt] ERROR: boost::filesystem::status: Permission denied: "C:\Xilinx"

Command Return Value: 256 [Error(s) occured]

=====================================================================
2 / 2 [0000:89:00.0] : xilinx_u30_gen3x4_base_2
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Command: /opt/xilinx/xrt/bin/xbmgmt program --base --device 0000:89:00.0

XRT build version: 2.11.691
Build hash: 3e695ed86d15164e36267fb83def6ff2aaecd758
Build date: 2021-11-18 18:16:39
Git branch: 2021.1
PID: 3707
UID: 0
[Wed Aug 24 22:11:21 2022 GMT]
HOST: xowalveoaccel40
EXE: /opt/xilinx/xrt/bin/unwrapped/xbmgmt2
[xbmgmt] ERROR: boost::filesystem::status: Permission denied: "C:\Xilinx"

Command Return Value: 256 [Error(s) occured]

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Summary:
   Installed device(s) : 4
          Shell Filter : 'u30'
      Number Evaluated : 2
                Passed : 0
                Failed : 2
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

ThomasXilinx, 2022-08-26:
This is a known issue which has already been fixed in a newer version of XRT: https://github.com/Xilinx/XRT/commit/aa67c81332cd755551d527452ae63c2af80b67ab
One possibility to move past the issue is to install a newer version of XRT in order to flash the cards, and then revert to the version of XRT included in the Video SDK to run your workloads.

closed 2022-08-26 by ThomasXilinx

shaunclimbon, 2022-08-30:
Thanks for the feedback.  I was also able to work-around it by using the Docker image.