Xilinx/video-sdk #43: Hardware accelerated decoding fails on M1 macs for certain videos encoded with h264

Mudpuppy12, 2022-08-29:
Playback on an M1 mac says video file corruption on a browser playback.

We have tracked down the issue which is similar below

https://trac.ffmpeg.org/ticket/9713
https://git.videolan.org/?p=ffmpeg.git;a=commitdiff;h=c534d9f72a89542ed639071b1ae15893aadf1f18

This bug also seems to be affecting the xlinux codec as well.

NastoohX, 2022-08-30:
Hi,
Thank you for bringing this issue and its solution to our attention! I'll inform our dev team of your findings.
Thanks again.
Cheers,

closed 2022-08-30 by NastoohX

Mudpuppy12, 2022-09-21:
Why was this issue closed? We are still seeing problems with this.

NastoohX, 2022-09-21:
Hi,
Thank you for the followup. As the problem is currently stated, it is either specific to M1 or version 4.4 of FFmpeg which our release is based on. We can't do much for the 1st case; however, we will revisit this problem, when we have updated our base FFmpeg.
Cheers,