Xilinx/video-sdk #44: Hardware accelerated decoding fails on M1 macs for certain videos encoded with h264 (still issue)

Mudpuppy12, 2022-09-21:
Reopening this issue, as it wasn't addressed.
https://github.com/Xilinx/video-sdk/issues/43

Playback on an M1 mac says video file corruption on a browser playback.
We have tracked down the issue which is similar below

https://trac.ffmpeg.org/ticket/9713
https://git.videolan.org/?p=ffmpeg.git;a=commitdiff;h=c534d9f72a89542ed639071b1ae15893aadf1f18

This bug also seems to be affecting the xlinux codec as well.

Please advise

NastoohX, 2022-09-23:
Not sure if this will help to resolve your issue, but give the attached patch a try and if possible let us know if it helped. Kindly, follow steps delineated in https://xilinx.github.io/video-sdk/v2.0/using_ffmpeg.html#using-the-git-patch-file, noting to replace 0001-Support-for-Xilinx-U30-SDK-v2-for-FFmpeg-n4.4.patch with the attached patch and switching to FFmpeg's n4.4.2 branch.
[patch.zip](https://github.com/Xilinx/video-sdk/files/9637487/patch.zip)

Cheers,

NastoohX, 2022-12-07:
Hi,
As there has not been any further feedback, it issue is now closed.
Cheers,

closed 2022-12-07 by NastoohX