Xilinx/video-sdk #45: u30 failed to load xclbins

MarcusWay45, 2022-10-05:
{
    "response": {
        "name": "load",
        "requestId": "1",
        "status": "failed",
        "data": {
            "failed": "xclLoadXclBin failed, rc = -22, failed to load /opt/xilinx/xcdr/xclbins/transcode.xclbin to device 0"
        }
    }
}
-----Load xrm plugins-----
{
    "response": {
        "name": "loadXrmPlugins",
        "requestId": "1",
        "status": "failed",
        "data": {
            "failed": "plugin xrmU30DecPlugin (version 2) is already loaded"
        }
    }
}

NastoohX, 2022-10-05:
Hi MarcusWay45,
Thank you for bringing this matter to our attention. For debugging purposes, could you, kindly, provide the following information:

1- Which version of video sdk is this instance running?
2- Is this an AWS instance?
3- What distro and kernel version is this instance running on, e.g., output of the following commands?

  `lsb_release`
  `uname -ar`

4- What is the output of the following commands?
  `source /opt/xilinx/xrt/setup.sh`
 `xbutil examine`

5- What is the full output of `source /opt/xilinx/xcdr/setup.sh` command?

Cheers,

MarcusWay45, 2022-10-06:
> Hi MarcusWay45, Thank you for bringing this matter to our attention. For debugging purposes, could you, kindly, provide the following information:
>
> 1- Which version of video sdk is this instance [...]

Thank you for your kindly response I'm sorry about not to mention about OS or Hardware 
1) I'm running Video-SDK 2.0 
2) I'm using 3 Cards of Alveo U30 with already set up and config following by the manual 
3)OS `Linux bva-Super-Server 4.15.0-193-generic #204-Ubuntu SMP Fri Aug 26 19:20:21 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

NastoohX, 2022-10-06:
Thank you for the info. The problem is due to incompatible version of XRT, i.e, 2.12.427, wrt video sdk 2.0, which requires version 2.11.691. (Kindly, note that we do not support mixed card environment.) To properly enable video sdk 2.0, you need to:
1- `sudo apt purge xrt`
2- Rerun install script, as per https://xilinx.github.io/video-sdk/v2.0/getting_started_on_prem.html#install-the-sdk
3- Reboot the chassis
Let us know if this resolves your issue.
Cheers,

MarcusWay45, 2022-10-07:
> Thank you for the info. The problem is due to incompatible version of XRT, i.e, 2.12.427, wrt video sdk 2.0, which requires version 2.11.691. (Kindly, note that we do not support mixed card [...]

Thank you,Will follow your instructions 
Peace

MarcusWay45, 2022-10-20:
Want to update about your instructions
 I Following  in this link https://xilinx.github.io/video-sdk/v2.0/getting_started_on_prem.html#install-the-sdk 
After Cold Boot Step My Machine ran into terminal mode when it boot you know how to fix it ?

MarcusWay45, 2022-10-25:
Could you explain me , How to deal with mixed card environment that you said

NastoohX, 2022-10-26:
Hi,
It very much depends on whether the other card(s) within your chassis is able to work with XRT version 2.11.691 or not. If so, we do not foresee any problems. (You will need to check the specific requirements wrt the other card(s).) If this is not the case and other card(s) do not need to directly interact with U30, then an alternative _could_ be to run multiple VMs on the host, where each card is mapped to a single VM, with the correct version of XRT.  It should be noted that we have not tested this option.
Cheers,

closed 2022-12-01 by NastoohX