Xilinx/video-sdk #50: ffmpeg xmahw_hal Could not get info for xclbin file

krlc, 2022-12-06:
I encountered an issue when running the ffmpeg on my Ubuntu machine. When I run any ffmpeg command, it results with: `[XMA] ERROR: ffmpeg xmahw_hal Could not get info for xclbin file /opt/xilinx/xcdr/xclbins/transcode.xclbin`. I'm not sure why.

I have a single Alveo U30. Firmware updated with `xball` according to the manual. `xbutil validate` passed for two devices.

There's a catch, though. The system I have is Ubuntu 22.04 with a custom compiled XRT (branch 2022.2) and XRM (master). I know the newer kernels aren't officially supported yet, but unfortunately downgrade is not an option for me. XRT and video-sdk installed successfully. Seems like it's an XRT issue. Is 2022.2 incompatible with current video-sdk? Should I use the u30 branch of XRT? Or is there any tested newer XRT versions that works with the latest video-sdk? If not yet, when will you officially support Ubuntu >20?

Thank you.

Full output of ffmpeg:
```
ffmpeg version n4.4.xlnx.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
  configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
No device set hence falling to default device 0
------------------i=0------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0 
------------------------------------------------------------

XRT build version: 2.14.0
Build hash: 50bd779580cf109b75d89b191d836a6a8f06156c
Build date: 2022-12-06 20:10:10
Git branch: 2022.2
PID: 32583
UID: 1000
[Tue Dec  6 21:34:07 2022 GMT]
HOST: test
EXE: /opt/xilinx/ffmpeg/bin/ffmpeg
[XMA] ERROR: ffmpeg xmahw_hal Could not get info for xclbin file /opt/xilinx/xcdr/xclbins/transcode.xclbin

ERROR: XMA Initialization failed. Program exiting
```

Output of xbutil examine:
```
System Configuration
  OS Name              : Linux
  Release              : 5.15.0-56-generic
  Version              : #62-Ubuntu SMP Tue Nov 22 19:54:14 UTC 2022
  Machine              : x86_64
  CPU Cores            : 128
  Memory               : 193262 MB
  Distribution         : Ubuntu 22.04.1 LTS
  GLIBC                : 2.35
  Model                : S8030GM2NE

XRT
  Version              : 2.14.0
  Branch               : 2022.2
  Hash                 : 50bd779580cf109b75d89b191d836a6a8f06156c
  Hash Date            : 2022-12-06 20:10:10
  XOCL                 : 2.14.0, 50bd779580cf109b75d89b191d836a6a8f06156c
  XCLMGMT              : 2.14.0, 50bd779580cf109b75d89b191d836a6a8f06156c

Devices present
BDF             :  Shell                     Platform UUID                         Device ID       Device Ready*  
------------------------------------------------------------------------------------------------------------------
[0000:22:00.1]  :  xilinx_u30_gen3x4_base_2  7F2A9619-6D07-9610-DEF6-0B81EC18DBAA  user(inst=129)  Yes            
[0000:21:00.1]  :  xilinx_u30_gen3x4_base_2  7F2A9619-6D07-9610-DEF6-0B81EC18DBAA  user(inst=128)  Yes            

* Devices that are not ready will have reduced functionality when using XRT tools
```

Output of `sudo /opt/xilinx/xrt/bin/xball --device-filter u30 xbmgmt program --base`:
```
Discovering installed devices....
Match: xilinx_u30_gen3x4_base_2
Match: xilinx_u30_gen3x4_base_2

=====================================================================
1 / 2 [0000:22:00.0] : xilinx_u30_gen3x4_base_2
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Command: /opt/xilinx/xrt/bin/xbmgmt --device 0000:22:00.0 program --base

Device is up-to-date.  No flashing to performed.

Command Return Value: 0 [Operation successful]

=====================================================================
2 / 2 [0000:21:00.0] : xilinx_u30_gen3x4_base_2
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Command: /opt/xilinx/xrt/bin/xbmgmt --device 0000:21:00.0 program --base

Device is up-to-date.  No flashing to performed.

Command Return Value: 0 [Operation successful]

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Summary:
   Installed device(s) : 2
          Shell Filter : 'u30'
      Number Evaluated : 2
                Passed : 2
                Failed : 0
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

kasivisw-amd, 2022-12-07:
I too faced the same issue as below

ubuntu@video-sdk3-0:~/video-sdk/examples/ffmpeg/tutorials$ ./01_ffmpeg_decode_only.sh TextInMotion-VideoSample-1080p.mp4
ffmpeg version n4.4.xlnx.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.5.0-3ubuntu1~18.04)
  configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
No device set hence falling to default device 0
------------------i=0------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0
------------------------------------------------------------

XRT build version: 2.13.466
Build hash: f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
Build date: 2022-04-14 17:45:07
Git branch: 2022.1
PID: 6113
UID: 1000
[Wed Dec  7 09:49:14 2022 GMT]
HOST: video-sdk3-0
EXE: /opt/xilinx/ffmpeg/bin/ffmpeg
[XMA] ERROR: ffmpeg xmahw_hal Could not get info for xclbin file /opt/xilinx/xcdr/xclbins/transcode.xclbin

ERROR: XMA Initialization failed. Program exiting

ubuntu@video-sdk3-0:~$ xbutil examine
System Configuration
  OS Name              : Linux
  Release              : 5.4.0-050400-generic
  Version              : #201911242031 SMP Mon Nov 25 01:35:10 UTC 2019
  Machine              : x86_64
  CPU Cores            : 16
  Memory               : 30099 MB
  Distribution         : Ubuntu 18.04.6 LTS
  GLIBC                : 2.27
  Model                : OpenStack Compute

XRT
  Version              : 2.13.466
  Branch               : 2022.1
  Hash                 : f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
  Hash Date            : 2022-04-14 17:45:07
  XOCL                 : 2.13.466, f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
  XCLMGMT              : 2.13.466, f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776

Devices present
BDF             :  Shell                     Platform UUID                         Device ID       Device Ready*
[0000:0c:00.1]  :  xilinx_u30_gen3x4_base_2  7F2A9619-6D07-9610-DEF6-0B81EC18DBAA  user(inst=129)  Yes
[0000:0b:00.1]  :  xilinx_u30_gen3x4_base_2  7F2A9619-6D07-9610-DEF6-0B81EC18DBAA  user(inst=130)  Yes

* Devices that are not ready will have reduced functionality when using XRT tools

NastoohX, 2022-12-07:
Hi,
Thank you for bringing this issue to our attention. It is indeed the case that different versions of video-sdk are meant to work with specific versions of XRT and XRM. Although, Ubuntu 2022 is not a supported OS within video-sdk v2.0; however, the following versions of XRT and XRM may provide a better outcome:
XRT: https://github.com/Xilinx/xrt/tree/22cbc46ebc878a0548f6329922807031e2c5d50f
XRM: https://github.com/Xilinx/xrm/tree/ff7479b8094515396dc2b51f433c89a626e1202e

Alternatively, you may contact AMD/Xilinx, at Video-SDK@amd.com, for Ubuntu 2022 support within video-sdk.

Cheers,

krlc, 2022-12-07:
It worked!!! I recompiled those commits, and now ffmpeg transcodes just fine! On Ubuntu 22! **I cannot thank you enough!** Huh interesting, what was the cause for this issue? I'll go through the changes later to find out.

Also, I noticed color difference and some weird visual artifacts on one frame appearing when doing h264 to h265 transcoding, compared to libx265. I haven't investigated it any further yet, but maybe you have any idea what's causing it? [https://imgur.com/a/sqhVJyS](https://imgur.com/a/sqhVJyS)

NastoohX, 2022-12-07:
Hi,
That's great. Those specific hashes were tested on 22.04, for U30, under somewhat of a different operating environment. (In general, there are no guarantees that tip of XRT or XRM repos will work with U30.)
As for the blocking issues, Kindly, open a new ticket. In that ticket please include:
1- Is this a repeatable scenario in Ubuntu 22.04
2- Is it also observed in Ubuntu 20.04 or any other supported OS with standard video-sdk installation
3- Does changing encode bit rate has any effect on the blocking artefacts
4- Short input and output clips that exhibit this problem
Cheers,

closed 2022-12-07 by NastoohX

krlc, 2022-12-08:
@kasivisw-amd  Did you compile XRT by yourself? You seem to have a stock Ubuntu 18 with kernel 5.4, which is officially supported, but XRT you used it from branch 2022.1? Have you tried using the one that comes with video-sdk?