Xilinx/video-sdk #54: retain inputs EIA-608/EIA-708 and ID3 timed text data using mpsoc_vcu_h264

scottf-tvw, 2023-01-02:
as referenced in issue #53, how would you retain and pass (map) both EIA-608/EIA-708 and ID3 data using mpsoc_vcu_h264 decoder/scaler/encoder if it exists in the source input stream? (using ffmpeg)
I have tried many options and failed to retain this data in the hls segments.

NastoohX, 2023-01-05:
Hi,
As noted in https://github.com/Xilinx/video-sdk/issues/53, we do not retain those SEI elements during transcode operation.  Depending on your flow, you may be able to extract the existing CC and reinsert them later on, e.g.:
Extract
ffmpeg -f lavfi -i "movie=cc.mp4[out0+subcc]" -map s cc.srt
, reinsert
ffmpeg  -f lavfi -i testsrc=duration=20:size=1920x1080:rate=30 -i cc.srt  -c:v libx264 -c:s mov_text -metadata:s:s:0 language=eng mix.mp4
and confirm
ffprobe -v error -show_entries stream=index,codec_name,codec_type mix.mp4 
[STREAM]
index=0
codec_name=h264
codec_type=video
[/STREAM]
[STREAM]
index=1
codec_name=mov_text
codec_type=subtitle
[/STREAM]

Hope this helps.
Cheers,

closed 2023-01-12 by NastoohX

NastoohX, 2023-01-30:
Hi,
For roadmap related questions, kindly, contact Video-SDK@amd.com.
Cheers,