Xilinx/video-sdk #59: incorrect references about lookahead not supported at 4k

shaunclimbon, 2023-01-20:
2.0 release added support for lookahead at 4k but multiple conflicting references still exist in the documentation:

- https://xilinx.github.io/video-sdk/v2.0/examples/ffmpeg/tutorials.html?highlight=lookahead#encoding-streams-to-4k
  - "The lookahead feature is not supported for 4k. FFmpeg will give an error if [-lookahead_depth](https://xilinx.github.io/video-sdk/v2.0/using_ffmpeg.html#cmdoption-lookahead_depth) is enabled when encoding to 4k."
-  https://xilinx.github.io/video-sdk/v2.0/specs_and_features.html#video-codec-unit
  - "Look-ahead driven video quality improvements through temporal adaptive and spatial adaptive quantization for resolutions up to 1080p"

NastoohX, 2023-01-24:
Hi,
Thank you for bringing this to our attention. Appropriate corrections will be made ASAP.
Cheers,

ThomasXilinx, 2023-02-02:
The documentation has now been updated accordingly.

closed 2023-02-02 by ThomasXilinx