Xilinx/video-sdk #61: NOT working with FFMPEG

notego, 2023-02-27:
Hi,

U30 device environment is successfully installed, but it does not work with sample code like below. there is no output file after the execution. Please feedback me with what is wrong.**

user@user-System-Product-Name:~/u30/video-sdk/examples/ffmpeg/tutorials$ ./01_ffmpeg_decode_only.sh Big_Buck_Bunny_360_10s_1MB.mp4 
ffmpeg version n4.4.xlnx.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
  configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
No device set hence falling to default device 0
------------------i=0------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0 
------------------------------------------------------------

Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'Big_Buck_Bunny_360_10s_1MB.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    title           : Big Buck Bunny, Sunflower version
    artist          : Blender Foundation 2008, Janus Bager Kristensen 2013
    composer        : Sacha Goedegebure
    encoder         : Lavf57.63.100
    comment         : Creative Commons Attribution 3.0 - http://bbb3d.renderfarming.net
    genre           : Animation
  Duration: 00:00:10.00, start: 0.000000, bitrate: 792 kb/s
  Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 640x360 [SAR 1:1 DAR 16:9], 789 kb/s, 30 fps, 30 tbr, 15360 tbn, 60 tbc (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
[MPSOC H.264 decoder @ 0x557a14074000] timing information from stream is not available
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (mpsoc_vcu_h264) -> rawvideo (native))
Press [q] to stop, [?] for help
[Parsed_xvbm_convert_0 @ 0x557a14964000] xvbm_conv:: wait for conversion to finish...
Output #0, rawvideo, to '/tmp/xil_dec_out.yuv':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    title           : Big Buck Bunny, Sunflower version
    artist          : Blender Foundation 2008, Janus Bager Kristensen 2013
    composer        : Sacha Goedegebure
    genre           : Animation
    comment         : Creative Commons Attribution 3.0 - http://bbb3d.renderfarming.net
    encoder         : Lavf58.76.100
  Stream #0:0(und): Video: rawvideo (I420 / 0x30323449), yuv420p, 640x360 [SAR 1:1 DAR 16:9], q=2-31, 82944 kb/s, 30 fps, 30 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
      encoder         : Lavc58.134.100 rawvideo
frame=  300 fps=0.0 q=-0.0 Lsize=  101250kB time=00:00:10.00 bitrate=82944.0kbits/s speed=23.7x    
video:101250kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%

NastoohX, 2023-02-27:
Hi,
Looking at the logs, it seems that proper yuv file was created, i.e., _Output #0, rawvideo, to '/tmp/xil_dec_out.yuv':_. Note that output is generated under /tmp folder. Can you, kindly, confirm?
Cheers,

notego, 2023-03-02:
Hi,
I cannot find the output file and now ffmpeg does not work as below.
./01_ffmpeg_decode_only.sh: line 30: ffmpeg: command not found

I reinstalled video-sdk, but it is not fixed. How do I reinstall the Xilinx ffmpeg?

Thanks.

NastoohX, 2023-03-02:
Excellent.
Cheers,

closed 2023-03-02 by NastoohX