Xilinx/video-sdk #63: Unable to open MPEG-TS livestream

koenkarsten, 2023-03-03:
After launching an AWS `vt1.3xlarge` instance with the `AMD-Xilinx Video SDK AMI for VT1 Instances (Ubuntu 20.04 LTS)` AMI image provided by Xilinx we're trying to ingest an MPEG-TS stream, and simply output it to a file with the provided `ffmpeg` binary with the following flags: `ffmpeg -c:v mpsoc_vcu_h264 -i 'http://<IP>/mpegts' -y transcoded.mp4`

Sadly this simple example gives the follwing error:
```
Impossible to convert between the formats supported by the filter 'Parsed_null_0' and the filter 'auto_scaler_0'
Error reinitializing filters!
Failed to inject frame into filter network: Function not implemented
Error while processing the decoded data for stream #0:1
Conversion failed!
```

Especially the `Failed to inject frame into filter network: Function not implemented` indicates network support is not implemented? We do notice every example in the documentation refers to transcoding files, is it true there is no network support or are we missing something. We're considering `mpsoc_vcu_h264` for livestream transcoding purposes at scale, and we're expecting this to be handled as with any other `ffmpeg` binary. Any thoughts?

koenkarsten, 2023-03-03:
Full output:
```
$ ffmpeg -c:v mpsoc_vcu_h264 -i 'http://<IP>/mpegts' -y transcoded.mp4
ffmpeg version n4.4.xlnx.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
  configuration: --prefix=/opt/xilinx/ffmpeg --datadir=/opt/xilinx/ffmpeg/etc --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --enable-libfreetype --enable-libfontconfig --extra-cflags=-I/opt/xilinx/xrt/include/xma2 --extra-ldflags=-L/opt/xilinx/xrt/lib --extra-libs=-lxma2api --extra-libs=-lxrt_core --extra-libs=-lxrt_coreutil --extra-libs=-lpthread --extra-libs=-ldl --disable-static --enable-shared
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100

<<<<<<<==  FFmpeg xrm ===>>>>>>>>
No device set hence falling to default device 0
------------------i=0------------------------------------------

   xclbin_name :  /opt/xilinx/xcdr/xclbins/transcode.xclbin
   device_id   :  0
------------------------------------------------------------

Input #0, mpegts, from 'http://<IP>/mpegts':
  Duration: N/A, start: 34685.734022, bitrate: N/A
  Program 1
  Stream #0:0[0x3e8]: Video: h264 (High) ([27][0][0][0] / 0x001B), yuv420p(tv, progressive), 1280x720 [SAR 1:1 DAR 16:9], 50 fps, 50 tbr, 90k tbn, 100 tbc
[MPSOC H.264 decoder @ 0x55bc54ac3000] timing information from stream is not available
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (mpsoc_vcu_h264) -> mpeg4 (native))
Press [q] to stop, [?] for help
Impossible to convert between the formats supported by the filter 'Parsed_null_0' and the filter 'auto_scaler_0'
Error reinitializing filters!
Failed to inject frame into filter network: Function not implemented
Error while processing the decoded data for stream #0:0
Conversion failed!
```

NastoohX, 2023-03-03:
Hi,
Thank you for your question. Seems that the issue is mainly due to FFmpeg cli, e.g., usage of http in mpeg ts. Kindly, execute the following steps and see if your issue is resolved:
1- Open  2 terminals
2- In one terminal run: ffmpeg  -f lavfi -i testsrc=duration=20:size=1920x1080:rate=60  -c:v mpsoc_vcu_h264 -b:v 8M -minrate 8M -maxrate 8M -bufsize 16M  -f mpegts udp://127.0.0.1:8000?pkt_size=1316
3- In the other terminal run a transcode pipeline: ffmpeg -re -y  -i udp://127.0.0.1:8000 -c:v mpsoc_vcu_hevc -f mpegts ~/work/t.ts
After terminating the transcode pipeline, you should be able to confirm proper transcode via mpegts:

ffprobe ./work/t.ts
ffprobe version n4.4.xlnx.1 Copyright (c) 2007-2021 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.5.0-3ubuntu1~18.04)
...
Input #0, mpegts, from '/home/nastooha/work/t.ts':
  Duration: 00:00:17.62, start: 1.416667, bitrate: 833 kb/s
  Program 1 
    Metadata:
      service_name    : Service01
      service_provider: FFmpeg
  Stream #0:0[0x100]: Video: hevc (Main) (HEVC / 0x43564548), yuv420p(tv), 1920x1080 [SAR 1:1 DAR 16:9], 60 fps, 60 tbr, 90k tbn, 60 tbc

Let us know if this resolves your problem.
Cheers,

NastoohX, 2023-03-17:
Hi,
Kindly, let us know if the above resolved you issue.
Cheers,

NastoohX, 2023-03-22:
Hi,
As there has not been any activity in this ticket, we are closing it. Feel free to reopen if needed.
Cheers,

closed 2023-03-22 by NastoohX