Xilinx/video-sdk #65: Option combination for optimal quality

notego, 2023-03-16:
There is low video quality issue event it has been encoded with same bitrate of original video source using the Xilinx FFMPEG option which has been guided in the github. (https://xilinx.github.io/video-sdk/v2.0/index.html)

Please give guide how to combinate the option for optimal video quality.

NastoohX, 2023-03-17:
Thank you for bringing this to our attention. This section of our documentation, https://xilinx.github.io/video-sdk/v2.0/tuning_encoding_quality.html, describes U30's VQ toolset. As to which combination would fit your needs, it is specific to your use case. In general, the more delay a pipeline can tolerate, the better its VQ. So, increasing LA, enabling adaptive quantisation, etc. would all be viable approaches. If you are able to provide a sample clip or the gop structure of your clip, we could provide further assistance.
Cheers,

NastoohX, 2023-03-22:
Hi,
Kindly, let us know if the above resolved you issue.
Cheers,

NastoohX, 2023-04-10:
Hi,
As there has not been any activity in this ticket, we are closing it. Feel free to reopen if needed.
Cheers,

closed 2023-04-10 by NastoohX