Xilinx/video-sdk #66: AV sink issue

notego, 2023-03-16:
There is AV sink issue when the video encoding is executed with the option like below. The video play rate worked by the speed of 0.9 and did not synchronized with audio.

ffmpeg -c:v mpsoc_vcu_h264 -i rtmp://[extrtmp2.slim.sbs.co.kr/push/S15_relay](http://extrtmp2.slim.sbs.co.kr/push/S15_relay) \
-map 0:v -map 0:a -c:a aac -b:a 128k -ar 48000 -ac 2 -c:v mpsoc_vcu_h264 -b:v 4500k -minrate 4500k -maxrate 4500k -bufsize 9000k \
-profile:v main -r 30000/1001 -vsync cfr -force_key_frames "expr:gte(t,n_forced*1)" -max_interleave_delta 0 \
-f flv -flvflags no_duration_filesize rtmp://[extrtmp2.slim.sbs.co.kr](http://extrtmp2.slim.sbs.co.kr/):1935/push/S15_U30

NastoohX, 2023-03-17:
Thank you for bringing this issue to our attention. Unfortunately, this stream is not available from our location and as such we cannot directly test it. Could you, kindly, provide responses to the following:
 
1- What is the original FPS of the stream? Was there a reason  to specify 29.97 at the output?
2-  It would be appropriate to move -vsync cfr to the decoding stage, i.e., ffmpeg -c:v mpsoc_vcu_h264 -vbsync cfr -f flv -i ... . Let us know if this improves your AV sync issue.
3- For debugging purposes, it is recommended to simplify the command line, e.g., ffmpeg  -c:v mpsoc_vcu_h264 -vsync cfr -f flv -i rtmp... -c:v mpsoc_vcu_h264 -f mp4 test.mp4 and see if problem is still observed.

Cheers,

NastoohX, 2023-03-22:
Hi,
Kindly, let us know if the above resolved you issue.
Cheers,

NastoohX, 2023-04-10:
Hi,
As there has not been any activity in this ticket, we are closing it. Feel free to reopen if needed.
Cheers,

closed 2023-04-10 by NastoohX