Xilinx/video-sdk #67: Webm video stream  [error] Error while decoding stream #0:1: Invalid data found when processing input

Mudpuppy12, 2023-03-16:
Issue:
    When using mpsoc_vcu_h264 to process a webm video into a mp4, We encountered an input error, and fail conversion to mp4. Using libx264 we encounter the same input error, but have a successful conversion to mp4 (continues to process after input error)

Details 
using 2.0 sdk - ami-03427f35504cbb6e5

Video file:
Can provide on demand.

mpsoc_vcu_h264:

    [ec2-user@ip-10-0-0-55 dennis]$ ffmpeg -loglevel repeat+level+error -nostdin -hide_banner -xlnx_hwdev 1 -i 1678991779096-video-1678991696697.webm -level 5.2 -c:v mpsoc_vcu_h264 -b:v 12000k -r 60 -f mp4 -y 1678991696697WebM.mp4
[error] Error while decoding stream #0:1: Invalid data found when processing input
 
Error while decoding stream #0:1: Invalid data found when processing input
[AVIOContext @ 0x1a59000] Statistics: 0 seeks, 464 writeouts
[aac @ 0x193e000] Qavg: 250.343
[aac @ 0x193e000] 2 frames left in the queue on closing
[AVIOContext @ 0x19c1000] Statistics: 59855813 bytes read, 0 seeks
Conversion failed!

---------------------------------------------------------------------------------------------------------------------------
libx264:

Error while decoding stream #0:1: Invalid data found when processing input4kbits/s dup=2103 drop=15 speed=0.262x
    Last message repeated 1 times
*** 7 dup!
[libx264 @ 0x5562a2697500] frame=5341 QP=18.00 NAL=0 Slice:B Poc:114 I:0    P:7    SKIP:8153 size=73 bytes
[libx264 @ 0x5562a2697500] frame=5342 QP=18.00 NAL=0 Slice:B Poc:118 I:0    P:4    SKIP:8156 size=71 bytes
[libx264 @ 0x5562a2697500] frame=5343 QP=10.93 NAL=2 Slice:P Poc:128 I:0    P:90   SKIP:8070 size=388 bytes
[libx264 @ 0x5562a2697500] frame=5344 QP=16.00 NAL=2 Slice:B Poc:124 I:0    P:36   SKIP:8124 size=106 bytes
[libx264 @ 0x5562a2697500] frame=5345 QP=18.00 NAL=0 Slice:B Poc:122 I:0    P:7    SKIP:8153 size=73 bytes
[libx264 @ 0x5562a2697500] frame=5346 QP=18.00 NAL=0 Slice:B Poc:126 I:0    P:4    SKIP:8156 size=72 bytes
[libx264 @ 0x5562a2697500] frame=5347 QP=11.33 NAL=2 Slice:P Poc:136 I:0    P:40   SKIP:8120 size=236 bytes
[libx264 @ 0x5562a2697500] frame=5348 QP=16.00 NAL=2 Slice:B Poc:132 I:0    P:35   SKIP:8125 size=107 bytes
Error while decoding stream #0:1: Invalid data found when processing input
    Last message repeated 1 times
*** 7 dup!
[libx264 @ 0x5562a2697500] frame=5349 QP=18.00 NAL=0 Slice:B Poc:130 I:0    P:6    SKIP:8154 size=73 bytes
[libx264 @ 0x5562a2697500] frame=5350 QP=18.00 NAL=0 Slice:B Poc:134 I:0    P:4    SKIP:8156 size=70 bytes
[libx264 @ 0x5562a2697500] frame=5351 QP=11.27 NAL=2 Slice:P Poc:144 I:0    P:48   SKIP:8112 size=323 bytes
[libx264 @ 0x5562a2697500] frame=5352 QP=16.00 NAL=2 Slice:B Poc:140 I:0    P:43   SKIP:8117 size=114 bytes
[libx264 @ 0x5562a2697500] frame=5353 QP=18.00 NAL=0 Slice:B Poc:138 I:0    P:9    SKIP:8151 size=77 bytes
[libx264 @ 0x5562a2697500] frame=5354 QP=18.00 NAL=0 Slice:B Poc:142 I:0    P:3    SKIP:8157 size=70 bytes
[libx264 @ 0x5562a2697500] frame=5355 QP=10.43 NAL=2 Slice:P Poc:152 I:2    P:27   SKIP:8131 size=199 bytes
[libx264 @ 0x5562a2697500] frame=5356 QP=15.00 NAL=2 Slice:B Poc:148 I:0    P:33   SKIP:8127 size=103 bytes
Error while decoding stream #0:1: Invalid data found when processing input

[aac @ 0x5562a2699040] Trying to remove 384 more samples than there are in the queue
frame= 9091 fps= 14 q=-1.0 Lsize=  220413kB time=00:02:31.78 bitrate=11895.6kbits/s dup=3933 drop=15 speed=0.24x
video:217758kB audio:2379kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.125527%
Input file #0 (1678991779096-video-1678991696697.webm):
  Input stream #0:0 (audio): 2530 packets read (1884530 bytes); 2530 frames decoded (7286400 samples);
  Input stream #0:1 (video): 5326 packets read (94525098 bytes); 5174 frames decoded;
  Total: 7856 packets (96409628 bytes) demuxed
Output file #0 (1678991696697WebM.mp4):
  Output stream #0:0 (video): 9091 frames encoded; 9091 packets muxed (222983849 bytes);
  Output stream #0:1 (audio): 7116 frames encoded (7286400 samples); 7117 packets muxed (2436245 bytes);
  Total: 16208 packets (225420094 bytes) muxed
7704 frames successfully decoded, 152 decoding errors
[AVIOContext @ 0x5562a26999c0] Statistics: 14 seeks, 872 writeouts
[libx264 @ 0x5562a2697500] frame I:54    Avg QP:10.06  size:266715
[libx264 @ 0x5562a2697500] frame P:2538  Avg QP:13.91  size: 67746
[libx264 @ 0x5562a2697500] frame B:6499  Avg QP:15.73  size:  5638
[libx264 @ 0x5562a2697500] consecutive B-frames:  1.9%  6.4%  5.8% 85.8%
[libx264 @ 0x5562a2697500] mb I  I16..4: 22.5% 51.7% 25.8%
[libx264 @ 0x5562a2697500] mb P  I16..4:  7.5% 15.7%  3.9%  P16..4: 26.9% 13.9%  6.9%  0.0%  0.0%    skip:25.2%
[libx264 @ 0x5562a2697500] mb B  I16..4:  0.4%  0.4%  0.1%  B16..8: 14.8%  2.0%  0.3%  direct: 2.2%  skip:79.8%  L0:44.6% L1:49.4% BI: 6.0%
[libx264 @ 0x5562a2697500] final ratefactor: 12.98
[libx264 @ 0x5562a2697500] 8x8 transform intra:56.6% inter:71.4%
[libx264 @ 0x5562a2697500] coded y,uvDC,uvAC intra: 42.9% 51.9% 35.7% inter: 11.2% 10.6% 1.6%
[libx264 @ 0x5562a2697500] i16 v,h,dc,p: 42% 33% 10% 15%
[libx264 @ 0x5562a2697500] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 40% 22% 20%  2%  3%  4%  3%  3%  3%
[libx264 @ 0x5562a2697500] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 43% 26% 12%  2%  4%  4%  3%  2%  2%
[libx264 @ 0x5562a2697500] i8c dc,h,v,p: 48% 24% 24%  4%
[libx264 @ 0x5562a2697500] Weighted P-Frames: Y:8.5% UV:7.6%
[libx264 @ 0x5562a2697500] ref P L0: 70.5% 16.3%  9.8%  3.3%  0.1%
[libx264 @ 0x5562a2697500] ref B L0: 91.3%  7.0%  1.7%
[libx264 @ 0x5562a2697500] ref B L1: 98.7%  1.3%
[libx264 @ 0x5562a2697500] kb/s:11773.39
[aac @ 0x5562a2699040] Qavg: 250.571
[AVIOContext @ 0x5562a266a600] Statistics: 96467708 bytes read, 0 seeks

Mudpuppy12, 2023-03-16:
I visually inspected the webm video and indeed is corrupted. I suppose this is more of a question of why is libx264 more tolerant than the mpsoc_vcu_h264

NastoohX, 2023-03-17:
Hi,
Thank you for bring this issue to our attention. Kindly, have a look at this ticket, https://github.com/Xilinx/video-sdk/issues/41, and see if it provides relevant guidance. If you could also share the offending clip, it would help us in debugging process.
Cheer,

NastoohX, 2023-03-22:
Hi,
Kindly, let us know if the above resolved you issue.
Cheers,

Mudpuppy12, 2023-03-22:
closing issue

closed 2023-03-22 by Mudpuppy12