Xilinx/video-sdk #74: 3.0 AMI-085c834167fc3623c .bashrc fix for ec2-user permission denied on login

Mudpuppy12, 2023-06-19:
Issue: 
When logging into the AMI, the following error is seen:

Last login: Mon Jun 19 16:04:48 2023 from ec2-52-21-153-129.compute-1.amazonaws.com

       __|  __|_  )
       _|  (     /   Amazon Linux 2 AMI
      ___|\___|___|

https://aws.amazon.com/amazon-linux-2/
30 package(s) needed for security, out of 63 available
Run "sudo yum update" to apply all updates.
-bash: /opt/xilinx/xcdr/setup.sh: Permission denied

The problem is the following in the .bashrc

/opt/xilinx/xcdr/setup.sh

The fix
if [ -n "$PS1" ]; then source /opt/xilinx/xcdr/setup.sh; fi

For the lazy:
sed -i 's/^\/opt\/xilinx\/xcdr\/setup.sh/if [ -n \"\$PS1\" ]; then source \/opt\/xilinx\/xcdr\/setup.sh; fi/g' .bashrc

NastoohX, 2023-06-20:
Hi,
Thank you for bringing this to our attention. (And doubly thank you for the workaround.) Issue has been escalated to our engineering team.
Cheers,

duthv-amd, 2023-08-11:
@NastoohX  do we have any updates here?

NastoohX, 2023-08-11:
Hi,
This issue has been filed internally, under https://jira.xilinx.com/browse/CR-1166411. Updates will be provided here as they become available.
Cheers,