Xilinx/video-sdk #77: Support for more recent versions of gstreamer

stemcc, 2023-07-02:
Do you have a timeframe for when the SDK will support versions of gstreamer more recent than 1.16? 

Of course, ideally, we wouldn’t have to use a patched gstreamer core and base but could instead build the specific plugins ourselves like they were any other gst plugins.

NastoohX, 2023-07-13:
Hi,
At this point, there no plans to update our Gstreamer version. (If this was to change, it will be announced in this forum.)
Regarding patches to Gstreamer core, they were required to accommodate HDR SEI and color space conversion. (See https://github.com/Xilinx/video-sdk-gstreamer/tree/3cea0c762a8960c89e6e4fe13509efc3813aa6a6/patches, for specific details.)
Cheers,

NastoohX, 2023-07-27:
Hi,
Closing this ticket, as currently there are no further solutions to explore.
Cheers,

closed 2023-07-27 by NastoohX