Xilinx/video-sdk #86: U30 does not appears, may damaged ?

cooliobr, 2024-01-23:
I bought a U30, tested it on various operating systems and different hardware setups, but it doesn't get detected on any of them. In some cases, it prevents the server from starting. Could it be a problem with the board or some configuration issue?

NastoohX, 2024-01-30:
Hi,
Please note that U30 is only supported under Linux. (See https://xilinx.github.io/video-sdk/v3.0/release_notes.html#supported-platforms-and-operating-systems for supported platforms.) If you haven't already, double check the installation process for your specific distro as per https://xilinx.github.io/video-sdk/v3.0/getting_started_on_prem.html#getting-started-with-alveo-u30-on-premises.
It would be helpful if you could provide the output of the following commands:
1. lspci -d 10ee:
2. sudo dmesg

The 1st one will indicate if kernel has been able to detect the board and the 2nd should note any boot-up issues.
Cheers,

cooliobr, 2024-02-01:
i have tried all, it not show on lspci or dmidecode ( populated slot )

NastoohX, 2024-02-01:
Hi,
That device doesn't show up in lspci or dmesg, is a strong indication of a hardware failure. Have you tried moving the card to a known working PCIe slot, e.g., switching the position with your GPU?
Cheers,

closed 2024-02-01 by NastoohX

cooliobr, 2024-02-01:
i have tried 3 different server and 2 different desktop, Ubuntu 18, 20, 22, different kernel versions. I will send this board do china.