Xilinx/video-sdk #87: Compile FFMPEG with openssl

andrewbeer-hub, 2024-01-30:
https://xilinx.github.io/video-sdk/v3.0/using_ffmpeg.html#rebuilding-ffmpeg

I tried to use the guide above to allow the xlinx ffmpeg executable to leverage https protocol.
I want to use s3 objects as ffmpeg sources.
ffmpeg -i http://commondatastorage.googleapis.com/gtv-videos-bucket/sample/BigBuckBunny.mp4 WORKS
ffmpeg -i https://commondatastorage.googleapis.com/gtv-videos-bucket/sample/BigBuckBunny.mp4 gives following error:
https protocol not found, recompile FFmpeg with openssl, gnutls or securetransport enabled.
Not so much an issue as a help request - thanks

NastoohX, 2024-02-01:
Hi,
Thank you for your question. You'll need to add openssl library to your build image. Kindly, have a look at https://trac.ffmpeg.org/wiki/CompilationGuide, for your specific OS, and follow the relevant/similar instructions to include openssl. For example, if you are building for a Debian based distro, you need to:

1. sudo apt install libssl-dev
2. include _--enable-openssl_ to your _configure_ command

Let us know how it goes.
Cheers,

NastoohX, 2024-02-12:
Closing this ticket due to inactivity. Feel free to reopen if needed.
Cheers,

closed 2024-02-12 by NastoohX