Xilinx/video-sdk #88: XmaSession statistics resources are not released

Keukhan, 2024-02-08:
Hello. experts!

I'm developing a streaming server and using XMA.
While creating and destroying the encoder repeatedly, the process suddenly terminated and the following log was output.

````
...
[XMA-Session-Stats] WARNING: Session id: 16037, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16038, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16044, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16045, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16046, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16047, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16053, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16054, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16055, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16056, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16062, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16063, type: encoder still has unused completd cu cmds
[XMA-Session-Stats] WARNING: Session id: 16064, type: encoder still has unused completd cu cmds
...
````

Analyzing the log, it was the xmaapi.cpp file, and it seems that sessions were accumulated in the g_xma_singleton->all_sessions_vec vector and could not be released. 
In the developed code, the **session = xma_enc_session_create** function and **xma_enc_session_destroy(session)** are called normally, but the XmaSession of g_xma_singleton->all_sessions_vec does not seem to be removed when the xma_enc_session_destroy function is called.

````
    for (auto& itr1: g_xma_singleton->all_sessions_vec) {
        xclLogMsg(NULL, XRT_INFO, "XMA-Session-Stats", "--------");
        XmaHwSessionPrivate *priv1 = (XmaHwSessionPrivate*) itr1.hw_session.private_do_not_use;
        if (priv1->kernel_complete_count != 0 && !priv1->using_cu_cmd_status) {
            xclLogMsg(NULL, XRT_WARNING, "XMA-Session-Stats", "Session id: %d, type: %s still has unused completd cu cmds", itr1.session_id, 
                xma_core::get_session_name(itr1.session_type).c_str());
        }
````

**Issue**
It is unavoidable if the policy is to continuously store statistical information, but as a result, CPU usage increases and processes suddenly terminate.

Is there a way to release resources when decoder, scaler and encoder are repeatedly create/destroy without terminating the process, as in my case? Please help.

NastoohX, 2024-02-08:
Hi
Thank you for bringing this issue to our attention. I have informed our engineering group of this problem and will provide an update as soon as we have one. In the mean time, could you, kindly, describe your use case in more details, e.g., frequency of creation, whether requested resources are monotonically incrementing, etc.?
Cheers,

NastoohX, 2024-02-26:
Hi,
Thank you for the extra info and steps to reproduce this issue. I have opened an internal ticket to track this issue and will provide updates here, as soon as they are available.
Cheers,
Internal ticket: CR-1191227

cdeshpan-xlnx, 2024-05-15:
@Keukhan Could you please provide exact source code of your app. I have been to trying to reproduce the issue with our XMA app and unable to reproduce yet. So I am interested in trying out with your application.

Keukhan, 2024-06-04:
@cdeshpan-xlnx 

Thank you for your reply.
I maintain an open source project called OvenMediaEngine, and XMA is used in this project. It is used by using the mpsoc_vcu_h264/hevc codec integrated in FFmpeg. 
Please let me know if you need more information to resolve the issue. 
I'm waiting for good news. :)

- Github : https://github.com/AirenSoft/OvenMediaEngine
- Decoder: https://github.com/AirenSoft/OvenMediaEngine/blob/master/src/projects/transcoder/codec/decoder/decoder_avc_xma.cpp
- Filter: https://github.com/AirenSoft/OvenMediaEngine/blob/master/src/projects/transcoder/filter/filter_rescaler.cpp#L185
- Encoder : https://github.com/AirenSoft/OvenMediaEngine/blob/master/src/projects/transcoder/codec/encoder/encoder_avc_xma.cpp