Xilinx/video-sdk #92: Can't have more than 32 channels for encoding by CU

richywalls, 2024-04-30:
Hi everyone, Is there a max capacity setting on the encoder ( by unit ) ? I'm running 10 encodings, all working ok, but when the 11th is added, it sends an error aboute no more CU, even though there is capacity, encoding usage it's less than 50%. 
I tried adding simple ffmpeg conversions ( one input, no scaling, one encoding) , 2 of this simple encodings work, but when running the third one ( up to here I have 32 channels used ) it throws an error saying no more capacity, with an usage of 65%. 
Encodings are real small, 480p, 360p and 240p.
I don't know if the 32 channel capacity is something I can configure or is a hardware maximum. 

error code when no more resources available: 
```
xrm_allocation: failed to allocate encoder cu  from reserve
``` 
use load of encoder in unit 1
```
      "usedLoad     ": "438885 of 1000000",
```

NastoohX, 2024-07-23:
Hi,
Sorry for the late reply. If this issue still persists, kindly, provide the following info:

1. Instance type, i.e., bare metal or VM
2. SDK version
3. Distro info
4. CLI that demonstrates the issue. (Failing that provide detail description on the input stream and encoder parameters.)

Kindly, review https://xilinx.github.io/video-sdk/v3.0/release_notes.html#limitations and https://xilinx.github.io/video-sdk/v3.0/specs_and_features.html#performance-tables to ensure that your requested jobs are within the specified limits.
Cheers,

richywalls, 2024-07-25:
Hi, 
1. AWS vt1.3xlarge, with AMI video_sdk_ga3.0_u30_ubuntu2204_ami_07212023-prod-qh6kiiqd5aops
2. have no idea how to obtain that, but found something it could help: 
```bash
/opt/xilinx/xrm$ cat version.json 
{
  "XRM_VERSION_STRING" : "1.5.103",
  "XRM_VERSION_RELEASE" : "202220",
  "XRM_GIT_BRANCH" : "master",
  "XRM_GIT_COMMIT_HASH" : "6c3ab244bbacc8d7edc77c347b19b0dece295d87",
  "XRM_GIT_COMMIT_DATE" : "Mon, 20 Jun 2022 22:26:31 -0700"
}
```
3.
```bash 
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.1 LTS
Release:        22.04
Codename:       jammy
```
4. running this process 7 times ( and with other ffmpeg running): 
```
ffmpeg -rw_timeout 5000000 -fflags +genpts+igndts+ignidx -rtbufsize 2000M -xlnx_hwdev 0 -c:v mpsoc_vcu_h264 -i rtmp://localhost/transcode/test -filter_complex "multiscale_xma=outputs=2: out_1_width=832:  out_1_height=468:  out_1_rate=full: out_2_width=640:  out_2_height=360:  out_2_rate=full: [480p][360p]" -map 0:a:0 -map 0:v:0 -c:v:0 copy -c:a copy -map 0:a:0 -map [480p] -vsync 0 -bf 0 -periodicity-idr 60 -qp-mode auto -g 60 -b:v:1 750K -minrate:v:1 750K -maxrate:v:1 750K -bufsize:v:1 750K -c:v:1 mpsoc_vcu_h264 -c:a copy -map 0:a:0 -map [360p] -vsync 0 -bf 0 -periodicity-idr 60 -qp-mode auto -g 60 -b:v:2 375K -minrate:v:2 375K -maxrate:v:2 375K -bufsize:v:2 375K -c:v:2 mpsoc_vcu_h264 -c:a copy -var_stream_map "v:0,a:0,name:original v:1,a:1,name:480p v:2,a:2,name:360p" -threads 0 -f hls -hls_list_size 10 -hls_time 10 -hls_flags independent_segments -hls_flags delete_segments -master_pl_name index.m3u8 -y /tmp/%v.m3u8
```
output of gpu usage: 
```
/opt/xilinx/xrm/bin/xrmadm /opt/xilinx/xrm/test/list_cmd.json{
    "response": {
        "name": "list",
        "requestId": "1",
        "status": "ok",
        "data": {
            "deviceNumber": "2",
            "device_0": {
                "dsaName    ": "xilinx_u30_gen3x4_base_2",
                "xclbinName ": "/opt/xilinx/xcdr/xclbins/transcode.xclbin",
                "uuid       ": "6d473db69dd58349a10612baa2049224",
                "isExclusive": "false",
                "cuNumber   ": "70",
                "cu_0": {
                    "cuId         ": "0",
                    "cuType       ": "IP Kernel",
                    "kernelName   ": "scaler",
                    "kernelAlias  ": "SCALER_MPSOC",
                    "instanceName ": "scaler_1",
                    "cuName       ": "scaler:scaler_1",
                    "kernelPlugin ": "/opt/xilinx/xma_plugins/libxmultiscaler.so",
                    "maxCapacity  ": "497664000",
                    "numChanInuse ": "16",
                    "usedLoad     ": "626621 of 1000000",
                    "reservedLoad ": "0 of 1000000",
                    "resrvUsedLoad": "0 of 1000000"
                },
                "cu_1": {
                    "cuId         ": "1",
                    "cuType       ": "IP Kernel",
                    "kernelName   ": "lookahead",
                    "kernelAlias  ": "LOOKAHEAD_MPSOC",
                    "instanceName ": "lookahead_1",
                    "cuName       ": "lookahead:lookahead_1",
                    "kernelPlugin ": "/opt/xilinx/xma_plugins/libxlookahead.so",
                    "maxCapacity  ": "497664000",
                    "numChanInuse ": "0",
                    "usedLoad     ": "0 of 1000000",
                    "reservedLoad ": "0 of 1000000",
                    "resrvUsedLoad": "0 of 1000000"
                },
                "cu_2": {
[... 295 lines cut ...]
[mpsoc_vcu_h264 @ 0x55fff8cfb000] Custom Rate Control Mode is Disabled
XRT build version: 2.11.722
Build hash: d1ecc641da7b8805e778926e3354e8dd3fdaa6a8
Build date: 2023-01-25 15:37:40
Git branch: 2021.1
PID: 4078589
UID: 1000
[Thu Jul 25 21:06:50 2024 GMT]
HOST: transcode-1
EXE: /opt/xilinx/ffmpeg/bin/ffmpeg
[XMA] WARNING: ffmpeg xma-vcu-encoder device warning: !! The specified Level is too low and will be adjusted !!

[mpsoc_vcu_h264 @ 0x55fff8cad000] Custom Rate Control Mode is Disabled
xrm_allocation: failed to allocate encoder cu  from reserve
[MPSOC VCU H264 encoder @ 0x55fff8dbe000] xrm_allocation: resource allocation failed
Error initializing output stream 0:5 -- Error while opening encoder for output stream #0:5 - maybe incorrect parameters such as bit_rate, rate, width or height
Conversion failed!
[XMA-Session-Stats] WARNING: Session id: 3, type: encoder still has unused completd cu cmds
````

In specific this
``` xrm_allocation: failed to allocate encoder cu  from reserve ```
In the encoder, there is 60% usage, the 40% remaining should be enough to add another encoding process like this, but the channels in use is at 31, if I drop encoding process to only use 1 instead of 3, then there will be no error.

Thanks for taking the time :)

NastoohX, 2024-07-25:
Thank you for the info. By the looks of it, it's the decoder that is running out of resources:
...
"cu_7": {
                    "cuId         ": "7",
                    "cuType       ": "Soft Kernel",
                    "kernelName   ": "kernel_vcu_decoder",
                    "kernelAlias  ": "",
                    "instanceName ": "",
                    "cuName       ": "kernel_vcu_decoder",
                    "kernelPlugin ": "",
                    "maxCapacity  ": "",
                    "numChanInuse ": "1",
                    "usedLoad     ": "1000000 of 1000000",
...

Do you happen to know the resolution of the incoming streams, and whether they are within the expected value: https://xilinx.github.io/video-sdk/v3.0/specs_and_features.html#performance-tables?
Cheers,

richywalls, 2024-07-26:
Hi, 
cu6 and cu7 are always maxed, heres the output without the 7 ffmpeg running ( other ffmpeg are running )
```
/opt/xilinx/xrm/bin/xrmadm /opt/xilinx/xrm/test/list_cmd.json
{
    "response": {
        "name": "list",
        "requestId": "1",
        "status": "ok",
        "data": {
            "deviceNumber": "2",
            "device_0": {
                "dsaName    ": "xilinx_u30_gen3x4_base_2",
                "xclbinName ": "/opt/xilinx/xcdr/xclbins/transcode.xclbin",
                "uuid       ": "6d473db69dd58349a10612baa2049224",
                "isExclusive": "false",
                "cuNumber   ": "70",
                "cu_0": {
                    "cuId         ": "0",
                    "cuType       ": "IP Kernel",
                    "kernelName   ": "scaler",
                    "kernelAlias  ": "SCALER_MPSOC",
                    "instanceName ": "scaler_1",
                    "cuName       ": "scaler:scaler_1",
                    "kernelPlugin ": "/opt/xilinx/xma_plugins/libxmultiscaler.so",
                    "maxCapacity  ": "497664000",
                    "numChanInuse ": "11",
                    "usedLoad     ": "427739 of 1000000",
                    "reservedLoad ": "0 of 1000000",
                    "resrvUsedLoad": "0 of 1000000"
                },
                "cu_1": {
                    "cuId         ": "1",
                    "cuType       ": "IP Kernel",
                    "kernelName   ": "lookahead",
                    "kernelAlias  ": "LOOKAHEAD_MPSOC",
                    "instanceName ": "lookahead_1",
                    "cuName       ": "lookahead:lookahead_1",
                    "kernelPlugin ": "/opt/xilinx/xma_plugins/libxlookahead.so",
                    "maxCapacity  ": "497664000",
                    "numChanInuse ": "0",
                    "usedLoad     ": "0 of 1000000",
                    "reservedLoad ": "0 of 1000000",
                    "resrvUsedLoad": "0 of 1000000"
                },
                "cu_2": {
                    "cuId         ": "2",
                    "cuType       ": "IP Kernel",
                    "kernelName   ": "lookahead",
                    "kernelAlias  ": "LOOKAHEAD_MPSOC",
                    "instanceName ": "lookahead_2",
                    "cuName       ": "lookahead:lookahead_2",
                    "kernelPlugin ": "/opt/xilinx/xma_plugins/libxlookahead.so",
                    "maxCapacity  ": "497664000",
                    "numChanInuse ": "0",
                    "usedLoad     ": "0 of 1000000",
                    "reservedLoad ": "0 of 1000000",
                    "resrvUsedLoad": "0 of 1000000"
                },
                "cu_3": {
                    "cuId         ": "3",
                    "cuType       ": "IP Kernel",
                    "kernelName   ": "decoder",
                    "kernelAlias  ": "DECODER_MPSOC",
                    "instanceName ": "decoder_1",
                    "cuName       ": "decoder:decoder_1",
                    "kernelPlugin ": "/opt/xilinx/xma_plugins/libvcu-xma-dec-plg.so",
                    "maxCapacity  ": "497664000",
                    "numChanInuse ": "11",
                    "usedLoad     ": "612030 of 1000000",
[... 196 lines cut ...]
                "xrmPluginFileName": "/opt/xilinx/xrm/plugin/libxrmU30Encoder.so"
            }
        }
    }
}
```
With this initial conditions I can run 7 more ffmpeg processes, so I assume that usedLoad is not relevant or maybe it's not showing correct info. 

The input signal has this params: 
```
Input #0, flv, from 'rtmp://localhost/transcode/test':
  Metadata:
    |RtmpSampleAccess: true
    Server          : NGINX RTMP (github.com/arut/nginx-rtmp-module)
    displayWidth    : 1280
    displayHeight   : 720
    fps             : 30
    profile         : 
    level           : 
  Duration: 00:00:00.00, start: 550983.010000, bitrate: N/A
  Stream #0:0: Audio: aac (LC), 48000 Hz, stereo, fltp, 196 kb/s
  Stream #0:1: Video: h264 (High), yuv420p(tv, progressive), 1280x720, 4194 kb/s, 30 fps, 30 tbr, 1k tbn, 60 tbc
```
That resolution i'ts in the expected value. 
Cheers

NastoohX, 2024-07-26:
Thank you for the info. This item is now internally tracked: CR-1207797
Cheers,