Xilinx/video-sdk #93: Trying to compile ffmpeg on AL2 ami but unable to configure due to /opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclLogMsg'

tonetechnician, 2024-07-24:
Hi there, 

I'm trying to compile an updated version of ffmpeg that support .exr DWAA encoding for AWS's VT1 instances. I've run through the [ffmpeg compilation docs](https://trac.ffmpeg.org/wiki/CompilationGuide/Centos) to compile x264 and x265 and [xilinx ffmpeg compilation docs](https://xilinx.github.io/video-sdk/v3.0/using_ffmpeg.html#using-the-source-code) to compile a custom version of ffmpeg. I'm using a Xilinx provided AMI (AMD-Xilinx-VT1-3.0-eks-1.22-3b50633b-1f1b-4c14-b26b-cb33681d076f). 

I've applied the git-patch given in this repo to the ffmpeg source code I'm basing from and made the necessary edits. When running the configure for ffmpeg, I'm running into an error that the xilinx libxma2api.so included with the ami are missing references to particular functions which should be included.

I run 

```.bash
PATH="$HOME/bin:$PATH" PKG_CONFIG_PATH="$HOME/ffmpeg_build/lib/pkgconfig" ./configure --prefix="$HOME/ffmpeg_build" --datadir="$HOME/ffmpeg_build/etc" --pkg-config-flags="--static" --extra-cflags="-I$HOME/ffmpeg_build/include" --extra-ldflags="-L$HOME/ffmpeg_build/lib"  --enable-x86asm --enable-libxma2api --disable-doc --enable-libxvbm --enable-libxrm --extra-cflags="-I/opt/xilinx/xrt/include/xma2" --extra-ldflags="-L/opt/xilinx/xrt/lib" --extra-libs=-lpthread  --extra-libs=-lm --disable-static --enable-shared --enable-libx264 --enable-libx265 --enable-gpl
```

And from the ffbuild/config.log I get

```.bash
check_func_headers xma.h xma_initialize -I/opt/xilinx/xrt/include/xma2 -I/opt/xilinx/xrt/include -L/opt/xilinx/xrt/lib -lxma2api
test_ld cc -I/opt/xilinx/xrt/include/xma2 -I/opt/xilinx/xrt/include -L/opt/xilinx/xrt/lib -lxma2api
test_cc -I/opt/xilinx/xrt/include/xma2 -I/opt/xilinx/xrt/include -L/opt/xilinx/xrt/lib
BEGIN /tmp/ffconf.fZ3WV6VN/test.c
    1   #include <xma.h>
    2   #include <stdint.h>
    3   long check_xma_initialize(void) { return (long) xma_initialize; }
    4   int main(void) { int ret = 0;
    5    ret |= ((intptr_t)check_xma_initialize) & 0xFFFF;
    6   return ret; }
END /tmp/ffconf.fZ3WV6VN/test.c
gcc -D_ISOC11_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -I/home/ec2-user/ffmpeg_build/include -I/opt/xilinx/xrt/include/xma2 -std=c11 -fomit-frame-pointer -fPIC -pthread -I/usr/include/libdrm -I/home/ec2-user/ffmpeg_build/include -I/home/ec2-user/ffmpeg_build/include -I/opt/xilinx/xrt/include/xma2 -I/opt/xilinx/xrt/include -L/opt/xilinx/xrt/lib -c -o /tmp/ffconf.fZ3WV6VN/test.o /tmp/ffconf.fZ3WV6VN/test.c
gcc -L/home/ec2-user/ffmpeg_build/lib -L/opt/xilinx/xrt/lib -Wl,--as-needed -Wl,-z,noexecstack -I/opt/xilinx/xrt/include/xma2 -I/opt/xilinx/xrt/include -L/opt/xilinx/xrt/lib -o /tmp/ffconf.fZ3WV6VN/test /tmp/ffconf.fZ3WV6VN/test.o -lxma2api -lm -lpthread
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclExecWait'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt_core::xclbin::get_kernel_arguments(axlf const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclLogMsg'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclAllocBO'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt_core::xclbin::get_softkernels(axlf const*)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `uuid_copy'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclOpenContext'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt_core::xclbin::get_axlf_section(axlf const*, axlf_section_kind)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt_core::config::detail::get_uint_value(char const*, unsigned int)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclFreeBO'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclLoadXclBin'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt_core::xclbin::get_cuisr(axlf const*)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt_core::xclbin::get_cus(ip_layout const*, bool)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt_core::xclbin::get_cus(ip_layout const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `pcidev::get_dev(unsigned int, bool)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `pcidev::get_dev_total(bool)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt::kernel::kernel(xrt::device const&, xrt::uuid const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, xrt::kernel::cu_access_mode)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclProbe'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclOpen'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt_core::config::detail::get_bool_value(char const*, bool)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xrt_core::config::detail::get_string_value(char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclMapBO'
collect2: error: ld returned 1 exit status
ERROR: libxma2api not found using pkg-config
```

I can confirm the libraries in the AMI are on the same version as one latest in this repo by checking

```.json
// /opt/xilinx/xrt/version.json
{
  "BUILD_VERSION" : "2.11.716",
  "BUILD_VERSION_DATE" : "Thu, 07 Jul 2022 20:49:26 +0000",
  "BUILD_BRANCH" : "2021.1",
  "VERSION_HASH" : "22cbc46ebc878a0548f6329922807031e2c5d50f",
  "VERSION_HASH_DATE" : "Wed, 29 Jun 2022 16:21:26 +0530"
}
```

I am able to find these functions in this repo so I'm not sure where it's going wrong that the configure is failing. 

Any ideas or thoughts to help me link these would be greatly appreciated!

NastoohX, 2024-07-25:
Hi,
Based on BUILD_VERSION, seems that you are not using the most recent version of the SDK. If this is not intentional, please refer to https://xilinx.github.io/video-sdk/v3.0/getting_started_on_vt1.html#installing-the-sdk-on-an-existing-ami for further details.
In the meantime, I have created an internal ticket for this: CR-1207699.
Cheers,

tonetechnician, 2024-07-26:
Thanks @NastoohX I'm using a prebuilt ami that has eks, so ideally we can get it to work with this version. I will also look into building on the latest version following those docs.