Xilinx/video-sdk #95: 403 Forbidden during install on Ubuntu focal

daverin, 2024-08-12:
My docker builds worked a few days ago but are now all failing for xrt packages install on ubuntu 20.04

```
99.16 E: Failed to fetch https://dwgrutst0mjb3.cloudfront.net/filestore/5f/5f7f6e72c4f52f43d709a4447e650a9c75ed9bce?response-content-type=application/x-debian-package&x-jf-traceId=ba386013109e11ed&X-Artifactory-repositoryKey=debian-packages-cache&X-Artifactory-projectKey=default&X-Artifactory-artifactPath=pool/xrt_202110.2.11.722_20.04-amd64-xrt.deb&X-Artifactory-username=anonymous&Expires=1723449143&Signature=I93dAiy82Ze-7oH~odlDWUcB6t8o-Ab6b-vkzihNsKxq9pljsqSTs-PDrbJOfM9cl4jkTYj-BkKAcoo9x92tazLvP8HPnNdjy6Ua9ipKSLbCJ7AYPn1U-FtAnBIVl6PrnnjYAScNyZfQQuxi1k9nYuk8mRNn-cZUyE686GI8NC2zy-qTykCuiBVEbNeeU5snqzzUA2Vm9y7fNxA6UjPJYi63TJtBP9lQ5Tzg9p4w75pUpQFkf8wbH8wrITB-lJhHI7Xf7KZGp8nm30oh~qrW-PJOwwmJAKfAFS8eus7wzcGdsm07cU0k55hr7R81~KVQ1n~-2QZJ63chN9KXR0nKqw__&Key-Pair-Id=K1N32JWIQUWNE3  403  Forbidden [IP: 52.85.42.198 443]
99.16 E: Failed to fetch https://dwgrutst0mjb3.cloudfront.net/filestore/e8/e879d3e3c08dc626ae3b37d477837bc09fdedb5b?response-content-type=application/x-debian-package&x-jf-traceId=b29b8039df358abd&X-Artifactory-repositoryKey=debian-packages-cache&X-Artifactory-projectKey=default&X-Artifactory-artifactPath=pool/xilinx-u30-gen3x4-validate_2-3380610_all.deb&X-Artifactory-username=anonymous&Expires=1723449148&Signature=MQ11iWfhjrNR8G6OZr7HhgZQDu-27NlCed0X7rxX2SvvyqjR8zLKKaaEGTRhqJEi1QcrdDbNOaZLCJenLocNCKwPFVucpr4qOXxuEgN6zgRgM~i~brUUUGAYxyAdCRt2qRPXFuWKPYEbKOGFFgkqkL0QfZHMAe~aECLxsxH518FRN1a7ba1gVp2ihHhx2zMhqDn7uroAJiO2IijTAfEwhNHHF-Aflf1sQV8xx2oUJJKJ7EULugxrAm8W~NPGkMoubPnt25aeJes4iO0R34GJQxiXyQ-JToWbI0RYz2UJvrjn0x9rnXXR8eNp09HL4aXB58-H8p0hSJ9ooK1daWeslA__&Key-Pair-Id=K1N32JWIQUWNE3  403  Forbidden [IP: 52.85.42.198 443]
99.16 E: Failed to fetch https://dwgrutst0mjb3.cloudfront.net/filestore/d3/d3d71484ddcd29df5c2b6b96e94139b58af4448e?response-content-type=application/x-debian-package&x-jf-traceId=2217843f4b919df5&X-Artifactory-repositoryKey=debian-packages-cache&X-Artifactory-projectKey=default&X-Artifactory-artifactPath=pool/xilinx-u30-gen3x4-base_2-3485939_all.deb&X-Artifactory-username=anonymous&Expires=1723449148&Signature=J5v6lJHknSAYjFbIp3x4YcNJH06QVUpydE~T0VeJf0t3y7zS07Mxb2k1fInJ4pRokfQSPeU5HV5gCQMz4aVmlwkwPlQzWdXJGY9MbIfBMMwEwepcQG9g2GhF6ay1cab3NdNbgIIk0ivPsjcCL~6IO110Ud-4wEEpkRc3Kt7fMAaRFz61HQ2Ortxjp4FV8CgrqP6-flbFqHK0lXx52Peo2QRfG9vfxZMC5QKw2MZUnknsHeglRjNfIPI-wWWe6sjWChOeFlgsjOVxIESzIPN8h6CpGPdbh4fKHP0jIkrSfLfr4KTgxv1-j-M1pzyBG3p78W1o8RPOZ8DDS0fHuYQUEQ__&Key-Pair-Id=K1N32JWIQUWNE3  403  Forbidden [IP: 52.85.42.198 443]
99.16 E: Failed to fetch https://dwgrutst0mjb3.cloudfront.net/filestore/30/3067e9bdf530d1b4cdd6b879acc5bfe4f7551195?response-content-type=application/x-debian-package&x-jf-traceId=422def8a45f95cea&X-Artifactory-repositoryKey=debian-packages-cache&X-Artifactory-projectKey=default&X-Artifactory-artifactPath=pool/xilinx-u30-gstreamer-1.16.2_1.1.0_20.04-amd64.deb&X-Artifactory-username=anonymous&Expires=1723449226&Signature=XQtIaWf4CyZzcS1gBBu3-L-hO0KPXdqLc28vkG3BlcD1IvwD7gEn7ZleFAT6nOQdf2hRq3EyPWA26ROkKKU3G9Y1woUJMrX57u~Q~o3f0e5pf4doyUjYQT0gHP-DwcSaSyL-NP70QHN2uyJbLW9-3eSNk0sl-E14Zwo7-7dmU597NZiN4Qh0AAi1gdW9BFLf7hBiUbNmzI9hsKqbWO1nLlV~oLgnAPfB9x5BdYZ4IbF1Hwyo5wT6QGtXwg4dh4M7QRwd8fIz4Zk-PmKMjxXmd2e37kXIQLckCMTLwPpddwjuK8KD~1tDhrLCJZwsEGJTGiQpmiYvDmA9g2R0t3FCgw__&Key-Pair-Id=K1N32JWIQUWNE3  403  Forbidden [IP: 52.85.42.198 443]
99.16 E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Please help. I was able to download and install xrt deb package via the amd downloads, but I'm not sure where to find xilinx-alveo-u30-core deb.

NastoohX, 2024-08-12:
Hi,
Thank you for bringing this issue to our attention. Can you, kindly, describe the steps that led to the above?
Cheers,

daverin, 2024-08-19:
Hi @NastoohX 

```
RUN echo 'deb [trusted=yes] https://packages.xilinx.com/artifactory/debian-packages focal main' > /etc/apt/sources.list.d/xilinx.list
RUN echo 'Acquire { https::Verify-Peer false }' > /etc/apt/apt.conf.d/99verify-peer.conf

RUN --mount=type=cache,sharing=locked,target=/var/cache/apt apt-get update -qq && apt-get install -qqy --no-install-recommends \
    apt-utils \
    sudo  \
    xrt=2.11.722 \
    xilinx-alveo-u30-core \
    xilinx-alveo-u30-examples \
    software-properties-common \
    && apt-mark hold xrt \
	&& rm -rf /var/lib/apt/lists/*

```

This fails 👆🏾

ramark3237, 2024-08-22:
Please try https://amd.github.io/ama-sdk/latest/troubleshooting.html#ubuntu-20-04-installation which helps to resolve the issue.

closed 2024-10-11 by NastoohX

NastoohX, 2025-05-29:
Hi Artem,
Can you double check that "apt-get" calls are pointing to the newly installed apt utility? (Perhaps take a look at its creation date.) Also, it could be that the PPA fix only updates "apt" and not "apt-get". Try changing apt-get calls to apt.
Cheers,