Xilinx/video-sdk #96: Accurate seeking in video

balaprassanna, 2024-09-24:
I have a custom workflow where I cannot generate the HLS content for the entire video file in a single command. I have to split and transcode the video as chunks (say each 15 minute is a chunk). As suggested in this [link](https://trac.ffmpeg.org/wiki/Seeking), -ss and -to can be used in ffmpeg to achieve the same. I am aware that -ss when used as input parameter does key frame based lookup and the transcoding starts from the nearest keyframe which is not accurate for my use case. I have observed that the same happens in U30 and MA35D. But, ffmpeg with libx264 provides accurate seek support even when -ss is used as input parameter if the file is transcoded. How do i achieve the same using Xilinx's video SDK?

> As of FFmpeg 2.1, when transcoding with `ffmpeg` (i.e. not [stream copying](https://ffmpeg.org/ffmpeg.html#Stream-copy)): "-ss" is also "frame-accurate" even as input option.

NastoohX, 2024-10-08:
Hi,
Thank you for bring this issue to our attention. I'll raise the issue with our dev team, and if there are specific hooks, they'll be noted here. Meanwhile, would using "-ss" after specifying the input give the required accuracy?
Cheers,

balaprassanna, 2024-10-09:
Hi,
Thanks for looking into this. Specifying -ss after the input results in slow processing, especially for large video files. According to the FFmpeg wiki [link](https://trac.ffmpeg.org/wiki/Seeking), using -ss after the input causes FFmpeg to transcode the video from the beginning up to the seek position, rather than performing a keyframe-based seek. This leads to the slower processing.

NastoohX, 2024-10-11:
Hi,
The feedback provided from R&D, indicates that our SDK does not offer accelerated parsing for fast skipping.
Cheers,

closed 2024-10-11 by NastoohX

bhargav-sudo, 2024-10-12:
Hi,

Can we expect this support in future SDK release atleast with MA35D ?