Xilinx/video-sdk #98: Getting Started on Amazon EC2 VT1 doesn't work anymore as of November 6th, 2024

kristof-mattei, 2024-11-13:
\<TDLR>

To install `xilinx-alveo-u30-ffmpeg` WITH `xrt=2.11.722`, we need `xilinx-alveo-u30-ffmpeg=3.0.0`.

`xilinx-alveo-u30-ffmpeg=3.0.0` depends on `xilinx-u30-xffmpeg=4.4.xlnx.2`, `xilinx-u30-launcher=1.1.1` & `xilinx-u30-jobslotreservation=1.1.1`.

BUT `xilinx-u30-launcher=1.1.1` & `xilinx-u30-jobslotreservation=1.1.1` depend on `xilinx-u30-xffmpeg=5.1.4.xlnx.3`

Were the old `1.1.1` versions of `xilinx-u30-launcher` & `xilinx-u30-jobslotreservation` overwritten by newer ones?

\</TDLR>

I'm following the instructions at https://xilinx.github.io/video-sdk/v3.0/getting_started_on_vt1.html#installing-the-sdk-on-an-existing-ami

Quote:
> Note: it is imperative for the XRT package to be set to version 2.11.722.

```
sudo apt-get update
sudo apt-get install xrt=2.11.722
sudo apt-mark hold xrt
sudo apt-get install xilinx-alveo-u30-core
sudo apt-get install xilinx-alveo-u30-ffmpeg
```

This fails with:
```
The following packages have unmet dependencies:
 xilinx-alveo-u30-core : Depends: xrt (= 2.11.730) but 2.11.722 is to be installed
```

Since the documentation requires the installation of `xrt=2.11.722` I went into the packages website and searched for the version published when our builds still worked: https://packages.xilinx.com/ui/packages/deb:%2F%2Fxilinx-alveo-u30-core?name=xilinx-alveo-u30-core&type=packages

Then we get into a whole dependency chain that I tried to manually solve:

I install `xilinx-alveo-u30-core=3.0.1`, which depends on `xilinx-u30-xcdr=3.0.0`, but `3.1.0` is to be installed.

I install `xilinx-u30-xcdr=3.0.0`, which depends on `xilinx-u30-xmpsoccodecs=1.2.0`, but `1.3.0` is to be installed.

I install `xilinx-u30-xmpsoccodecs=1.2.0` which installs, and then I go back up, installing `xilinx-u30-xcdr=3.0.0` and `xilinx-alveo-u30-core=3.0.1`.

However, I then try to install `xilinx-alveo-u30-ffmpeg`, but cannot because it depends on `xilinx-alveo-u30-core>=3.1.0`, but I have it pinned on `3.0.1`. Looking at the packages website I see a version from before November 6th, namely `3.0.0`: 

https://packages.xilinx.com/ui/packages/rpm:%2F%2Fxilinx-alveo-u30-ffmpeg?name=xilinx-alveo-u30-ffmpeg&type=packages

Trying to install `xilinx-alveo-u30-ffmpeg=3.0.0` yields: 

```
 xilinx-alveo-u30-ffmpeg : Depends: xilinx-u30-xffmpeg (= 4.4.xlnx.2) but it is not going to be installed
                           Depends: xilinx-u30-launcher (= 1.1.1) but it is not going to be installed
                           Depends: xilinx-u30-jobslotreservation (= 1.1.1) but it is not going to be installed
```

And this is an impossible situation. There are no older versions than `xilinx-u30-launcher=1.1.1` & `xilinx-u30-jobslotreservation=1.1.1`, both last modified on December 1st, 2022:

[image]
[image]

Notice the 'Modified' date. The date above does show November 7th, 2024, but there are no newer or older versions than `1.1.1`. 

Since the only version available of `xilinx-u30-launcher` and `xilinx-u30-jobslotreservation` needs `xilinx-u30-xffmpeg=5.1.4.xlnx.3` (https://packages.xilinx.com/ui/repos/tree/DebianInfo/debian-packages-cache/pool/xilinx-u30-launcher_1.1.1_22.04-amd64.deb > Dependencies, https://packages.xilinx.com/ui/repos/tree/DebianInfo/debian-packages-cache/pool/xilinx-u30-jobslotreservation_1.1.1_22.04-amd64.deb > Dependencies)

AND

Since `xilinx-alveo-u30-ffmpeg=3.0.0` depends on `xilinx-u30-xffmpeg=4.4.xlnx.2` I am now unable to build the image for AWS' VT1.

My initial thought is that `xilinx-u30-launcher=1.1.1` and `xilinx-u30-jobslotreservation=1.1.1` were overridden by a newer version without bumping the package number

NastoohX, 2024-11-18:
Hi,
Thank you for bringing this problem to our attention. Indeed, this seems to be an issue with our PPA. Meanwhile, direct download from, https://packages.xilinx.com/artifactory/debian-packages/pool/xilinx-alveo-u30-core_3.0.1_22.04-amd64.deb, may address the issue. (Modify the _22.04 portion of the URL to match your distro.)
Cheers,

georgesbiaux, 2024-11-26:
Having the same exact problem than @kristof-mattei. Could't not yet figure out how to install directly using deb files. Is there a fix of the PPA planned @NastoohX ?

NastoohX, 2024-11-26:
Indeed. There is an issue in our PPA. Unfortunately, this issue is exacerbated by the fact that it takes couple of days for PPA caches to clear out. Our IT team is root-causing this.
Cheers,

scottlamb, 2024-11-26:
> Indeed. There is an issue in our PPA. Unfortunately, this issue is exacerbated by the fact that it takes couple of days for PPA caches to clear out. Our IT team is root-causing this.

When you change packages, are you adding a file to the PPA with the new version number, or are you replacing an existing file? Both recent history—in which the package resolver was suddenly unable to find `xilinx-alveo-u30-core` versions compatible with the locked `xrt=2.11.722`—and this comment about caching problems suggest to me that you're doing the latter. But this is deeply wrong.

scottlamb, 2024-11-26:
Additionally, do you follow semantic versioning? It appears that between `xrt=2.1.722` and `xrt=2.1.730` you switched a `pts` field from 32-bit to 64-bit? I suppose the documented "imperative" to stay on 2.1.722?

> Note: it is imperative for the XRT package to be set to version 2.11.722.

It is deeply confusing to us that you continue to release new versions but advise your customers not to use them. What's the plan here?

Xaeroxe, 2024-11-27:
Managed to get a workaround going by extracting the debian packages from the last known working build. Did this with `dpkg-repack`. Package `xilinx-u30-xcdr_3.1.0_amd64.deb` has a post installation step which removes data from `/opt/xilinx/xcdr/xclbins/transcode.xclbin`. This step is not idempotent, and will fail if it has already been applied. So it was necessary to obtain `transcode.xclbin` in its original state from [here](https://github.com/Xilinx/app-xcdr/tree/265ba34519c5b6ac51fad5c527f9e8643f21844c/xclbins). I then used `ar` to unpack the debian file, and `tar` to unpack the `data.tar.zst` file. I placed the original `transcode.xclbin` back in its place, rebuilt `data.tar.zst`, and then repackaged the debian archive with `ar`. If you want to do something similar in your own setup please remember that the final `ar` command for packing the debian must order the files like so

1. `debian-binary`
2. `control.tar.xz`
3. `data.tar.xz`

It's not pretty but it got us running again.

NastoohX, 2024-12-03:
Hi,
Thank you all for your patience. PPA is up and running again. We, unfortunately, had a release-escape, where code from our internal PPA got into the public one, and caused the above unforeseen versioning issues. Again, thank you all for your workarounds and investigative works.
Cheers,

georgesbiaux, 2024-12-04:
Thanks @NastoohX, it works well now

closed 2024-12-05 by NastoohX

Xaeroxe, 2025-01-14:
@NastoohX It seems a similar problem happened with the rpm repo. Using this with Amazon Linux 2 I'm getting this error. I think a similar fix is needed there.

```
$ sudo yum install -y xilinx-alveo-u30-core xilinx-u30-xffmpeg

...

--> Finished Dependency Resolution
Error: Package: xilinx-u30-xffmpeg-5.1.4.xlnx.3-1.x86_64 (Amzn2Artifactory)
           Requires: xilinx-u30-xcdr >= 3.1
           Installing: xilinx-u30-xcdr-3.0.0-1.x86_64 (Amzn2Artifactory)
               xilinx-u30-xcdr = 3.0.0-1
 You could try using --skip-broken to work around the problem
 You could try running: rpm -Va --nofiles --nodigest
```