Xilinx/video-sdk #100: Advanced recovery for Alveo U30MA that does not show up in lspci

justjanne, 2024-11-27:
**Steps to reproduce**

- Plug card in
- Boot system

**Expected behavior**

- Card shows up under lspci

**Actual behavior**

- Card is not recognized in UEFI or lspci

**Attempted recovery steps**

- The issue remains under cold and warm reboot
- The issue remains with fastboot disabled
- Both lights glow blue after initialization

**Further questions**

The documentation states:

> In the event that your card has become corrupted and that a reset is not sufficient, you will need to recover it. This is normally be done in-band via a set of simple commands. Should the card’s flash devices become corrupted to the point where it is no longer detected by PCIe, out-of-band recovery with a JTAG cable may be required.
> […]
> Should there be an issue with the standard in-band recovery process, it is still possible to recover the card using out-of-band methods. Please [contact Xilinx](https://github.com/Xilinx/video-sdk/issues) for more details.

1. Is such a recovery method possible in this situation?
2. What additional hardware and software (e.g., alveo programming cable, vivado, etc) is needed?
3. If not, would it be possible to send the card in and have Xilinx do the recovery? If so, how much would that cost?

**Additional information**

Purchased from this [eBay Listing](https://www.ebay.de/itm/145139334030) ([archive.org](https://web.archive.org/web/20241127185251/https://www.ebay.de/itm/145139334030)|[archive.is](https://archive.is/d6Kel))

> YI BAO TONG TRADING CO LIMITED  
> FLAT RM D18 3F WONG KING INDUSTRIAL, BUILDING NO 2 TAI YAU STREET KL  
> Hong Kong 852  
> Hongkong

```
# uname -a
Linux encoder-justjanne 5.15.0-126-generic #136-Ubuntu SMP Wed Nov 6 10:38:22 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

```
# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy
```

```
# dpkg -l | grep -i xilinx
ii  xilinx-alveo-u30-core                  3.0.1                                   amd64        Alveo Core Package
ii  xilinx-container-runtime               1.1.27                                  amd64        Xilinx container runtime is an extension of runC, with modification to add xilinx devices before running containers. Since it is a runC compliant runtime, xilinx container runtime can be integrate with various contianer orchestrators, including docker and podman.
ii  xilinx-sc-fw-u30                       6.3.8-1.cd35f69                         all          Xilinx xilinx-sc-fw-u30 deployment package.
ii  xilinx-u30-gen3x4-base                 2-3485939                               all          Xilinx Inc xilinx-u30-gen3x4-base deployment package.
ii  xilinx-u30-gen3x4-validate             2-3380610                               all          Xilinx Inc xilinx-u30-gen3x4-validate validate package.
ii  xilinx-u30-xcdr                        3.0.0                                   amd64        Xilinx Transcoder Application for MPSoC Device
ii  xilinx-u30-xlookahead                  1.2.0                                   amd64        Xilinx Lookahead Plugin
ii  xilinx-u30-xma-multiscaler             1.1.1                                   amd64        Xilinx Multi-Scaler Plugin
ii  xilinx-u30-xmpsoccodecs                1.2.0                                   amd64        Xilinx MPSoC H264/HEVC Encoder and Decoder Plugins
ii  xilinx-u30-xrm-decoder                 1.1.1                                   amd64        Xilinx MPSoC XRM Decoder Plugin
ii  xilinx-u30-xrm-encoder                 1.1.1                                   amd64        Xilinx MPSoC XRM Encoder Plugin
ii  xilinx-u30-xrm-multiscaler             1.1.1                                   amd64        Xilinx MPSoC XRM Scaler Plugin
ii  xilinx-xvbm                            1.0.3                                   amd64        Xilinx Video Buffer Manager (XVBM)
ii  xrm                                    1.5.103                                 amd64        Xilinx FPGA Resource Manager (XRM)
hi  xrt                                    2.11.722                                amd64        Xilinx RunTime stack for use with Xilinx FPGA platforms
```

[**dmesg.log**](https://github.com/user-attachments/files/17938910/dmesg.log)

[**lspci.log**](https://github.com/user-attachments/files/17938908/lspci.log)

NastoohX, 2024-12-03:
Hi,
Looked at both dmesg and lspci logs, and by the looks of it this card is not detected by the kernel. A typical dmesg looks like:
`
[   13.224360] xocl: loading out-of-tree module taints kernel.
[   13.224963] xocl: module verification failed: signature and/or required key missing - tainting kernel
[   13.236152] xocl 0000:21:00.1:  ffff94b7043ef0d0 xocl_fill_dsa_priv: found vsec cap, platform type 1
[   13.236398] xocl 0000:21:00.1:  ffff94b7043ef0d0 get_header_from_vsec: Does not get UUID ROM
[   13.236444] xocl 0000:21:00.1:  ffff94b7043ef0d0 __xocl_subdev_create: Created subdev rom inst 0 level 2
[   13.236601] xocl 0000:21:00.1: mb_scheduler.u.12582912 ffff94af9c681010 exec_create: did not get IRQ resource
[...
[   13.236688] xocl 0000:21:00.1:  ffff94b7043ef0d0 exec_reset: exec->xclbin(00000000-0000-0000-0000-000000000000),xclbin(00000000-0000-0000-0000-000000000000)
[   13.334198] xclmgmt init()
[   13.335942] xclmgmt 0000:21:00.0:  ffff94b7043ed0d0 xclmgmt_probe: Driver: 2.11.722,d1ecc641da7b8805e778926e3354e8dd3fdaa6a8
[   13.335953] xclmgmt 0000:21:00.0:  ffff94b7043ed0d0 xclmgmt_probe: probe(pdev = 0x00000000ef42597b, pci_id = 0x0000000092054eb4)
[   13.336039] xclmgmt 0000:21:00.0:  ffff94b7043ed0d0 xocl_fill_dsa_priv: found vsec cap, platform type 1
[   13.336611] xclmgmt 0000:21:00.0:  ffff94b7043ed0d0 xclmgmt_probe: minimum initialization done
[   13.336947] xclmgmt 0000:21:00.0:  ffff94b7043ed0d0 __xocl_subdev_create: Created subdev rom inst 0 level 0
[   ...
`
As such, the issue is either with the card itself, chassis' pcie slot, or bios configuration. To rule out issue with the card, you may try the following:

1. Swap to a known working slot and see if problem persists.
2. Depending on you bios, you may be able to detect installed cards. If so, verify if that other cards and this card are visible to bios.
3. Do a physical inspection of the card, for visible defects.

I've forward this ticket to our support team for further feedback if any.
Cheers,

justjanne, 2024-12-04:
I've tested all PCIe slots with other cards, an AMD RX5700XT shows up in all of them, this card does not.

There are no visible defects on the card. The blue LEDs do light up.

NastoohX, 2024-12-06:
Hi,
We have the following generic guide for debugging Alveo cards:  https://xilinx.github.io/Alveo-Cards/master/debugging/build/html/docs/common-steps.html . Try to see if some of the suggestion apply to your case, e.g., disabling the fast boot. 
Cheers,

justjanne, 2024-12-14:
I have chosen to RMA the card instead.

closed 2024-12-14 by justjanne