Xilinx/video-sdk #101: Ubuntu jammy package feed is broken

tamasgp, 2024-11-28:
During on-premise installation the following error is visible (xrt=2.11.722 is already installed and set to hold):

**apt-get install xilinx-alveo-u30-core**
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 **xilinx-u30-xcdr : Depends: xilinx-u30-xmpsoccodecs (= 1.2.0) but 1.3.0 is to be installed**
E: Unable to correct problems, you have held broken packages.

NastoohX, 2024-12-03:
Hi,
Indeed, we had an issue with our public PPA, which has now been resolved. Please retry the installation procedure.
Cheers,

tamasgp, 2024-12-05:
Thanks, I confirm that now all the dependencies are fine!

closed 2024-12-05 by tamasgp