Xilinx/video-sdk #102: Transformation job failed when using more than one ama_u30 device with Xilinx

dannylu88, 2025-01-06:
**Cloud Provider:** AWS
**Instance:** VT1.3xlarge node group in EKS, which has 1 U30 accelerator cards, and 2 addressable devices
**AMI:** AMD Xilinx Video SDK AMI with ECS support for VT1 Instances (AL2)
**EKS version:** 1.3.1

**Device Plugin version:** 1.2.0
`        env:
        - name: U30NameConvention
          value: CommonName
        - name: U30AllocUnit
          value: Device
`

**Other notes:**
/opt/xilinx/xrt/setup.sh and xbutil examine return no error. Ran from the container which also built from AL2

```
bash-4.2# source /opt/xilinx/xrt/setup.sh
XILINX_XRT        : /opt/xilinx/xrt
PATH              : /opt/xilinx/xrt/bin:/opt/xilinx/ffmpeg/bin:/opt/xilinx/xrt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
LD_LIBRARY_PATH   : /opt/xilinx/xrt/lib:/opt/xilinx/xrt/lib:/opt/xilinx/ffmpeg/lib
PYTHONPATH        : /opt/xilinx/xrt/python:/opt/xilinx/xrt/python
```

```
bash-4.2# xbutil examine
System Configuration
  OS Name              : Linux
  Release              : 4.14.305-227.531.amzn2.x86_64
  Version              : #1 SMP Tue Feb 14 09:55:28 UTC 2023
  Machine              : x86_64
  CPU Cores            : 12
  Memory               : 22589 MB
  Distribution         : Amazon Linux 2
  GLIBC                : 2.26
  Model                : vt1.3xlarge

XRT
  Version              : 2.11.722
  Branch               : 2021.1
  Hash                 : d1ecc641da7b8805e778926e3354e8dd3fdaa6a8
  Hash Date            : 2023-01-25 23:37:46
  XOCL                 : 2.11.716, 22cbc46ebc878a0548f6329922807031e2c5d50f
  XCLMGMT              : unknown, unknown

Devices present
  [0000:00:1f.0] : xilinx_u30_gen3x4_base_2 
  [0000:00:1e.0] : xilinx_u30_gen3x4_base_2 
```

It run fine with only one job per node. But when the 2nd job that request the 2nd device on the same node, before the 1st job finishes. it gives error below
It also run fine when we request 2 devices, which is one job per node. 

```
[Mon Jan  6 20:54:31 2025 GMT]
HOST: ats-job-8e88e6806ead404e99aed20c51617535-92kcw
EXE: /opt/xilinx/ffmpeg/bin/ffmpeg
[XMA] ERROR: ffmpeg xma-vcu-encoder VCU_PREINIT failed : 

[XMA] ERROR: ffmpeg xmaencoder Initalization of encoder plugin failed. Return code -1

[MPSOC VCU H264 encoder @ 0x23b4000] [error] encoder error: ERROR: Unable to allocate MPSoC encoder session : ffmpeg pid 231 on device index =  0 cu index = 38
[error] Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
[AVIOContext @ 0x2559000] [verbose] Statistics: 0 seeks, 0 writeouts
[aac @ 0x24fb000] [info] Qavg: 65251.434
[aac @ 0x24fb000] [warning] 2 frames left in the queue on closing
[aac @ 0x27a1000] [info] Qavg: 65251.434
[aac @ 0x27a1000] [warning] 2 frames left in the queue on closing
[aac @ 0x24d1000] [info] Qavg: 65251.434
[aac @ 0x24d1000] [warning] 2 frames left in the queue on closing
[aac @ 0x2708000] [info] Qavg: 65251.434
[aac @ 0x2708000] [warning] 2 frames left in the queue on closing
[AVIOContext @ 0x248d000] [verbose] Statistics: 14659107 bytes read, 6 seeks
[info] Conversion failed!

More logs
```
[Mon Jan  6 22:32:09 2025 GMT]
HOST: ats-job-5bce0393e1994f1b9bd274cad229e3c3-h46h8
EXE: /opt/xilinx/ffmpeg/bin/ffmpeg
[XMA] WARNING: ffmpeg xma-vcu-encoder device warning: !! Warning : Adapting profile to support bitdepth and chroma mode
!! The specified Level is too low and will be adjusted !!

[mp4 @ 0x187f000] [error] Application provided duration: -9223372036854775808 / timestamp: -9223372036854775808 is out of range for mov/mp4 format
[mp4 @ 0x187f000] [warning] pts has no value
[info] Output #0, mp4, to '/mnt/ats/52b0f9df-fecf-45cb-a9d2-2a60fa1b19a4/output1/video_test.mp4':
[info]   Metadata:
[info]     operational_pattern_ul: 060e2b34.04010101.0d010201.01010900
[info]     uid             : 011e1261-0006-4901-802b-182482600231
[info]     generation_uid  : 011e1261-0006-4903-802b-182482600231
[info]     timecode        : 16:36:08:14
[info]     material_package_umid: 0x060A2B340101010501010D4313000000CD0E3939610006800000852482600231
[info]     modification_date: 2023-04-27T14:20:34.000000Z
[info]     product_uid     : 060e2b34-0401-010d-0e15-0056454f0800
[info]     encoder         : Lavf58.76.100
[info]   Stream #0:0: Video: h264, 1 reference frame (avc1 / 0x31637661), xv15(tv, bt709/unknown/unknown, progressive, left), 1080x720 (0x0) [SAR 32:27 DAR 16:9], q=2-31, 3500 kb/s, 0.04 fps, 24k tbn
[info]     Metadata:
[info]       file_package_umid: 0x060A2B340101010501010D4313000000CD0E3938610006800000852482600231
[info]       encoder         : Lavc58.134.100 mpsoc_vcu_h264
[info]   Stream #0:1: Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, mono, fltp (24 bit), delay 1024, 96 kb/s
[info]     Metadata:
[info]       file_package_umid: 0x060A2B340101010501010D4313000000CD0E3938610006800000852482600231
[info]       encoder         : Lavc58.134.100 aac
[info] frame=    1 fps=0.0 q=0.0 size=       0kB time=00:00:00.46 bitrate=   0.9kbits/s speed=1.42x    
[XMA] ERROR: ffmpeg xma-vcu-encoder Error: (run_recv_cmd) VCU_RECEIVE failed: 

[XMA] ERROR: ffmpeg xma-vcu-encoder Error: (process_cmd_response) VCU command failed: 

```

NastoohX, 2025-01-09:
Hi,
Thank you for bringing this issue to our attention. Can you provide further details on the following:

1. How many containers are used?
2. Is XRMD required for load balancing or is application meant to take care of this? If the former, where is XRMD begin run, i.e., on the host or in a container?
3. Did the requested loads exceed the capacity of any one device as reported by xrmadm /opt/xilinx/xrm/test/list_cmd.json?
Cheers,

dannylu88, 2025-01-13:
@NastoohX 
Hi, thanks for the response!

**How many containers are used?**
We have a deployment that act as a webhook to spin up transformation jobs. Transformation job is always a 1:1 mapping to container (k8s pod). Everything works fine when only one transformation job in a VT1.3xlarge (2 devices) node. The problem occur when we have 2 concurrent jobs (in separate containers) in the same VT1.3xlarge node. To put this in short, when a different container in the same node, use another device that is available on the node, problems occur.

**Is XRMD required for load balancing or is application meant to take care of this? If the former, where is XRMD begin run, i.e., on the host or in a container?**
I don't think we have any load balancing here, since transformation job is directly spin up by a deployment that act as a webhook accepter. 

**Did the requested loads exceed the capacity of any one device as reported by xrmadm /opt/xilinx/xrm/test/list_cmd.json?**
We are using the AWS marketplace prebuilt AMI from AMD, so didn't deep dive into the host setup. `xrmadm` is not installed on the host and here is the config.
`[ssm-user@ip-10-216-7-67 bin]$ cat /opt/xilinx/xrm/test/list_cmd.json
{
    "request": {
        "name": "list",
        "requestId": 1
    }
}`

Do we need xrmadm and some extra configuration using it, in order to have things work properly?

NastoohX, 2025-01-15:
Hi,
A few questions to diagnose the issue:

1. For each device, output of xrmadm /opt/xilinx/xrm/test/list_cmd.json # Please note the xrmadm at the beginning of the command
2.  For each device, output of systemctl status xrmd
3.  For each device, output of xbutil examine
4. Means by which transcode jobs are assigned to different devices, e.g., in case of FFmpeg this would be through -xlnx_hwde or -lxlnx_hwde

What we are trying to establish is to see if unintentionally all transcode jobs are being routed to device 0, which is the default device.
Cheers,

ybonnel, 2025-01-17:
Hi @NastoohX ,

Just wanted to be sure to understand, when you are saying "For each device", what do you mean?

To simplify, we have 1 host, and 2 container (each on in a pod associated with a job).

I will assume you mean "For each container".

To run `xrmadm`, I need to know in which rpm package it is available :)
For now, we install `xrt-2.11.722 xilinx-alveo-u30-core xilinx-u30-xcdr xilinx-u30-xffmpeg-4.4.xlnx.2-1` in our container, and I don't have `xrmadm`.
For ffmpeg option, I need to know more because I didn't find any information about those options, which one do you want? Are they options of the h264 codec or global option?

Waiting your answer I will try things, and let you know if I success on xrmadm and ffmpeg options before your answer :)

ybonnel, 2025-01-17:
Oh got it for `xlnx_hwde`.
For now we don't use it, we can, but which component will give use the device on which we need to "affect" the command?

I believed when running ffmpeg without specify the device to use with xlnx_hwde, it will try to use a free device, but maybe I miss something.

ybonnel, 2025-01-17:
Hello,

I tried to reproduce to have logs and it works like expected now... (I was able to have 4 container running ffmpeg in concurrence on 1 vt1.6x without any problem).

My guess is : the error was in fact coming from a concurrence on /dev/fuse which are using on the input and output file of ffmpeg, 1 week ago we changed the management of /dev/fuse on kube to use a device plugin so maybe it fixed the pb.

I will try to reproduce with our old configuration for fuse to confirm, but I guess you can close the issue.

ybonnel, 2025-01-17:
I was able to reproduce with our old configuration for `/dev/fuse`.

Really sorry for the noise, feel free to close the issue.

closed 2025-02-06 by dannylu88