Xilinx/video-sdk #103: xilinx-u30-xffmpeg cannot be installed with xrt-2.11.722 on Amazon Linux 2

Xaeroxe, 2025-02-18:
Commands run

```
amazon-linux-extras install -y epel
yum install -y yum-plugin-versionlock
yum install -y xrt-2.11.722
yum versionlock xrt-2.11.722
yum install -y xilinx-alveo-u30-core xilinx-u30-xffmpeg
```

Error output

```
Error: Package: xilinx-u30-xffmpeg-5.1.4.xlnx.3-1.x86_64 (Amzn2Artifactory)
           Requires: xilinx-u30-xcdr >= 3.1
           Installing: xilinx-u30-xcdr-3.0.0-1.x86_64 (Amzn2Artifactory)
               xilinx-u30-xcdr = 3.0.0-1
```

Base AMI details

```json
{
    "Images": [
        {
            "Architecture": "x86_64",
            "CreationDate": "2025-02-01T08:34:55.000Z",
            "ImageId": "ami-04681163a08179f28",
            "ImageLocation": "amazon/amzn2-ami-kernel-5.10-hvm-2.0.20250201.0-x86_64-gp2",
            "ImageType": "machine",
            "Public": true,
            "OwnerId": "137112412989",
            "PlatformDetails": "Linux/UNIX",
            "UsageOperation": "RunInstances",
            "State": "available",
            "BlockDeviceMappings": [
                {
                    "DeviceName": "/dev/xvda",
                    "Ebs": {
                        "DeleteOnTermination": true,
                        "SnapshotId": "snap-05d22b8ee1da53e2e",
                        "VolumeSize": 8,
                        "VolumeType": "gp2",
                        "Encrypted": false
                    }
                }
            ],
            "Description": "Amazon Linux 2 Kernel 5.10 AMI 2.0.20250201.0 x86_64 HVM gp2",
            "EnaSupport": true,
            "Hypervisor": "xen",
            "ImageOwnerAlias": "amazon",
            "Name": "amzn2-ami-kernel-5.10-hvm-2.0.20250201.0-x86_64-gp2",
            "RootDeviceName": "/dev/xvda",
            "RootDeviceType": "ebs",
            "SriovNetSupport": "simple",
            "VirtualizationType": "hvm",
            "DeprecationTime": "2025-07-01T00:00:00.000Z"
        }
    ]
}
```

Mudpuppy12, 2025-04-06:
I had similar issue trying to patch: 

In order to patch the system I did the following:

1. Remove the older libraries as per documentation https://xilinx.github.io/video-sdk/v3.0/getting_started_on_vt1.html
2. sudo yum update the system.
3.sudo yum install xilinx-alveo-u30-core
4. Then

sudo yum install xilinx-alveo-u30-ffmpeg
sudo yum install xilinx-alveo-u30-gstreamer
sudo yum install xilinx-alveo-u30-examples

5. Reboot.

However once patched, same issue as above:
using ami-03427f35504cbb6e5

yum versionlocked as described in documentation

# Added locks on Thu Jan  9 19:04:20 2025
0:xrt-2.11.722-1.*
versionlock.list (END)

[root@ip-10-0-0-214 ~]# yum update
Loaded plugins: dkms-build-requires, extras_suggestions, langpacks, priorities, update-motd, versionlock
amzn2-core                                                                         | 3.6 kB  00:00:00
230 packages excluded due to repository priority protections
Resolving Dependencies
--> Running transaction check
---> Package xilinx-u30-xffmpeg.x86_64 0:4.4.xlnx.2-1 will be updated
--> Processing Dependency: xilinx-u30-xffmpeg = 4.4.xlnx.2 for package: xilinx-alveo-u30-ffmpeg-3.0.0-1.x86_64
---> Package xilinx-u30-xffmpeg.x86_64 0:5.1.4.xlnx.3-1 will be an update
--> Processing Dependency: xilinx-u30-xcdr >= 3.1 for package: xilinx-u30-xffmpeg-5.1.4.xlnx.3-1.x86_64
---> Package xilinx-u30-xmpsoccodecs.x86_64 0:1.2.0-1 will be updated
--> Processing Dependency: xilinx-u30-xmpsoccodecs = 1.2.0 for package: xilinx-u30-xcdr-3.0.0-1.x86_64
---> Package xilinx-u30-xmpsoccodecs.x86_64 0:1.3.0-1 will be an update
--> Finished Dependency Resolution
Error: Package: xilinx-u30-xffmpeg-5.1.4.xlnx.3-1.x86_64 (Amzn2Artifactory)
           Requires: xilinx-u30-xcdr >= 3.1
           Installed: xilinx-u30-xcdr-3.0.0-1.x86_64 (@Amzn2Artifactory)
               xilinx-u30-xcdr = 3.0.0-1
Error: Package: xilinx-alveo-u30-ffmpeg-3.0.0-1.x86_64 (@Amzn2Artifactory)
           Requires: xilinx-u30-xffmpeg = 4.4.xlnx.2
           Removing: xilinx-u30-xffmpeg-4.4.xlnx.2-1.x86_64 (@Amzn2Artifactory)
               xilinx-u30-xffmpeg = 4.4.xlnx.2-1
           Updated By: xilinx-u30-xffmpeg-5.1.4.xlnx.3-1.x86_64 (Amzn2Artifactory)
               xilinx-u30-xffmpeg = 5.1.4.xlnx.3-1
Error: Package: xilinx-u30-xcdr-3.0.0-1.x86_64 (@Amzn2Artifactory)
           Requires: xilinx-u30-xmpsoccodecs = 1.2.0
           Removing: xilinx-u30-xmpsoccodecs-1.2.0-1.x86_64 (@Amzn2Artifactory)
               xilinx-u30-xmpsoccodecs = 1.2.0-1
           Updated By: xilinx-u30-xmpsoccodecs-1.3.0-1.x86_64 (Amzn2Artifactory)
               xilinx-u30-xmpsoccodecs = 1.3.0-1
 You could try using --skip-broken to work around the problem
 You could try running: rpm -Va --nofiles --nodigest

sudo yum remove xvbm xilinx-u30-xvbm xrmu30decoder xrmu30scaler xrmu30encoder xmpsoccodecs xmultiscaler xlookahead xmaapps xmapropstojson xffmpeg launcher jobslotreservation xcdr

sudo yum remove xrm xilinx-container-runtime xilinx-xvbm xilinx-u30-xrm-decoder xilinx-u30-xrm-encoder xilinx-u30-xrm-multiscaler xilinx-u30-xma-multiscaler xilinx-u30-xlookahead xilinx-u30-xmpsoccodecs xilinx-u30-xma-apps xilinx-u30-xmapropstojson xilinx-u30-xffmpeg xilinx-u30-launcher xilinx-u30-jobslotreservation xilinx-u30-xcdr xilinx-u30-gstreamer-1.16.2 xilinx-u30-vvas xilinx-sc-fw-u30 xilinx-u30-gen3x4-base xilinx-u30-gen3x4-validate

[root@ip-10-0-0-214 ~]# sudo su -
Last login: Sun Apr  6 17:24:39 UTC 2025 on pts/0
[root@ip-10-0-0-214 ~]# yum update
Loaded plugins: dkms-build-requires, extras_suggestions, langpacks, priorities, update-motd, versionlock
Amzn2Artifactory                                                                   | 1.4 kB  00:00:00
Artifactory                                                                        | 1.4 kB  00:00:00
BaseArtifactory                                                                    | 1.4 kB  00:00:00
amzn2-core                                                                         | 3.6 kB  00:00:00
amzn2extra-docker                                                                  | 2.9 kB  00:00:00
[... 217 lines cut ...]
---> Package xilinx-u30-xffmpeg.x86_64 0:4.4.xlnx.2-1 will be updated
--> Processing Dependency: xilinx-u30-xffmpeg = 4.4.xlnx.2 for package: xilinx-alveo-u30-ffmpeg-3.0.0-1.x86_64
---> Package xilinx-u30-xffmpeg.x86_64 0:5.1.4.xlnx.3-1 will be an update
--> Processing Dependency: xilinx-u30-xcdr >= 3.1 for package: xilinx-u30-xffmpeg-5.1.4.xlnx.3-1.x86_64
---> Package xilinx-u30-xmpsoccodecs.x86_64 0:1.2.0-1 will be updated
--> Processing Dependency: xilinx-u30-xmpsoccodecs = 1.2.0 for package: xilinx-u30-xcdr-3.0.0-1.x86_64
--> Finished Dependency Resolution
Error: Package: xilinx-u30-xffmpeg-5.1.4.xlnx.3-1.x86_64 (Amzn2Artifactory)
           Requires: xilinx-u30-xcdr >= 3.1
           Installed: xilinx-u30-xcdr-3.0.0-1.x86_64 (@Amzn2Artifactory)
               xilinx-u30-xcdr = 3.0.0-1
Error: Package: xilinx-alveo-u30-ffmpeg-3.0.0-1.x86_64 (@Amzn2Artifactory)
           Requires: xilinx-u30-xffmpeg = 4.4.xlnx.2
           Removing: xilinx-u30-xffmpeg-4.4.xlnx.2-1.x86_64 (@Amzn2Artifactory)
               xilinx-u30-xffmpeg = 4.4.xlnx.2-1
           Updated By: xilinx-u30-xffmpeg-5.1.4.xlnx.3-1.x86_64 (Amzn2Artifactory)
               xilinx-u30-xffmpeg = 5.1.4.xlnx.3-1
Error: Package: xilinx-u30-xcdr-3.0.0-1.x86_64 (@Amzn2Artifactory)
           Requires: xilinx-u30-xmpsoccodecs = 1.2.0
           Removing: xilinx-u30-xmpsoccodecs-1.2.0-1.x86_64 (@Amzn2Artifactory)
               xilinx-u30-xmpsoccodecs = 1.2.0-1
           Updated By: xilinx-u30-xmpsoccodecs-1.3.0-1.x86_64 (Amzn2Artifactory)
               xilinx-u30-xmpsoccodecs = 1.3.0-1
 You could try using --skip-broken to work around the problem
 You could try running: rpm -Va --nofiles --nodigest

Mudpuppy12, 2025-04-06:
[root@ip-10-0-0-214 pluginconf.d]# yum info xilinx-u30-xcdr
Loaded plugins: dkms-build-requires, extras_suggestions, langpacks, priorities, update-motd, versionlock
230 packages excluded due to repository priority protections
Installed Packages
Name        : xilinx-u30-xcdr
Arch        : x86_64
Version     : 3.0.0
Release     : 1
Size        : 13 M
Repo        : installed
From repo   : Amzn2Artifactory
Summary     : Xilinx Transcoder Application for MPSoC Device
License     : unknown
Description : DESCRIPTION
            : ===========
            :
            : This is an installer created using CPack (http://www.cmake.org). No additional installation
            : instructions provided.

[root@ip-10-0-0-214 pluginconf.d]# yum update  xilinx-u30-xcdr
Loaded plugins: dkms-build-requires, extras_suggestions, langpacks, priorities, update-motd, versionlock
230 packages excluded due to repository priority protections
No packages marked for update

Mudpuppy12, 2025-04-06:
IS this card still supported anymore?

Mudpuppy12, 2025-04-06:
[ec2-user@ip-10-0-0-214 ~]$ rpm -qa | grep ffmpeg
xilinx-alveo-u30-ffmpeg-3.0.0-1.x86_64
xilinx-u30-xffmpeg-4.4.xlnx.2-1.x86_64
[ec2-user@ip-10-0-0-214 ~]$ yum info xilinx-u30-xffmpeg-5.1.4.xlnx.3-1.x86_64
Loaded plugins: dkms-build-requires, extras_suggestions, langpacks, priorities, update-motd, versionlock
230 packages excluded due to repository priority protections
Available Packages
Name        : xilinx-u30-xffmpeg
Arch        : x86_64
Version     : 5.1.4.xlnx.3
Release     : 1
Size        : 8.2 M
Repo        : Amzn2Artifactory
Summary     : FFmpeg with Xilinx Accelerators
License     : unknown
Description : DESCRIPTION
            : ===========
            : This is an installer created using CPack (http://www.cmake.org). No additional installation
            : instructions provided.

Xaeroxe, 2025-04-06:
For what it’s worth my team already has plans to decommission our fleet of these. We’ve been unimpressed with the software support. It’s a shame because the hardware has been great.

Mudpuppy12, 2025-04-06:
Did you forget to push     Requires: xilinx-u30-xcdr >= 3.1 to the repo

Mudpuppy12, 2025-04-06:
Yeah, The support was great a year ago. Then they got purchased by AMD, then.......